home-assistant/operating-system #3528: Home Assistant OS 13.0 Breaks Hikvision and all Streaming Generic Cameras

bkarakashev, 2024-08-14:
### Describe the issue you are experiencing

After updating to OS 13.0 All my cameras stop working.
I was using Hikvision and also Generic Camera integration.

After rolling back to 12.4 everything starts working as normal.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

13.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Proxmox using HAOS VM
8GB RAM
i5 CPU

### Steps to reproduce the issue

1. Update to OS 13.0

### Anything in the Supervisor logs that might be useful for us?

```txt
Error from stream worker: Error demuxing stream while finding first packet (ERRORTYPE_110, Operation timed out, rtsp://****:****@192.168.0.104:554/Streaming/channels/301)

Error from stream worker: Error opening stream (INVALIDDATA, Invalid data found when processing input, rtsp://****:****@192.168.0.104:554/Streaming/channels/101)
```

### Anything in the Host logs that might be useful for us?

```txt
Error from stream worker: Error demuxing stream while finding first packet (ERRORTYPE_110, Operation timed out, rtsp://****:****@192.168.0.104:554/Streaming/channels/301)

Error from stream worker: Error opening stream (INVALIDDATA, Invalid data found when processing input, rtsp://****:****@192.168.0.104:554/Streaming/channels/101)
```

### System information

## System Information

version | core-2024.8.1
installation_type | Home Assistant OS
hassio | true
os_version | 6.6.33-haos
arch | x86_64

host_os | Home Assistant OS 12.4
supervisor_version | supervisor-2024.08.0

tjmv, 2024-08-14:
Just wanted to post the same thing. Basically the same specs and also using Proxmox (latest update).

Besides the camera feeds not being available, I also checked a 20MB file download and that didn't work normally either. The MQTT addon didn't receive much data either. Pings were working and occasional packet got through, but not much more than that. A downgrade to 12.4 got things back in working order. Didn't have to restore any backups for this.

sairon, 2024-08-14:
It _could_ be [this kernel regression](https://bugzilla.kernel.org/show_bug.cgi?id=219129). There's 6.6.46 kernel that will be included in today's dev build. As I'm not able to reproduce the issues myself, I will much appreciate if anyone here could check if it's fixed by that.

Oldefa1985, 2024-08-14:
I have exactly the same issue, all Hikvision cameras are not getting streams - sensors are unavailable and inactive (switching between both)

Error from stream worker: Error demuxing stream while finding first packet (ERRORTYPE_110, Operation timed out, rtsp://****:****@172.16.30.2:554/Streaming/channels/101)

sairon, 2024-08-15:
Please test with the latest dev (`ha os update --version 13.1.dev20240815`), it contains latest kernel which fixes the above regresison.

Oldefa1985, 2024-08-15:
[image]

Cant seem to make it work, github was down yesterday when i tried

sairon, 2024-08-15:
@Oldefa1985 There were some issues with the Cloudflare cache, can you try again?

tjmv, 2024-08-15:
Still the same 404 issue for me.

qwerk, 2024-08-15:
Same issue here but with blueiris integration. reverting to 12.4 solved it.

Oldefa1985, 2024-08-15:
@sairon im still getting error 404 when executing the commandline - yesterday it was a different error code since the service was unavailable.

sairon, 2024-08-15:
That's strange, can you also check Supervisor logs (`ha su logs`) for the exact error?

Oldefa1985, 2024-08-15:
[image]

sairon, 2024-08-15:
@Oldefa1985 Oh, I realize what's wrong. It's not possible to update to dev OS if you're on stable channel. You'll need to switch to beta at least:
```
ha su options --channel=dev
ha su reload
```
Then wait until you see `Fetching update data from https://version.home-assistant.io/beta.json` in `ha su logs`. Afterwards the update should pass.

Sorry for not realizing that sooner. You don't need to switch Core to beta version - you might just need to upgrade Supervisor first if it complains about that. After finishing the OS upgrade, you can switch back to stable using `ha su options --channel=stable`, so you won't be notified about beta releases.

Oldefa1985, 2024-08-15:
@sairon 

[image]

it now works with the camera streams 

When can i expect HA to be updated in the stable channel? :)

sairon, 2024-08-15:
@Oldefa1985 Great, thanks for confirmation! It will take a couple of days to make it to stable but as it's going to contain essentially the same commits as the current dev, you should be fine staying on this dev build - it's not inferior or less stable than a tagged stable build.

Oldefa1985, 2024-08-15:
thank you @sairon !!

wasted most of my evening yesterday to figure out that it was a HAOS problem - i changed something in my router so i thought i fucked up somewhere in between - when i realized that it wasnt the network, i looked at HA and found this post :)

i changed my channel back to stable (so i suppose it will upgrade to the stable version next time when its released), since i still have the .dev version.

Really great to see the developer help out in this manner - thank you again :)

qwerk, 2024-08-15:
I also had to update the supervisor to a DEV version. I can confirm blueiris is also working again.
[image]

Oldefa1985, 2024-08-15:
me too @qwerk - it asked for it to be updated before it would work - forgot to include that :)

crazyasyou, 2024-08-16:
OS 13.0 broke my Blue Iris MQTT to HA - Other MQTT from my Tasmota devices was working still so I spend ages troubleshooting Blue Iris - I'm running my prod HA in a proxmox VM  - Tested MQTT BI>HA on my dev HA RasPi 4, and it worked, a roll back to 12.4 on the proxmox VM restored my BI to HA MQTT

closed 2024-11-22 by github-actions[bot]