home-assistant/operating-system #3530: HAOS 13.0 breaks import of KNX Project File

zaphood1967, 2024-08-14:
### Describe the issue you are experiencing

After updating to HAS 13.0 the import of KNX Project files takes forver and never finishes. 
Deleting the project file changes nothing, nor does rebooting HA or using a different ETS Project. 
After Rollback to 12.4 , import works again as expected. 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

13.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

HAOS as VM unter Proxmox 8.2.3.

### Steps to reproduce the issue

1. Install 13.0
2. Try to import an ETS (KNX) Project into HA

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing special, seems the process is locked
```

### Anything in the Host logs that might be useful for us?

```txt
nothing of relevance either
```

### System information

## System Information

version | core-2024.8.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.33-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1456
Downloaded Repositories | 72

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 21. März 2025 um 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | eu-central-1-5.ui.nabu.casa
certificate_status | ready
instance_id | f55559b461a44114b45df9211966e4d4
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 62.3 GB
disk_used | 50.7 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Spotify Connect (0.13.0), File editor (5.8.0), Samba share (12.3.2), Terminal & SSH (9.14.0), Log Viewer (0.17.0), Studio Code Server (5.15.0), Piper (1.5.2), ESPHome (2024.7.3), TasmoAdmin (0.30.4), Mosquitto broker (6.4.1), Network UPS Tools (0.13.1), AppDaemon (0.16.6), Music Assistant Server (2.1.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 44
views | 18
mode | storage

</details>

<details><summary>MercedesME 2020</summary>

api_endpoint_reachable | ok
-- | --
websocket_connection_state | connected
cars_connected | 1
version | v0.16.0

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 9. August 2024 um 20:12
-- | --
current_recorder_run | 14. August 2024 um 12:02
estimated_db_size | 2432.03 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

..

zaphood1967, 2024-08-15:
This is weird. Out of curiosity, I updated HAOS via cli and tested the ETS-import again - now it works. Seems the update of HAOS via GUI does something different which then breaks the import of the ETS file? (testet it 4 times yesterday via GUI)

zaphood1967, 2024-08-15:
Solved

closed 2024-08-15 by zaphood1967

sairon, 2024-08-19:
> Seems the update of HAOS via GUI does something different which then breaks the import of the ETS file?

It does not, the issue is probably caused by some other factor, as the KNX import itself has little to do with the OS. Could be caused by #3532 though, which caused intermittent network issues on some VMs.