home-assistant/operating-system #3531: HA OS 13.0 breaks usb Aeotec Z-wave stick?

laageide, 2024-08-14:
### Describe the issue you are experiencing

Aeotec Z-wave gen5+ stick dissapears from system after updating to HA OS 13.0.
It is not detected as connected anymore.
Downgrading to HA OS 12.4 solves the problem and z-stick is working as normal.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

13.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Using an dell laptop with deconz zigbee, aeotec z-stick gen5+ and skyconnect dongle connected. 

### Steps to reproduce the issue

1. updated HA OS 12.4 to 13.0
2. system restarted and failed to connect to z wave stick.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Did not backup logs before reverting to version 12.4 
#sorry about that 🥴
```

### Anything in the Host logs that might be useful for us?

```txt
Did not backup logs before reverting to version 12.4 
#sorry about that 🥴
```

### System information

this is system information with version 12.4 installed

## System Information

version | core-2024.8.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.33-haos
arch | x86_64
timezone | Europe/Oslo
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4826
Installed Version | 1.34.0
Stage | running
Available Repositories | 1391
Downloaded Repositories | 31

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 468.7 GB
disk_used | 16.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | deCONZ (6.23.0), Studio Code Server (5.15.0), Z-Wave JS UI (3.10.0), Terminal & SSH (9.14.0), AdGuard Home (5.1.1), Mosquitto broker (6.4.1), Node-RED (18.0.5), Samba share (12.3.2), Silicon Labs Multiprotocol (2.4.5), Matter Server (6.4.1), AppDaemon (0.16.6), Music Assistant Server (2.1.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 14
views | 18
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 6, 2024 at 20:40
-- | --
current_recorder_run | August 14, 2024 at 23:07
estimated_db_size | 1094.44 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

phaase1, 2024-08-15:
I have the same problem using a Raspberry 4. 
 
Question to "laageide" , how can I downgrade to HA OS 12.4?

sairon, 2024-08-15:
The root cause of some USB issues was not resolved in Raspberry Pi kernel - however, those were supposedly present on RPi 3 only. Moreover, there was no change between 12.4 and 13.0 for Raspberry Pi that could change the behavior in either way. The Aeotec Z-wave stick has enumeration issues though (it's a hardware defect) and sometimes requires to be unplugged multiple times (or a reboot can also help), so I think this is the case as well. Hard to tell without any logs though.

laageide, 2024-08-15:
I did not try to physically disconnect/reconnect the z-wave stick since i did the update remotely. But I tried to reboot the entire system without any luck. Can give it another go when I'm back home and have access to the hardware.

phaase1, 2024-08-15:
@laageide: Thank you very much!

@sairon:  Re-booting and swapping USB slots didn't change anything.

Here are Core and Z-Wave logs I got.

[a0d7b954_zwavejs2mqtt_2024-08-15T09-17-55.874Z.log](https://github.com/user-attachments/files/16623764/a0d7b954_zwavejs2mqtt_2024-08-15T09-17-55.874Z.log)

[home-assistant_2024-08-15T09-16-53.841Z.log](https://github.com/user-attachments/files/16623779/home-assistant_2024-08-15T09-16-53.841Z.log)

JoDoTmm, 2024-08-15:
Same problem here:

After OS update 13:  failed to setup in Devices and services. Failed to connect to host core-zwave-js:3000 ssl:default [None].

Unplugging and retrying / rebooting does not solve the problem. 

I'm on a Home Assistant Yellow

phaase1, 2024-08-15:
@sairon: To build on what you said, just downgraded to 12.4 and had the same problem after reboot. Swapped again USB slot of Aeotec Z-wave gen5+ stick and now all is running fine again after reboot.
So obviously the problem is the stick with the issues described above.

sairon, 2024-08-15:
Host would be needed here. Ideally `ha host logs -b0 -n1000` to get the everything since boot.

JoDoTmm, 2024-08-15:
I had the same problem,

Installed 12.4 and problem persisted. Turned out the Integration looking for the wrong USB port. Used the configure option and selected the right USB port. No problems  now on 12.4. Will try the same on 13.0.

phaase1, 2024-08-15:
@sairon 
[host_2024-08-15T09-33-42.236Z.log](https://github.com/user-attachments/files/16624163/host_2024-08-15T09-33-42.236Z.log)

laageide, 2024-08-15:
An update from my side here : I just updated to HA OS 13.0 again and everything is woring perfectly on my system.
I did not do anything to Aeotec Z-stick or the Z-wave JS integration, it just worked.

JoDoTmm, 2024-08-15:
> "Do you want to use the Z-Wave JS Supervisor add-on?"
> 
> When/where do you see this option?

Settings->Devices & services Select Z-Wave, icecream menu (vertical dots and select (re)configure). 
You now get a floating frame with " Select connection method, do you want to use the z-wave JS Supervisor add-on? and a check box to select the option "use the z-wave JS supervisor add-on" , make sure the check box is selected and press submit. 
In the new floating frame select the correct USB device path (mine is : ttyACM0 - /dev/serial/by-id/usb-0658_0200-if00, s/n: n/a - 0658:0200).

Regards,

John.

RonaldRegister, 2024-11-11:
any news if this is fixed with 13.2 ?
I'm stuck on 12.2 and afraid to update to anything more recent

SebuZet, 2024-11-18:
Im having similar issue with my dongle on intel machine. It doesn't work after reboot. I have to unplug the device and plug it again - then it works.

cjlee89, 2025-02-17:
Is this similar to #3362 ?

SebuZet, 2025-02-18:
The issue still exists in current version.

sairon, 2025-03-28:
The Z-Wave stick is known to have issues as already stated in the documentation. The previous OS versions had patches that tried to address those, but those were now dropped in 15.0. There are some workarounds (e.g. using a power USB hub) but the stick is not a properly compliant USB device so it can't be addressed on the OS level. See also my comment here: https://github.com/home-assistant/operating-system/issues/3963#issuecomment-2746133403

closed 2025-03-28 by sairon

SebuZet, 2025-03-29:
I have connected this stick and zigbee one trough USB hub and it helps.

LeSpocky, 2025-03-29:
> I have connected this stick and zigbee one trough USB hub and it helps.

Slightly off-topic, but I found connecting those sticks to a USB hub with a cable like a meter away from the RaspberryPi is useful for another reason: the radio connection is much more stable, especially with Z-Wave. I suspect less electromagnetic disturbance from the Pi itself.