home-assistant/operating-system #3532: Kernel Issue - Bad GSO

greggitter, 2024-08-15:
### Describe the issue you are experiencing

After updating to HAOS 13.0, this error seems to have started presenting itself along with very slow restart time.  This wasn't obvious until the proxmox host was restarted.  From the log....

[image]

This bug was replied on the forums, including here as possible issue/cause:

https://bugzilla.kernel.org/show_bug.cgi?id=219129

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.44-haos, Home Assistant OS 13.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Proxmox 8 (latest)

### Steps to reproduce the issue

1.  Reboot proxmox hardware, problem occurs...but messages continue (see above)
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing related.
```

### Anything in the Host logs that might be useful for us?

```txt
2024-08-14 23:52:39.374 homeassistant kernel: enp6s18: bad gso: type: 1, size: 19
2024-08-14 23:52:40.355 homeassistant kernel: enp6s18: bad gso: type: 1, size: 137
2024-08-14 23:52:41.353 homeassistant kernel: enp6s18: bad gso: type: 1, size: 137
2024-08-14 23:52:45.689 homeassistant kernel: enp6s18: bad gso: type: 1, size: 48
2024-08-14 23:52:46.354 homeassistant kernel: enp6s18: bad gso: type: 1, size: 141
2024-08-14 23:52:47.853 homeassistant kernel: enp6s18: bad gso: type: 1, size: 141
2024-08-14 23:52:49.353 homeassistant kernel: enp6s18: bad gso: type: 1, size: 141
2024-08-14 23:52:49.354 homeassistant kernel: enp6s18: bad gso: type: 1, size: 141
2024-08-14 23:52:50.853 homeassistant kernel: enp6s18: bad gso: type: 1, size: 141
2024-08-14 23:52:52.353 homeassistant kernel: enp6s18: bad gso: type: 1, size: 141
2024-08-14 23:52:55.379 homeassistant kernel: enp6s18: bad gso: type: 1, size: 61
2024-08-14 23:53:05.675 homeassistant kernel: enp6s18: bad gso: type: 1, size: 106
2024-08-14 23:53:06.355 homeassistant kernel: enp6s18: bad gso: type: 1, size: 133
2024-08-14 23:53:06.356 homeassistant kernel: enp6s18: bad gso: type: 1, size: 141
2024-08-14 23:53:07.354 homeassistant kernel: enp6s18: bad gso: type: 1, size: 133
2024-08-14 23:53:07.355 homeassistant kernel: enp6s18: bad gso: type: 1, size: 141
2024-08-14 23:53:07.373 homeassistant kernel: enp6s18: bad gso: type: 1, size: 52
2024-08-14 23:53:08.355 homeassistant kernel: enp6s18: bad gso: type: 1, size: 137
2024-08-14 23:53:09.354 homeassistant kernel: enp6s18: bad gso: type: 1, size: 137
2024-08-14 23:53:10.354 homeassistant kernel: enp6s18: bad gso: type: 1, size: 137
2024-08-14 23:53:11.354 homeassistant kernel: enp6s18: bad gso: type: 1, size: 137
2024-08-14 23:53:12.354 homeassistant kernel: enp6s18: bad gso: type: 1, size: 137
2024-08-14 23:53:12.355 homeassistant kernel: enp6s18: bad gso: type: 1, size: 137
2024-08-14 23:53:14.354 homeassistant kernel: enp6s18: bad gso: type: 1, size: 137
2024-08-14 23:53:18.354 homeassistant kernel: enp6s18: bad gso: type: 1, size: 137
2024-08-14 23:53:23.407 homeassistant kernel: enp6s18: bad gso: type: 1, size: 106
2024-08-14 23:53:24.357 homeassistant kernel: enp6s18: bad gso: type: 1, size: 133
2024-08-14 23:53:24.357 homeassistant kernel: enp6s18: bad gso: type: 1, size: 141
[... 136 lines cut ...]
version_api | ok
installed_addons | File editor (5.8.0), Mosquitto broker (6.4.1), TasmoAdmin (0.30.4), ESPHome (2024.7.3), MQTT Explorer (browser-1.0.1), SQLite Web (4.2.0), Home Assistant Google Drive Backup (0.112.1), Advanced SSH & Web Terminal (18.0.0), Terminal & SSH (9.14.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 8
views | 12
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 20, 2024 at 9:21 PM
-- | --
current_recorder_run | August 14, 2024 at 6:52 PM
estimated_db_size | 1772.25 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

sairon, 2024-08-15:
This might be potential root cause of #3517 and #3528 and was fixed by #3529. We'll do a bit more of testing and release 13.1 with the fixed kernel ASAP.

closed 2024-08-15 by sairon

vampywiz17, 2024-08-16:
I also face these issue, but different simtome.. i can not able to upload anything that use TCP port... smb, scp, web etc...

it also affect my other server (Alpine linux) that run 6.6.44 same issue. 6.6.45 also bugged, 6,6,46 (edge versin in alpine) are working well again :)