home-assistant/operating-system #3533: HA OS 13 fails to install on ODROID-XU4

Frans1960, 2024-08-15:
### Describe the issue you are experiencing

After activating the installation from the HA settings/update the system start with the update and is busy for a long time and noting is working any more.
After a manual reboot I get the previous installed HA OS 12.4

### What operating system image do you use?

odroid-xu4 (Hardkernel ODROID-XU4)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Moved the system from SDCARD to a SSD by USB drive

### Steps to reproduce the issue

1. Start the installation of HA OS 13
2. Manual reboot
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No useful logs
```

### Anything in the Host logs that might be useful for us?

```txt
No useful logs
```

### System information

## System Information

version | core-2024.8.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.33-haos
arch | armv7l
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4973
Installed Version | 1.34.0
Stage | running
Available Repositories | 1389
Downloaded Repositories | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 58.7 GB
disk_used | 11.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | odroid-xu4
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Samba share (12.3.2), Mosquitto broker (6.4.1), Node-RED (18.0.5), InfluxDB (5.0.0), MariaDB (2.7.1), phpMyAdmin (0.9.1), Grafana (10.0.0), HAOS-SBFspot (2024.7.1), Advanced SSH & Web Terminal (18.0.0), ESPHome (2024.7.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 2
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 8 augustus 2024 om 08:15
-- | --
current_recorder_run | 15 augustus 2024 om 10:36
estimated_db_size | 234.18 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

Wyman42, 2024-08-17:
same issue with HA on Raspberry pi3 and HA OS 12.3

sairon, 2024-08-21:
Can you connect an HDMI display and check what happens after the upgrade, or after running `ha os boot-slot other` (if `ha os info` confirms OS 13.0 is indeed in the other slot)?

Frans1960, 2024-08-21:
Hello 

 

 

After running ha os boot-slot other, I got this response: Post "http://supervisor/os/boot-slot": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

 

 

 

Met vriendelijke groet,

 

Frans Kleisman

Jan van Galenstraat 8

7572AZ Oldenzaal

Mob +316 25072748

 ***@***.***> ***@***.***

Frans1960, 2024-08-21:
Hello Ian,

I have an update and after running the ha os info command:

 

~ # ha os info

board: odroid-xu4

boot: B

boot_slots:

  A:

    state: inactive

    status: good

    version: "12.4"

  B:

    state: booted

    status: good

    version: "13.0"

data_disk: C400-MTFDDAC064MAM-000000001241091B5CA7

update_available: false

version: "13.0"

version_latest: "13.0"

 

Hope this heps

sairon, 2024-08-21:
I'm a bit confused from your posts what exactly went through or not but from this information it seems that 13.0 is now running fine:
```
version: "13.0"
version_latest: "13.0"
```

Frans1960, 2024-08-21:
Hello Ian,

No with OS13, I get a lot of pages that are not loading: Error while loading page automation., Error while loading page automation., Error while loading page state. Etc.

How can I solve that?

 

 

 

Met vriendelijke groet,

 

Frans Kleisman

Jan van Galenstraat 8

7572AZ Oldenzaal

Mob +316 25072748

 ***@***.***> ***@***.***

Frans1960, 2024-08-22:
Yesterday I upgraded to 13.1 and now it’s running fine. With 13 I had issues as described in my previous mail

sairon, 2024-08-22:
> No with OS13, I get a lot of pages that are not loading: Error while loading page automation., Error while loading page automation., Error while loading page state. Etc.

That doesn't sound like an OS issue, rather than something that the reboot after the update triggered. However, as it's resolved now, I'm closing this issue.

closed 2024-08-22 by sairon

Frans1960, 2024-08-22:
The issues are solved with installation of OS 13.1 and all is running fine again