home-assistant/operating-system #3534: HA OS 13.0 task kthreadd blocked for more than 120 seconds

rrooggiieerr, 2024-08-15:
### Describe the issue you are experiencing

Since I upgraded my Raspberry Pi 4 to the latest HA OS 13.0 the system freezes regularly with the message

```
task kthreadd:2 blocked for more than 120 seconds.
```

[image]

Once frozen I can still ping my HA server, but can't SSH into it anymore

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

3.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Raspberry Pi 4 with 1 GB memory and WD Purple SC QD101 microSDXC 64 GB storage, original power supply and powered USB hub, Phoscon RaspBee II Zigbee module

### Steps to reproduce the issue

1. Upgrade to latest HA OS 13.0
2. After 5 to 10 minutes the system freezes
3. After about 20 minutes the system restarts and it starts over again

about 1 in 4 times the system boots up correctly and I can access HA, but then after some hours it crashes again

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't access my system
```

### Anything in the Host logs that might be useful for us?

```txt
Can't access my system
```

dpgh947, 2024-08-15:
I just hit this. I got in to the CLI and then discovered via command "ha os info" that there are 2 boot slots that it flip flops between when an update is applied. If it fails to boot 3 times it reverts to the other slot, but this problem is obviously letting the os come up far enough to prevent that triggering. I then used command "ha os boot-slot other" to force it to the other slot, which had previous 12.4 in it. The system then came up OK.

EastArctica, 2024-08-16:
I'm dealing with this too! Crashing every few minutes. Occasionally I'll also see it throw up a stack trace too but I don't know of any way to have those traces be saved. I'll try the boot slot change now...

rrooggiieerr, 2024-08-17:
@dpgh947 Thanks for looking into this, I have just rolled back to 12.4 using `ha os update --version 12.4`

Using your proposed `ha os info` I also see 2 boot slots:
```
board: rpi4-64
boot: B
boot_slots:
  A:
    state: inactive
    status: good
    version: "13.0"
  B:
    state: booted
    status: good
    version: "12.4"
data_disk: WX64G-0x14ec9103
update_available: true
version: "12.4"
version_latest: "13.0"
```

DeXter3306, 2024-08-19:
Thanks for the tip of reverting back to 12.4 !
Got the same issue on RPi 3B+ with USB SSD (also tried with a USB HDD).
When the Pi is hanging like that, is it possible to extract some logs or so ? Really want to help on this case.

sairon, 2024-08-19:
The error messages basically mean the system is too busy to handle kernel tasks. Can you watch memory and CPU usage at the [Hardware](https://my.home-assistant.io/create-link/?redirect=hardware) page to check whether it's not hitting limits before it becomes unresponsive? The common denominator for this report and @dipseth's in the other issue is RPi with 1GB RAM, which is sufficient only for very simple HA setup.

dpgh947, 2024-08-19:
Mine is a pi4 with 2gb, never had a problem before.

EastArctica, 2024-08-19:
> The error messages basically mean the system is too busy to handle kernel tasks. Can you watch memory and CPU usage at the [Hardware](https://my.home-assistant.io/create-link/?redirect=hardware) [...]

I have pi4 with 4GB and am experiencing this. Although, when recording both `top` from the os and the info from the hardware tab I was unable to get the device to crash after 1 hour and 20 minutes. Perhaps I was only encountering this when I was doing something on home assistant?

DeXter3306, 2024-08-19:
> The error messages basically mean the system is too busy to handle kernel tasks. Can you watch memory and CPU usage at the [Hardware](https://my.home-assistant.io/create-link/?redirect=hardware) [...]

I understand that the RPi 3B+ is quite old and after 5 years of loyal duty I should consider to replace it but still, I'm trying to understand why it happened after the 13.0 update and it's not occuring when I run 12.4.

The ram shortage was quite mitigated by my bigger SWAP config.

In order to downgrade to 12.4, I had to stop every add-ons, and avoid connecting to the UI (otherwise the crash would occur). When I downgraded to 12.4, I re-enabled the add-ons and I have 0 issues since then.

EastArctica, 2024-08-19:
> I understand that the RPi 3B+ is quite old and after 5 years of loyal duty I should consider to replace it but still, I'm trying to understand why it happened after the 13.0 update and it's not [...]

This reminds me that in between when it was crashing and now, where it is no longer crashing, I increased my swap to prevent ESPHome crashes.

dipseth, 2024-08-19:
> > > The ram shortage was quite mitigated by my bigger SWAP config

I had done something similar, adding a 600mb SWAP file.

rrooggiieerr, 2024-08-19:
@sairon Are you saying this as a HA OS developer, or just making a statement? What are your sources that 1 GB is not be enough? My current memory usage on 12.4 is 676.3 MiB/75 % with 53 integrations active, 155 devices and 3 add ons active

rrooggiieerr, 2024-08-19:
@EastArctica you can run ESPHome on your laptop or local PC, this wil definitely take some load away from your HA device

sairon, 2024-08-20:
@rrooggiieerr

> Are you saying this as a HA OS developer, or just making a statement?

I don't know what you mean by either of those, so I will elaborate a bit. On one hand, 1 GB RAM can be fine, from those who opted in for the [analytics](https://analytics.home-assistant.io/), 7 % are running on RPi 3 which only has 1 GB of RAM. However, your mileage may vary. 3 add-ons and 53 integrations is still something I'd call conservative usage, as especially the amount of add-ons makes a bigger difference. Compared to you, dipseth has 18 add-ons, and many people are using HA as a self-hosting platform, where RAM starts to be scarce very soon. For other platforms minimum of 2 GB RAM is generally recommended [in the docs](https://www.home-assistant.io/installation/linux), for CM4 in Yellow it's [recommended as well](https://yellow.home-assistant.io/faq/#how-do-i-find-a-raspberry-pi-compute-module-4). From my purely personal experience, only one real HA deployment out of three I manage, only one sits just slightly below 1 GB after a while of usage.

In your case, the remaining 25 % of RAM can become insufficient quite quickly. It also needs to be considered the system usually performs well if it also has some RAM available for page caches, if not, it can lead to higher I/O and combined with swapping on a (rather) slow media, it can be very detrimental to performance. Which leads me to another thought - the SD card you are using - while it's a good choice in terms of endurance - might not be the best for this usage in overall. Per the description it's optimized for use in security cameras, and has no Application Performance Class, so there's no IOPS guarantee. For HAOS it's recommended to use cards of [A2 class](https://www.home-assistant.io/installation/raspberrypi) which perform better in scenarios like this.

In summary, the issue definitely looks like performance related. To see what's going on, having an HDMI display connected, typing `login` in the HA CLI and checking `free` and `uptime` (or just watching `top`) when the system becomes unstable could help diagnose if the theory is right or not.

DeXter3306, 2024-08-20:
Thanks for the insight.
But, if I may, I still don't understand why it is working very well on 12.4 and not on 13.0, like my workload doesn't change between both.
That's puzzling me right now :)

sairon, 2024-08-20:
@DeXter3306 It's hard to tell without any details about your deployment. Doing what I suggested in the last paragraph of the previous post could help. Also, the kernel doesn't log all information by default, there might be more in `dmesg` or host logs (i.e. `ha host logs` or `journalctl -e`).

EastArctica, 2024-08-20:
> @rrooggiieerr
>
>
> I don't know what you mean by either of those, so I will elaborate a bit. On one hand, 1 GB RAM can be fine, from those who opted in for the [...]

I made a backup of my current system and reverted back to one from when I was experiencing issues before and alas! I am experiencing them once again (didn't think I'd be happy to say that). So far I've been doing most of my monitoring and diag through ssh to the OS, but when the host seems to "die" it tends to kill the network too. Through the terminal on the OS itself (via keyboard + HDMI), how can I run shell commands? I'm sorta just stuck in the home assistant CLI and don't know how to break out of it...

dpgh947, 2024-08-22:
I tried 13.1 this morning, still broken. HA was just starting to come up, then it rebooted, over and over. This time I managed to get in to ssh and issue the boot-slot other command to get back to 12.4 rather than getting the pi out again and plugging in a screen, so I can't confirm the same messages, but whatever it is, I still cannot upgrade my 2gb pi4.

DeXter3306, 2024-10-03:
Hey, I'm also using the serial port for my RaZberry.

Gonioul, 2024-10-05:
I don't know if it'll help, but after googling a lot, I came across this issue.
I have a completely different setup (ha-supervised on old 32 raspbian, RPI4 8GB w/M.2 on usb) but had similar problems (kworker at 100%).
HA didn't crash, but could take 1/2 hour to start, with lots of hung tasks messages.
I use official docker repos.

After lots of google searches, I decided to roll back the move to a 64bit kernel which promised a boost (arm_64bits=1 in boot config, which is not a problem since distrib is 32bits) and force supervisor to use 32 bit.
And now it works like a charm.

So looks like there's a bug in docker on 64bit raspberry kernels if it's the same issue.

GeraldPape, 2024-10-06:
Seems I have the same problem. RPI4 with 8GB.
Stops responding after some time, sometimes more, sometimes less. Console with the same errors as in th intial picture.
the SMB Share add-in still worked, also grott mostly keeps working.  But web ui and APP dead. Only chance is to cold boot the raspi.
What do you need to inspect the problem?

Gonioul, 2024-12-05:
Follow up from my previous post.

I had kworker+flush issues at 100% when updating, with hung tasks, could take 30+ minutes.

Then I found this: https://lore.kernel.org/all/bug-217965-13602@https.bugzilla.kernel.org%2F/T/

Ext4 kernel bug, upgrade kernel or apply workaround (change ext4 fs/mount options).
Solved my case.

regystro, 2024-12-07:
> Follow up from my previous post.
>
> I had kworker+flush issues at 100% when updating, with hung tasks, could take 30+ minutes.
>
> Then I found this: [...]

I upgraded HA OS to 14.0, which updates Raspberry Pi kernel to 6.6.51 - stable_20241008, but it didn't fix it for me.
In case you used the other option (`mount -o remount,stripe=0 <dev>` / `tune2fs -f -E stride=0 <dev>`), can you tell which <dev> did you use? I have 8 sdaX partitions and don't know how to tell apart which one...

Thank you.

davidparsson, 2025-03-08:
Has anybody had any success with 14.2? (I doubt it, but closing this without a resolution would be a shame.)

LasticotSoftware, 2025-05-14:
Same issue on a Pi 4 with 1GB RAM with a fresh install of HA 15.1 and core 2025.4.

Now unable to use a recent version of Home Assistant. Considering buying a Pi 5.

closed 2025-08-21 by github-actions[bot]