home-assistant/operating-system #3535: HA OS 13.0 breaks Zigbee2MQTT

rhpijnacker, 2024-08-15:
### Describe the issue you are experiencing

After updating HA OS, Zigbee2MQTT fails to start.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

13.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Proxmox 8.1.3
USB - Zigbee coordinator: https://slae.sh/projects/cc2652/

### Steps to reproduce the issue

1. Update HA OS to version 13.0
2. Boot
3. Zigbee2MQTT cannot lock zigbee stick

### Anything in the Supervisor logs that might be useful for us?

```txt
[23:03:05] INFO: Preparing to start...
[23:03:05] INFO: Socat not enabled
[23:03:05] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-08-15 23:03:07] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-15 23:03:07] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2024-08-15 23:03:07] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
[2024-08-15 23:03:07] info: 	z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-15 23:03:07] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_93_30_74-if00-port0\"},\"adapter\":{\"concurrent\":null,\"delay\":null,\"disableLED\":false}}"'
[2024-08-15 23:03:08] debug: 	zh:adapter: Failed to validate path: 'Error: spawn udevadm ENOENT'
[2024-08-15 23:03:08] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_93_30_74-if00-port0"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
[2024-08-15 23:03:08] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_93_30_74-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-08-15 23:03:08] error: 	z2m: Error while starting zigbee-herdsman
[2024-08-15 23:03:08] error: 	z2m: Failed to start zigbee
[2024-08-15 23:03:08] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-08-15 23:03:08] error: 	z2m: Exiting...
[2024-08-15 23:03:08] error: 	z2m: Error: Error Resource temporarily unavailable Cannot lock port
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2024.8.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.44-haos
arch | x86_64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4868
Installed Version | 1.34.0
Stage | running
Available Repositories | 1457
Downloaded Repositories | 22

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 62.3 GB
disk_used | 26.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.18.0), WireGuard (0.10.2), Advanced SSH & Web Terminal (18.0.0), AdGuard Home (5.1.1), Mosquitto broker (6.4.1), Zigbee2MQTT (1.39.1-1), Studio Code Server (5.15.0), Home Assistant Google Drive Backup (0.112.1), Grott Beta branch (2.8) (0.1.5), Bluetooth Presence Monitor (1.0.0), ESPHome (2024.7.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 6
views | 20
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 6, 2024 at 08:31
-- | --
current_recorder_run | August 15, 2024 at 22:57
estimated_db_size | 1042.08 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

rhpijnacker, 2024-08-17:
FYI:
It looks like more USB devices stopped working.

* Downgraded HA OS to 12.4
USB devices still do not work

* Downgraded Home Assistant to 2024.7.4
USB devices still do not work

Does the upgrade to 13.0 change anything persistently that is not reverted when downgrading to 12.4?

sairon, 2024-08-19:
> Does the upgrade to 13.0 change anything persistently that is not reverted when downgrading to 12.4?

No, it's more likely that something changed on the host or in the VM configuration in the meantime. Check the host logs when USB devices are enumerated - if there are some errors or if the devices can't be seen. Eventually check the VM config if the USB passthrough is still configured correctly.

dsan9088, 2024-08-22:
I had this same issue after upgrading to a new CM4 last week.  You'll have to go to your zigbee coordinator's firmware update page from a Chrome browser and do a serial update using an extender.  Mine was a little more starightforward using SkyConnect, but yours seems to be here: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin
After I upgraded the firmware, I had to go into my Zigbee2MQTT configuration and change the serial adapter from EZSP to ember.  Hope this helps