home-assistant/operating-system #3540: HAOS 13.0 RPI4 unstabled/tainted Install

dipseth, 2024-08-17:
### Describe the issue you are experiencing

Since Installing 13.0 on RPI4 the system is completely unstable and is either in state of frozen /restarting.   Shortly before the freezing begins this Error message appears:

```
[ 363.489746] Tainted: G C [ 6.6.31-haos-raspi #1 363.489765] "echo 0> /proc/sys/kernel/hung_task_timeout_secs" disables this message. 
[ 363.489973] INFO: task kworker/u10:1:88 blocked for more than 120 seconds. 
[ 363.489994] Tainted: G C 6.6.31-haos-raspi #1 
[ 363.490011] "echo > /proc/sys/kernel/hung_task_timeout_secs" disables this message. 
[ 363.490265] 363.490244] INFO: task usb-storage: 93 blocked for more than 120 seconds. Tainted: G C 6.6.31-haos-raspi #1 
[ 363.490281] "echo > /proc/sys/kernel/hung_task_timeout_secs" disables this message. 
[ 363.490519] INFO: task kworker/2:2:95 blocked for more than 120 seconds. Tainted: G 6.6.31-haos-raspi #1 
[ 363.490537] C 
[ 363.490553] "echo > /proc/sys/kernel/hung_task_timeout_secs" disables this message. 
[ 363.490725] INFO: task kworker/u13:2:266 blocked for more than 120 seconds. 
[ 363.4907471 Tainted: G C 6.6.31-haos-raspi #1 
[ 363.4907641 "echo 0> /proc/sys/kernel/hung_task_timeout_secs" disables this message. 
[ 363.491155] INFO: task NetworkManager: 448 blocked for more than 120 seconds. 
[ 363.491175] Tainted: Gc Tainted: G C 6.6.31-haos-raspi #1 
[ 363.491192] " echo 0> /proc/sys/kernel/hung_task_timeout_secs" disables this message. 
[ 363.4914001 INFO: task rpcbind: 457 blocked for more than 120 seconds. 
[363.491419 ] 6.6.31-haos-raspi #1 [363.4916681 INFO: task wpa_supplicant: 460 blocked for more than 120 seconds. proc/sys/kernel/hung_task_timeout_secs" disables this message.
```

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

13.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Canakit RPi4-64 1GB Ram

Note bug occured on SD card (LEXAR ) and with SSD equipment below:

CanaKit 3.5A
Kingston A400 
USB 3.1 variant of the StarTech 2.5″ SATA
Sabrent powered USB hub 

### Steps to reproduce the issue

1. Installed 13.0
2. Booted up 
3. Experienced unresponsive system
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Dont believe so

2024-08-17 07:25:39.734 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-08-17 07:30:27.291 WARNING (MainThread) [supervisor.addons.options] Option 'secrets' does not exist in the schema for AppDaemon (a0d7b954_appdaemon)
2024-08-17 07:35:28.924 WARNING (MainThread) [supervisor.addons.options] Option 'secrets' does not exist in the schema for AppDaemon (a0d7b954_appdaemon)
2024-08-17 07:39:02.759 WARNING (MainThread) [supervisor.addons.options] Option 'secrets' does not exist in the schema for AppDaemon (a0d7b954_appdaemon)
2024-08-17 07:40:30.963 WARNING (MainThread) [supervisor.addons.options] Option 'secrets' does not exist in the schema for AppDaemon (a0d7b954_appdaemon)
2024-08-17 07:44:52.013 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-08-17 07:45:32.960 WARNING (MainThread) [supervisor.addons.options] Option 'secrets' does not exist in the schema for AppDaemon (a0d7b954_appdaemon)
2024-08-17 07:50:35.291 WARNING (MainThread) [supervisor.addons.options] Option 'secrets' does not exist in the schema for AppDaemon (a0d7b954_appdaemon)
2024-08-17 07:55:36.350 WARNING (MainThread) [supervisor.addons.options] Option 'secrets' does not exist in the schema for AppDaemon (a0d7b954_appdaemon)
2024-08-17 08:00:39.083 WARNING (MainThread) [supervisor.addons.options] Option 'secrets' does not exist in the schema for AppDaemon (a0d7b954_appdaemon)
2024-08-17 08:05:40.329 WARNING (MainThread) [supervisor.addons.options] Option 'secrets' does not exist in the schema for AppDaemon (a0d7b954_appdaemon)
[... 193 lines cut ...]
dashboards | 4
-- | --
resources | 18
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 13, 2024 at 9:27 AM
-- | --
current_recorder_run | August 17, 2024 at 8:15 AM
estimated_db_size | 116.53 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

Cikkajes, 2024-08-18:
Had the same issue, especially after updating the Core too to 2024.8.2. HA was freezing and rebooting constantly and I saw this in the logs:

Fatal Python error: Illegal instruction

Now reverted back to 2024.8.2 and it seems a little more stable but still not 100% online and quite frustrating.

sairon, 2024-08-19:
This is same as #3534, let's follow up there.

closed 2024-08-20 by sairon