home-assistant/operating-system #3544: error on installation OS 13.0

nielswest65, 2024-08-20:
### Describe the issue you are experiencing

while installing 13.0 for the second time (went back to 2024.8.2 because of issues after upgrade, but kept python errors) is got an error in the install log on Python? it works for now, where up till now withing 24 hours my p4 would crash on .py errors there were two errors about the same content, I could not clip it because the install was done.

[supervisor.docker.homeassistant] Starting Home Assistant ghcr.io/home-assistant/raspberrypi4-64-homeassistant with version landingpage
/usr/local/lib/python3.12/site-packages/cpe/comp/cpecomp1_1.py:57: SyntaxWarning: invalid escape sequence '\w'
_STRING = "\w\.\-,\(\)@\#"
/usr/local/lib/python3.12/site-packages/cpe/comp/cpecomp2_2.py:52: SyntaxWarning: invalid escape sequence '\d'
_VALUE_PATTERN = "^([\d\w\._\-~%]+)$"
/usr/local/lib/python3.12/site-packages/cpe/cpe2_3_fs.py:97: SyntaxWarning: invalid escape sequence '\:'
_parts_pattern = "^cpe:2.3:{0}\:{1}\:{2}\:{3}\:{4}\:{5}\:{6}\:{7}\:{8}\:{9}\:{10}$".format(
[supervisor.os.manager] Detect Home Assistant Operating System 13.0 / BootSlot A

another bug in the list is about zigbee2mqtt I had that to, I think because of an .py issue?

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

13.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

usb ssd 500gb, sonoff 3.0 zigbee stick, Ikea temp/hum/air. no sd-card. Pi4 2gb 

### Steps to reproduce the issue

1. version 13.0 with Pi imager on 500gb SSD USB
2. start Pi4 2gb
3. follow on screen (no errors)
4. open http://home.assistant:8123
5. show details on install
6. error shows in logging.
7.
8.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
not available yet
```

### Anything in the Host logs that might be useful for us?

```txt
nay
```

### System information

not available yet

### Additional information

not available yet

sairon, 2024-08-20:
The warnings are harmless, see home-assistant/supervisor#5226 for details.

Is there anything that exactly doesn't work or you're just being cautious because of the log entries?

nielswest65, 2024-08-20:
> The warnings are harmless, see [home-assistant/supervisor#5226](https://github.com/home-assistant/supervisor/issues/5226) for details.
> 
> Is there anything that exactly doesn't work or you're just being cautious because of the log entries?

My English is not perfect :-) 

1 I had OS 12.8? working and no issues
2 I upgraded to 13.0 then issues started on Z2Mq and several HACS integrations all on .py componants
3 went back to previous version doing db restore.
4 within every 24 hours the system would crash, 

so I did another full rebuils

1) Pi imager putting 13,0 on my disk - then I got the error. as I had these Py errors before. my conclusion was that it went wrong on Installation of 13,0 the strange thing is that after restore to prevrious versions the errors stayded.

sairon, 2024-08-20:
> My English is not perfect :-)

No worries, patience is more important :)

> I am an enduser not a developer :-) I can only read such errors, mostly not realy understand them

Not a problem either. However, from the information you supplied I don't see anything that could cause issues. If something's not right, more logs will be needed. If the errors appear in Home Assistant Core logs (I guess that's what you call "Py errors") the OS might not be at fault and reporting an issue in the [Core repository](https://github.com/home-assistant/core) might be more appropriate. To troubleshoot OS issues, we usually need you to fetch the Host logs (or eventually capture a screen on a display connected to your RPi) and sometimes Supervisor logs as well. Describing the behavior in detail is also desirable ("crash" is just too vague).

nielswest65, 2024-08-20:
Alright, then after the restore is done, I will see if I can find old and new logfiles,
thank you for your answers and questions. :-)

nielswest65, 2024-08-20:
I have a logfile, it is from restoring a backup after putting 13.0 OS as a base on the 500gb SSD
[supervisor_2024-08-20T11-04-58.988Z.log](https://github.com/user-attachments/files/16674794/supervisor_2024-08-20T11-04-58.988Z.log)
[home-assistant_2024-08-20T11-04-35.978Z.log](https://github.com/user-attachments/files/16674795/home-assistant_2024-08-20T11-04-35.978Z.log)

Core 2024.8.2
Supervisor  2024.08.0
Operating System 13.0
Frontend 20240809.0

sairon, 2024-08-20:
I don't see anything wrong in Supervisor logs, and there are no Host logs to check if there's any OS issue, but all the errors in the Home Assistant logs are caused likely by Core integrations, their configuration, or third-party custom components.  From this perspective the OS is not at fault (and there is nothing that would raise suspicion it is either).

closed 2024-08-20 by sairon