home-assistant/operating-system #3545: 13.1 rc1 will not load

AllardKatan, 2024-08-20:
### Describe the issue you are experiencing

I am trying to upgrade to the 13.1. rc version for beta testing but I don't get it to load. See also my comments in 'discussion' but I guess that wasn't the place for it? 
The short of it: when I attempt to upgrade, the system simply boots back up in the existing 12.4 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Host OS: Linux Mint 22. Hypervisor: Virtualbox 7.0.16. USB devices: sonoff zigbee dongle 'e', USB ssd drive (passthrough enabled in Virtualbox but not physically connected).
Three consecutive upgrade attempts put me into 'rescue mode'

### Steps to reproduce the issue

1. Click 'install' on the upgrade pop-up in OR type `os upgrade` in the CLI
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-08-19 17:44:34.129 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0
2024-08-19 17:44:34.176 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
2024-08-19 17:44:35.131 INFO (MainThread) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0 - 172.30.32.4
2024-08-19 17:44:35.164 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-observer with version 2023.06.0
2024-08-19 17:44:35.221 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-multicast with version 2024.03.0
2024-08-19 17:44:35.267 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
2024-08-19 17:44:35.888 INFO (MainThread) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/amd64-hassio-multicast with version 2024.03.0 - Host
2024-08-19 17:44:35.898 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
2024-08-19 17:44:35.933 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/qemux86-64-homeassistant with version 2024.8.2
2024-08-19 17:44:36.108 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 12.4 / BootSlot A
2024-08-19 17:44:37.031 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
2024-08-19 17:44:37.031 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/9074a9fa repository
2024-08-19 17:44:37.031 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/cebe7a76 repository
2024-08-19 17:44:37.034 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/65f67727 repository
2024-08-19 17:44:37.036 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
2024-08-19 17:44:37.043 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/d5369777 repository
2024-08-19 17:44:37.053 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
2024-08-19 17:44:37.774 INFO (MainThread) [supervisor.store] Loading add-ons from store: 83 all - 83 new - 0 remove
2024-08-19 17:44:37.871 INFO (MainThread) [supervisor.addons.manager] Found 7 installed add-ons
2024-08-19 17:44:37.919 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-configurator with version 5.8.0
2024-08-19 17:44:37.937 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-mosquitto with version 6.4.1
2024-08-19 17:44:37.956 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-samba with version 12.3.2
2024-08-19 17:44:37.974 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-ssh with version 9.14.0
2024-08-19 17:44:38.003 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/esphome/esphome-hassio with version 2023.12.9
2024-08-19 17:44:38.025 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/amd64 with version 5.15.0
2024-08-19 17:44:38.040 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/sabeechen/hassio-google-drive-backup-amd64 with version 0.112.1
2024-08-19 17:44:38.499 INFO (MainThread) [supervisor.backups.manager] Found 3 backup files
2024-08-19 17:44:38.515 INFO (MainThread) [supervisor.discovery] Loaded 2 messages
2024-08-19 17:44:38.516 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
2024-08-19 17:44:38.516 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup
2024-08-19 17:44:38.516 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2024-08-19 17:44:38.516 INFO (MainThread) [supervisor.resolution.checks.base] Run check for detached_addon_removed/addon
2024-08-19 17:44:38.516 INFO (MainThread) [supervisor.resolution.checks.base] Run check for detached_addon_missing/addon
2024-08-19 17:44:38.516 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-08-19 17:44:38.517 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-08-19 17:44:38.517 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
2024-08-19 17:44:38.517 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-08-19 17:44:38.517 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - setup
[... 48 lines cut ...]
2024-08-19 17:46:10.586 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2024-08-19 17:46:10.588 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
2024-08-19 17:46:10.589 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2024-08-19 17:46:10.613 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2024-08-19 17:46:10.721 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-08-19 17:46:10.721 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-08-19 17:46:10.721 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2024-08-19 17:46:11.219 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-08-19 17:46:11.219 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-08-19 17:46:11.219 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-08-19 17:46:11.359 INFO (MainThread) [supervisor.host.services] Updating service information
2024-08-19 17:46:11.371 INFO (MainThread) [supervisor.host.network] Updating local network information
2024-08-19 17:46:11.573 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2024-08-19 17:46:11.585 INFO (MainThread) [supervisor.host.manager] Host information reload completed
```

### Anything in the Host logs that might be useful for us?

```txt
(not available)
```

### Additional information

I can't seem to find the log file that holds the full set of messages that appear on boot (bootlog, I guess that is called) That may contain useful information but it's not in the 'host' log. Where is that located, or is it not stored?

sairon, 2024-08-20:
Right, there's no information about the issue in these logs. The messages shown at boot are the host logs, but by default the log contains only 100 last lines. You can try running `ha host logs -n2000 -b-1` to get more lines and logs from the previous boot, by adjusting the `-b` parameter you can go back in time (`-b0`is the current boot). By appending something like `> /mnt/data/supervisor/homeassistant/hostlogs.txt`, you can save the output to the HA configuration folder and download it from there.

If the error happens after mounting the data disk, it should be persisted, but if it happens early in the boot, it may be empty. In that case watching VirtualBox console to determine where it fails would be needed as the first step.

AllardKatan, 2024-08-20:
I did another attempt to upgrade through the UI, and it came back up again as 12.4. This is the log file: [hostlogs.txt](https://github.com/user-attachments/files/16675922/hostlogs.txt)
I don't think this is useful? Unless I'm reading it wrong the first line already indicates that this is not the 13.1 version. The -1 log showed information about an earlier boot.

AllardKatan, 2024-08-20:
I will take a look at the console when I am at home, I can't do that from here. Everything flashes by real fast though, too fast to read sometimes. I'll see if I can use a screengrabber to record it, unless you know of a better trick.

sairon, 2024-08-20:
OVA also logs the kernel buffer to the `ttyS0` serial console, you can enable logging to file in VM settings. e.g. like this:
[image]

AllardKatan, 2024-08-20:
OK, I stopped the VM using the ACPI shutdown command, changed the settings to reflect your screenshot above, then restarted the VM. It booted normally into 12.4. Then I typed 'os update' into the terminal CLI. It restarted but ended up in 'rescue mode'. I got a prompt and typed 'exit'. Then in booted back up normally in 12.4. 
Logfile is attached. I don't know what to look for of course, but is that 'not found' message right at the start expected?

[haos.log](https://github.com/user-attachments/files/16681613/haos.log)

sairon, 2024-08-21:
> It restarted but ended up in 'rescue mode'. I got a prompt and typed 'exit'. Then in booted back up normally in 12.4.

I don't see any attempt to boot to OS 13 in that log. Can you check manually select the other boot slot in the boot loader before it boots automatically?

> Logfile is attached. I don't know what to look for of course, but is that 'not found' message right at the start expected?

That's okay, it's the emulated firmware's attempt to boot from optical drive that doesn't exist.

AllardKatan, 2024-08-21:
Okay, I'm not quite sure what  happened, but it seems the problems is fixed with the 13.1 release :-) . I wasn't aware the release was already done. I did the update a few moments ago from the console. The first attempt it selected slot B (okay=1 , try =1), and since things were weird before I selected the other slot (A, ok=1, try=0). That booted int o 12.4, so I did another `os update` , and selected slot B ('try =2') . Then I was a bit surprised to see that A: I was now on 13.1 without an rc suffix and B: the logfile was empty. I had cleared the contents and saved the logfile before starting this procedure but contrary to earlier attempts there was nothing added to the file. Anyway, the problem seems to have been fixed, which is good.

Devoman71, 2024-08-22:
I have just broken my live enstance by applying OS 13..1 so the issue seems not resolved as it's not booting up. I'm using a generic Intel machine. 
I will have to rebuild from my backups and I will not use OS 13,1

AllardKatan, 2024-08-22:
Well, I didn't say ALL issues were resolved. But since in my case the system still started, and the problem only occurred in the beta release, I think yours is a different issue and should be reported as such.

AllardKatan, 2024-08-25:
Well, it seems I have to reopen this issue. I rebooted my VM today, and to my great surprise it booted back up in OS 12.4! I had a notification badge for upgrading both OS (to 13.1) and core (to 8.3 from 8.2). I started with the core update, but that resulted in a very long wait with nothing happening except the instance going offline. I could not even shut it down from the Virtualbox manager app, had to kill it from the host OS. I have no idea what happened, the instance probably booted from another slot that still had 12.4 in it? After killing the VM I could start it again and it booted back into 12.4. I don't have too much time right now to tinker, but if there are any tests I can do let me know