home-assistant/operating-system #3546: When upgrading anything higher than `12.2` Bluetooth stops working

ovflowd, 2024-08-20:
### Describe the issue you are experiencing

When upgrading to any version greater than 12.2 for `ha os`, Home Assistant will report that no Bluetooth Adapters were found. Downgrading with `ha os 12.2` fixes the issue.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Nothing plugged in or modified on Raspberry Pi

### Steps to reproduce the issue

1. Upgrade HA OS

### Anything in the Supervisor logs that might be useful for us?

```txt
/usr/local/lib/python3.12/site-packages/cpe/comp/cpecomp1_1.py:57: SyntaxWarning: invalid escape sequence '\w'
  _STRING = "\w\.\-,\(\)@\#"
/usr/local/lib/python3.12/site-packages/cpe/comp/cpecomp2_2.py:52: SyntaxWarning: invalid escape sequence '\d'
  _VALUE_PATTERN = "^([\d\w\._\-~%]+)$"
/usr/local/lib/python3.12/site-packages/cpe/cpe2_3_fs.py:97: SyntaxWarning: invalid escape sequence '\:'
  _parts_pattern = "^cpe:2.3:{0}\:{1}\:{2}\:{3}\:{4}\:{5}\:{6}\:{7}\:{8}\:{9}\:{10}$".format(
```

### Anything in the Host logs that might be useful for us?

```txt
Nope
```

### System information

### System Information (HA OS 12.2)

version | core-2024.8.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.1.73-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4998
Installed Version | 1.34.0
Stage | running
Available Repositories | 1389
Downloaded Repositories | 5
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 6 September 2024 at 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-19.ui.nabu.casa
certificate_status | ready
instance_id | *****
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 57.8 GB
disk_used | 9.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.14.0), File editor (5.8.0), Mosquitto broker (6.4.1), Matter Server (6.4.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 1
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 11 August 2024 at 16:43
-- | --
current_recorder_run | 20 August 2024 at 13:55
estimated_db_size | 318.94 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### System Information (HA OS 13.0)

version | core-2024.8.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.31-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4998
Installed Version | 1.34.0
Stage | running
Available Repositories | 1389
Downloaded Repositories | 5
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 6 September 2024 at 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-19.ui.nabu.casa
certificate_status | ready
instance_id | a87c99d630414ddaa77e9c4c38c8b34a
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 57.8 GB
disk_used | 9.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.14.0), File editor (5.8.0), Mosquitto broker (6.4.1), Matter Server (6.4.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 1
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 11 August 2024 at 16:43
-- | --
current_recorder_run | 20 August 2024 at 14:06
estimated_db_size | 318.94 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Additional information

### Integration Error

[image]

### Integration Error Logs

https://gist.github.com/ovflowd/fdafa097ef9bcda56f26af6a383dbcc1

sairon, 2024-08-21:
It is probably some intermittent issue. Try unplugging the RPi for a while and cold-booting it. Once booted to OS 13, please share the host logs (ideally obtained with `ha host logs -n1000 -b0` to get all events since the boot).

bferrara, 2024-08-22:
I had the same Raspberry Pi Bluetooth integration error after updating to 13.1 - the cold-boot suggestion from @sairon worked for me. It was working again after the cold-boot.

kebner, 2024-08-22:
Hi,
tried that cold boot but it did not help. I got an exception like this
``Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:241
integration: Home Assistant WebSocket API ([documentation](https://www.home-assistant.io/integrations/websocket_api), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
First occurred: 9:05:53 AM (1 occurrences)
Last logged: 9:05:53 AM

[546853661440] Unexpected exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 363, in establish_connection
    await client.connect(
  File "/usr/local/lib/python3.12/site-packages/habluetooth/wrappers.py", line 310, in connect
    connected = await super().connect(**kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 273, in connect
    await self.get_services(
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 661, in get_services
    self.services = await manager.get_services(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 666, in get_services
    await self._wait_for_services_discovery(device_path)
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 799, in _wait_for_services_discovery
    raise BleakError("failed to discover services, device disconnected")
bleak.exc.BleakError: failed to discover services, device disconnected

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohomekit/controller/ble/connection.py", line 47, in establish_connection
    return await retry_establish_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 472, in establish_connection
    _raise_if_needed(name, device.address, exc)
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 339, in _raise_if_needed
    raise BleakConnectionError(msg) from exc
bleak_retry_connector.BleakConnectionError: LEDVANCE A50 F-7806 [CD:7F:2E:0D:5A:83] (id=2B:CD:42:80:D9:AE) - CD:7F:2E:0D:5A:83: Failed to connect after 9 attempt(s): failed to discover services, device disconnected

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 241, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2763, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2806, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 999, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1071, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 626, in async_handle_light_on_service
    await light.async_turn_on(**filter_turn_on_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/light.py", line 192, in async_turn_on
    await self.async_put_characteristics(characteristics)
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/entity.py", line 140, in async_put_characteristics
    return await self._accessory.put_characteristics(payload)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/connection.py", line 1004, in put_characteristics
    await self.pairing.put_characteristics(characteristics)
  File "/usr/local/lib/python3.12/site-packages/aiohomekit/controller/ble/pairing.py", line 197, in _async_operation_lock_wrap
    return await func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 505, in _async_wrap_bluetooth_connection_error_retry
    return await func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohomekit/controller/ble/client.py", line 76, in _async_disconnect_on_missing_services_wrap
    return await func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohomekit/controller/ble/pairing.py", line 211, in _async_restore_and_resume
    await self._populate_accessories_and_characteristics()
  File "/usr/local/lib/python3.12/site-packages/aiohomekit/controller/ble/pairing.py", line 1101, in _populate_accessories_and_characteristics
[... cut ...]
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohomekit/controller/ble/connection.py", line 57, in establish_connection
    raise AccessoryDisconnectedError(ex) from ex
aiohomekit.exceptions.AccessoryDisconnectedError: LEDVANCE A50 F-7806 [CD:7F:2E:0D:5A:83] (id=2B:CD:42:80:D9:AE) - CD:7F:2E:0D:5A:83: Failed to connect after 9 attempt(s): failed to discover services, device disconnected``

This happened in the past "sometimes", but now it is the regular case after upgrading to 13 on a RPi 4

ovflowd, 2024-08-22:
Why are y'all hijacking my support thread? Please open your own if you also have issues lol.

kebner, 2024-08-22:
Relax

ovflowd, 2024-08-22:
@kebner, the reasoning behind my ask is people searching for similar issues will be better served with individual threads tailored for their devices.

ovflowd, 2024-08-22:
@sairon unfortunately, cold boot did not help, here are the logs after boot:

<summery>Logs</summary>
<details>

```
2024-05-27 21:48:35.271 homeassistant kernel: kvm [1]: IPA Size Limit: 40 bits
2024-05-27 21:48:35.271 homeassistant kernel: kvm [1]: vgic interrupt IRQ9
2024-05-27 21:48:35.271 homeassistant kernel: kvm [1]: VHE mode initialized successfully
2024-05-27 21:48:35.271 homeassistant kernel: Initialise system trusted keyrings
2024-05-27 21:48:35.271 homeassistant kernel: workingset: timestamp_bits=46 max_order=20 bucket_order=0
2024-05-27 21:48:35.271 homeassistant kernel: zbud: loaded
2024-05-27 21:48:35.271 homeassistant kernel: squashfs: version 4.0 (2009/01/31) Phillip Lougher
2024-05-27 21:48:35.271 homeassistant kernel: NFS: Registering the id_resolver key type
2024-05-27 21:48:35.271 homeassistant kernel: Key type id_resolver registered
2024-05-27 21:48:35.271 homeassistant kernel: Key type id_legacy registered
2024-05-27 21:48:35.271 homeassistant kernel: nfs4filelayout_init: NFSv4 File Layout Driver Registering...
2024-05-27 21:48:35.271 homeassistant kernel: nfs4flexfilelayout_init: NFSv4 Flexfile Layout Driver Registering...
2024-05-27 21:48:35.271 homeassistant kernel: Key type asymmetric registered
2024-05-27 21:48:35.271 homeassistant kernel: Asymmetric key parser 'x509' registered
2024-05-27 21:48:35.271 homeassistant kernel: Block layer SCSI generic (bsg) driver version 0.4 loaded (major 246)
2024-05-27 21:48:35.271 homeassistant kernel: io scheduler mq-deadline registered
2024-05-27 21:48:35.271 homeassistant kernel: io scheduler kyber registered
2024-05-27 21:48:35.271 homeassistant kernel: io scheduler bfq registered
2024-05-27 21:48:35.271 homeassistant kernel: irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7c502000, parent irq: 27)
2024-05-27 21:48:35.271 homeassistant kernel: irq_brcmstb_l2: registered L2 intc (/soc/intc@7d503000, parent irq: 28)
2024-05-27 21:48:35.271 homeassistant kernel: irq_brcmstb_l2: registered L2 intc (/soc/intc@7d508380, parent irq: 29)
2024-05-27 21:48:35.271 homeassistant kernel: irq_brcmstb_l2: registered L2 intc (/soc/intc@7d508400, parent irq: 30)
2024-05-27 21:48:35.271 homeassistant kernel: irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7d510600, parent irq: 31)
2024-05-27 21:48:35.271 homeassistant kernel: irq_brcmstb_l2: registered L2 intc (/soc/intc@7d517b00, parent irq: 32)
2024-05-27 21:48:35.271 homeassistant kernel: Serial: 8250/16550 driver, 1 ports, IRQ sharing enabled
2024-05-27 21:48:35.271 homeassistant kernel: 107d50c000.serial: ttyS0 at MMIO 0x107d50c000 (irq = 34, base_baud = 6000000) is a Broadcom BCM7271 UART
2024-05-27 21:48:35.271 homeassistant kernel: serial serial0: tty port ttyS0 registered
2024-05-27 21:48:35.272 homeassistant kernel: iproc-rng200 107d208000.rng: hwrng registered
2024-05-27 21:48:35.272 homeassistant kernel: vc-mem: phys_addr:0x00000000 mem_base=0x3fc00000 mem_size:0x40000000(1024 MiB)
2024-05-27 21:48:35.272 homeassistant kernel: bcm2712-iommu-cache 1000005b00.iommuc: bcm2712_iommu_cache_probe
2024-05-27 21:48:35.272 homeassistant kernel: brd: module loaded
2024-05-27 21:48:35.272 homeassistant kernel: loop: module loaded
2024-05-27 21:48:35.272 homeassistant kernel: zram: Added device: zram0
2024-05-27 21:48:35.272 homeassistant kernel: zram: Added device: zram1
2024-05-27 21:48:35.272 homeassistant kernel: zram: Added device: zram2
2024-05-27 21:48:35.272 homeassistant kernel: bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
2024-05-27 21:48:35.272 homeassistant kernel: Loading iSCSI transport class v2.0-870.
2024-05-27 21:48:35.272 homeassistant kernel: tun: Universal TUN/TAP device driver, 1.6
2024-05-27 21:48:35.272 homeassistant kernel: igb: Intel(R) Gigabit Ethernet Network Driver
2024-05-27 21:48:35.272 homeassistant kernel: igb: Copyright (c) 2007-2014 Intel Corporation.
2024-05-27 21:48:35.272 homeassistant kernel: dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
2024-05-27 21:48:35.272 homeassistant kernel: dwc_otg: FIQ enabled
2024-05-27 21:48:35.272 homeassistant kernel: dwc_otg: NAK holdoff enabled
2024-05-27 21:48:35.272 homeassistant kernel: dwc_otg: FIQ split-transaction FSM enabled
2024-05-27 21:48:35.272 homeassistant kernel: Module dwc_common_port init
2024-05-27 21:48:35.272 homeassistant kernel: usbcore: registered new interface driver uas
2024-05-27 21:48:35.272 homeassistant kernel: usbcore: registered new interface driver usb-storage
2024-05-27 21:48:35.272 homeassistant kernel: usbcore: registered new interface driver usbserial_generic
2024-05-27 21:48:35.272 homeassistant kernel: usbserial: USB Serial support registered for generic
2024-05-27 21:48:35.272 homeassistant kernel: mousedev: PS/2 mouse device common for all mice
2024-05-27 21:48:35.272 homeassistant kernel: rpi-rtc soc:rpi_rtc: registered as rtc0
2024-05-27 21:48:35.272 homeassistant kernel: rpi-rtc soc:rpi_rtc: setting system clock to 1970-01-01T00:00:08 UTC (8)
2024-05-27 21:48:35.272 homeassistant kernel: Driver for 1-wire Dallas network protocol.
2024-05-27 21:48:35.273 homeassistant kernel: bcm2835-wdt bcm2835-wdt: Poweroff handler already present!
2024-05-27 21:48:35.273 homeassistant kernel: bcm2835-wdt bcm2835-wdt: Broadcom BCM2835 watchdog timer
2024-05-27 21:48:35.273 homeassistant kernel: device-mapper: ioctl: 4.48.0-ioctl (2023-03-01) initialised: dm-devel@redhat.com
[... cut ...] 8
2024-08-22 07:19:20.860 homeassistant kernel: eth0: renamed from vethfd37703
2024-08-22 07:19:20.878 homeassistant NetworkManager[493]: <info>  [1724311160.8781] device (veth1bc3571): carrier: link connected
2024-08-22 07:19:20.880 homeassistant kernel: hassio: port 8(veth1bc3571) entered blocking state
2024-08-22 07:19:20.880 homeassistant kernel: hassio: port 8(veth1bc3571) entered forwarding state
2024-08-22 07:19:23.024 homeassistant kernel: hwmon hwmon3: Voltage normalised
```

</details>

hiagocosta, 2024-08-29:
i'm facing the same error on my beelink GK Mini with HAOS

eerico, 2024-10-18:
same with integrated bleutooth on debian on a lenovo laptop

ovflowd, 2024-10-18:
Bump @sairon, anything that can be done from your side? Is this issue being tracked? 🤔

ovflowd, 2025-01-29:
Hey there 👋 -- @sairon -- I believe the firmware patch for `BCM4345C0` needs to be updated after some investigation, it seems that the patch when applied crashes (I checked `dsmeg -i bcrm` to check if it had issues loading, and it seems so)

I noticed this patch exists on the web https://github.com/RPi-Distro/bluez-firmware/blob/master/broadcom/BCM4345C0.hcd and wonder if it differs from the one applied by HAOS.

Besides, any HAOS image above 12.2 fails to have Bluetooth initialized (Even 14.2 has the same issue)

Doing cold reboots also do not help. Only way to fix the issue is to always rollback to `ha os update --version 12.2`

closed 2025-05-07 by github-actions[bot]