home-assistant/operating-system #3547: New OS 13 on OVA not showed

Cippo2019, 2024-08-20:
### Describe the issue you are experiencing

hello 
i am on OS12.4 OVA vm.
I dont see proposal for version 13.
I have also look into my test envirinment (another OVA OS) and also here i dont see version 13 like new os.

any idea ?
wbr

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

2024.8.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

intel nuc

### Steps to reproduce the issue

1.
2.
3.
...
as described below

### Anything in the Supervisor logs that might be useful for us?

```txt
no other infoes
```

### Anything in the Host logs that might be useful for us?

```txt
no other infoes
```

### System information

no other infoes

### Additional information

no other infoes

sairon, 2024-08-21:
It is intentional due to a network driver bug, see https://github.com/home-assistant/version/pull/388

closed 2024-08-21 by sairon

Cippo2019, 2024-08-21:
a ok !! 
thanks a lot !
:-)