home-assistant/operating-system #3553: Intel(R) Wi-Fi 6 AX101, REV=0x370 failing

RCG-SW, 2024-08-22:
### Describe the issue you are experiencing

Just installed HAOS on new Intel NUC N100 and Wifi is not loading
` dmesg|grep iwlwifi`
`[    1.285553] iwlwifi 0000:00:14.3: enabling device (0000 -> 0002)
[    1.303390] iwlwifi 0000:00:14.3: Detected crf-id 0x501, cnv-id 0x80400 wfpm id 0x80000030
[    1.303743] iwlwifi 0000:00:14.3: PCI dev 54f0/0244, rev=0x370, rfid=0x10c000
[    1.307215] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-83.ucode failed with error -2
[    1.307823] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-82.ucode failed with error -2
[    1.308665] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-81.ucode failed with error -2
[    1.309307] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-80.ucode failed with error -2
[    1.310581] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-79.ucode failed with error -2
[    1.311314] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-78.ucode failed with error -2
[    1.317983] iwlwifi 0000:00:14.3: api flags index 2 larger than supported by driver
[    1.318348] iwlwifi 0000:00:14.3: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.0.2.36
[    1.319049] iwlwifi 0000:00:14.3: loaded firmware version 77.f92b5fed.0 so-a0-hr-b0-77.ucode op_mode iwlmvm
[    1.391318] iwlwifi 0000:00:14.3: Detected Intel(R) Wi-Fi 6 AX101, REV=0x370
[    1.397851] iwlwifi 0000:00:14.3: WRT: Invalid buffer destination
[    1.500033] iwlwifi 0000:00:14.3: WFPM_UMAC_PD_NOTIFICATION: 0x1f
[    1.500085] iwlwifi 0000:00:14.3: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[    1.500094] iwlwifi 0000:00:14.3: WFPM_AUTH_KEY_0: 0x90
[    1.500104] iwlwifi 0000:00:14.3: CNVI_SCU_SEQ_DATA_DW9: 0x10
[    1.500117] iwlwifi 0000:00:14.3: Detected RF HR1 B3, rfid=0x10c000
[    1.501261] iwlwifi 0000:00:14.3: RFIm is deactivated, reason = 4
[    1.541769] iwlwifi 0000:00:14.3: base HW address: 90:09:df:00:50:55
[    1.665156] iwlwifi 0000:00:14.3 wlp0s20f3: renamed from wlan0
[    2.362223] iwlwifi 0000:00:14.3: WRT: Invalid buffer destination
[    2.460192] iwlwifi 0000:00:14.3: WFPM_UMAC_PD_NOTIFICATION: 0x1f
[    2.460401] iwlwifi 0000:00:14.3: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[    2.460412] iwlwifi 0000:00:14.3: WFPM_AUTH_KEY_0: 0x90
[    2.460470] iwlwifi 0000:00:14.3: CNVI_SCU_SEQ_DATA_DW9: 0x10
[    2.461418] iwlwifi 0000:00:14.3: RFIm is deactivated, reason = 4
[    2.506186] iwlwifi 0000:00:14.3: Registered PHC clock: iwlwifi-PTP, with index: 1
[    6.377919] iwlwifi 0000:00:14.3: Microcode SW error detected. Restarting 0x0.
[    6.378234] iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
[    6.378260] iwlwifi 0000:00:14.3: Transport status: 0x0000004B, valid: 6
[    6.378293] iwlwifi 0000:00:14.3: Loaded firmware version: 77.f92b5fed.0 so-a0-hr-b0-77.ucode
[    6.378326] iwlwifi 0000:00:14.3: 0x00000071 | NMI_INTERRUPT_UMAC_FATAL
[    6.378355] iwlwifi 0000:00:14.3: 0x00008A16 | trm_hw_status0
[    6.378379] iwlwifi 0000:00:14.3: 0x00000000 | trm_hw_status1
[    6.378402] iwlwifi 0000:00:14.3: 0x004D930E | branchlink2
[    6.378428] iwlwifi 0000:00:14.3: 0x004CF3DA | interruptlink1
[    6.378452] iwlwifi 0000:00:14.3: 0x004CF3DA | interruptlink2
[    6.378475] iwlwifi 0000:00:14.3: 0x0000B7F0 | data1
[    6.378498] iwlwifi 0000:00:14.3: 0x00000010 | data2
[    6.378522] iwlwifi 0000:00:14.3: 0x00000000 | data3
[    6.378547] iwlwifi 0000:00:14.3: 0x000441F5 | beacon time
[    6.378575] iwlwifi 0000:00:14.3: 0x003C8B85 | tsf low
[    6.378600] iwlwifi 0000:00:14.3: 0x00000000 | tsf hi
[    6.378624] iwlwifi 0000:00:14.3: 0x00000000 | time gp1
[    6.378648] iwlwifi 0000:00:14.3: 0x003D4397 | time gp2
[    6.378673] iwlwifi 0000:00:14.3: 0x00000001 | uCode revision type
[    6.378702] iwlwifi 0000:00:14.3: 0x0000004D | uCode version major
[    6.378732] iwlwifi 0000:00:14.3: 0xF92B5FED | uCode version minor
[    6.378762] iwlwifi 0000:00:14.3: 0x00000370 | hw version
[    6.378788] iwlwifi 0000:00:14.3: 0x00C80002 | board version
[    6.378817] iwlwifi 0000:00:14.3: 0x8044FF00 | hcmd
[    6.378841] iwlwifi 0000:00:14.3: 0xE6E20000 | isr0
[    6.378863] iwlwifi 0000:00:14.3: 0x61000000 | isr1
[    6.378886] iwlwifi 0000:00:14.3: 0x48F00002 | isr2
[    6.378908] iwlwifi 0000:00:14.3: 0x00C3400C | isr3
[    6.378931] iwlwifi 0000:00:14.3: 0x00000000 | isr4
[    6.378967] iwlwifi 0000:00:14.3: 0x0102001C | last cmd Id
[    6.378997] iwlwifi 0000:00:14.3: 0x0000B7F0 | wait_event
[    6.379023] iwlwifi 0000:00:14.3: 0x000000C4 | l2p_control
[    6.379050] iwlwifi 0000:00:14.3: 0x00010034 | l2p_duration
[    6.380239] iwlwifi 0000:00:14.3: 0x00000007 | l2p_mhvalid
[    6.381329] iwlwifi 0000:00:14.3: 0x00000000 | l2p_addr_match
[    6.382321] iwlwifi 0000:00:14.3: 0x00000009 | lmpm_pmg_sel
[    6.383147] iwlwifi 0000:00:14.3: 0x00000000 | timestamp
[... 346 lines cut ...]
version_api | ok
installed_addons | EMQX (0.6.2), Studio Code Server (5.15.0), Samba share (12.3.2), ESPHome (2024.8.0), Advanced SSH & Web Terminal (18.0.0), Assist Microphone (1.2.0), Duck DNS (1.18.0), Let's Encrypt (5.1.1), MariaDB (2.7.1), Matter Server (6.4.1), openWakeWord (1.10.0), Piper (1.5.2), CEC Scanner (3.0), Whisper (2.1.2), AirCast (4.2.1), Network UPS Tools (0.13.1), Nginx Proxy Manager (1.0.1), Uptime Kuma (0.12.1), Frigate (Full Access) (0.14.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 5
views | 11
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 12, 2024 at 12:10 AM
-- | --
current_recorder_run | August 21, 2024 at 9:39 PM
estimated_db_size | 230.99 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

sairon, 2024-08-27:
I am able to reproduce it here on my end with the same WiFi card on Beelink EQ12, it seems to be a bug in the kernel/firmware. Is this something that used to work for you in the past? I just tested couple of versions but it seems not. It will be worth trying other distributions (ideally those using 6.6 LTS kernel too) to check if there's some kernel+firmware combination that works.

RCG-SW, 2024-08-27:
> CAUTION: This email originated from outside the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I just got this NUC and is did not work on the previous version of the OS.

domitilo, 2024-08-29:
Same issue here. But it was working previously.

I bought a Beelink Mini S12 Pro on August 21st, whilst at my holiday home.
 I installed HASS OS smoothly following the guide from an Ubuntu USB Drive. I can connect to my WiFi 5g without problem. Configuring everything I want to.
I have updated to 13.1 and the new version of the HA, and moved back to my usual home (different router, different WiFi..) and wireless interface started to fail as described here.

I've tried to revert to previous version of OS and HA. Nothing.
Tried to restore previous backup made on my first installation. Nothing.
Tried completely new installation following the same steps as the first time. Nothing.

Seems there is no way to connect, however the device is working as I can list SSIDs on this home, but impossible to connect.

I'll try to create a new WiFi spot, 2.4Ghz and see what happens.

I'm not sure if it can be related to the issues on the driver for this AX101, or by the type of WiFi network trying to connect to.

mmornati, 2024-08-31:
The same here on a Beelink S12 Pro (having an Intel AX101) Wifi Card.
I did not test in the previous HassOS as I just started migrating from an RPi4, but on the latest one (13.1) is failing loading the firmware.

The WiFi card, as I'm having a second S12 Pro with Ubuntu is working correctly in the latest Ubuntu relaleses (and I tested on this server to prepare the migration on a live Ubuntu and worked too).

I tried different solutions to configure the firmware but so far I found any workaround here.

RCG-SW, 2024-09-04:
Looks like there are new drivers for wifi ...
https://www.intel.com/content/www/us/en/download/824804/intel-wireless-wi-fi-drivers-for-linux.html

flamme-demon, 2025-01-30:
Problem persists with Beelink Mini S13 and Intel Wi-Fi 6 AX101 on 14.2
I wanted to update the firmware to the latest version.

```bash
cd /lib/firmware
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/iwlwifi-so-a0-hr-b0-89.ucode
modprobe -r iwlwifi
```

But I stumbled across a mistake and, not wanting to break the system, I hesitated to go any further.

> modprobe: can't change directory to '/lib/modules': No such file or directory

Is there any way to force the firmware to run on HAOS?

sairon, 2025-01-31:
As generic-x86-64 kernel update is already merged, you can check whether just that didn't fix the issue, as it's not clear if it's a driver or firmware problem. Just update the OS using `ha os update --version 15.0.dev20250130` to see if it got better there. If not, then hopefully linux-firmware update might fix it.

flamme-demon, 2025-01-31:
@sairon I can confirm that I was able to get the wifi card to work in the  :  Operating System 15.0.dev20250130 

```
[    1.653519] iwlwifi 0000:00:14.3: enabling device (0000 -> 0002)
[    1.674176] Modules linked in: snd_intel_dspcfg(+) snd_hda_codec x86_pkg_temp_thermal snd_hwdep snd_hda_core snd_pcm coretemp iwlwifi(+) i915(+) ttm cfg80211 drm_buddy drm_display_helper igen6_edac
[    1.675521] Modules linked in: snd_intel_dspcfg(+) snd_hda_codec x86_pkg_temp_thermal snd_hwdep snd_hda_core snd_pcm coretemp iwlwifi(+) i915(+) ttm cfg80211 drm_buddy drm_display_helper igen6_edac
[    1.677567] iwlwifi 0000:00:14.3: Detected crf-id 0x501, cnv-id 0x80400 wfpm id 0x80000030
[    1.677612] iwlwifi 0000:00:14.3: PCI dev 54f0/0244, rev=0x370, rfid=0x10c000
[    1.677617] iwlwifi 0000:00:14.3: Detected Intel(R) Wi-Fi 6 AX101
[    1.677740] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-89.ucode failed with error -2
[    1.677758] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-88.ucode failed with error -2
[    1.677770] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-87.ucode failed with error -2
[    1.677783] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-86.ucode failed with error -2
[    1.677795] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-85.ucode failed with error -2
[    1.677807] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-hr-b0-84.ucode failed with error -2
[    1.686787] iwlwifi 0000:00:14.3: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.0.2.41
[    1.687031] iwlwifi 0000:00:14.3: loaded firmware version 83.e8f84e98.0 so-a0-hr-b0-83.ucode op_mode iwlmvm
[    1.875865] iwlwifi 0000:00:14.3: WFPM_UMAC_PD_NOTIFICATION: 0x20
[    1.875921] iwlwifi 0000:00:14.3: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[    1.875930] iwlwifi 0000:00:14.3: WFPM_AUTH_KEY_0: 0x90
[    1.875938] iwlwifi 0000:00:14.3: CNVI_SCU_SEQ_DATA_DW9: 0x10
[    1.875999] iwlwifi 0000:00:14.3: Detected RF HR1 B3, rfid=0x10c000
[    1.877306] iwlwifi 0000:00:14.3: RFIm is deactivated, reason = 4
[    1.917969] iwlwifi 0000:00:14.3: base HW address: 5c:b4:7e:b8:c9:28
[    1.987730] iwlwifi 0000:00:14.3 wlo1: renamed from wlan0
[    3.186368] iwlwifi 0000:00:14.3: WFPM_UMAC_PD_NOTIFICATION: 0x20
[    3.186417] iwlwifi 0000:00:14.3: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[    3.186469] iwlwifi 0000:00:14.3: WFPM_AUTH_KEY_0: 0x90
[    3.186482] iwlwifi 0000:00:14.3: CNVI_SCU_SEQ_DATA_DW9: 0x10
[    3.188481] iwlwifi 0000:00:14.3: RFIm is deactivated, reason = 4
[    3.234650] iwlwifi 0000:00:14.3: Registered PHC clock: iwlwifi-PTP, with index: 0
[    3.499480] iwlwifi 0000:00:14.3: Microcode SW error detected. Restarting 0x0.
[    3.501014] iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
[    3.502221] iwlwifi 0000:00:14.3: Transport status: 0x0000004A, valid: 6
[    3.503336] iwlwifi 0000:00:14.3: Loaded firmware version: 83.e8f84e98.0 so-a0-hr-b0-83.ucode
[    3.504356] iwlwifi 0000:00:14.3: 0x00000071 | NMI_INTERRUPT_UMAC_FATAL    
[    3.505280] iwlwifi 0000:00:14.3: 0x000002F0 | trm_hw_status0
[    3.506136] iwlwifi 0000:00:14.3: 0x00000000 | trm_hw_status1
[    3.506930] iwlwifi 0000:00:14.3: 0x004D9024 | branchlink2
[    3.507681] iwlwifi 0000:00:14.3: 0x004CF2F2 | interruptlink1
[    3.508399] iwlwifi 0000:00:14.3: 0x004CF2F2 | interruptlink2
[    3.509099] iwlwifi 0000:00:14.3: 0x00005FF4 | data1
[    3.509796] iwlwifi 0000:00:14.3: 0x00000010 | data2
[    3.510489] iwlwifi 0000:00:14.3: 0x00000000 | data3
[    3.511179] iwlwifi 0000:00:14.3: 0x003B18A4 | beacon time
[    3.511870] iwlwifi 0000:00:14.3: 0x00059089 | tsf low
[    3.512564] iwlwifi 0000:00:14.3: 0x00000000 | tsf hi
[    3.513229] iwlwifi 0000:00:14.3: 0x00000000 | time gp1
[    3.513881] iwlwifi 0000:00:14.3: 0x0006495A | time gp2
[    3.514528] iwlwifi 0000:00:14.3: 0x00000001 | uCode revision type
[    3.515151] iwlwifi 0000:00:14.3: 0x00000053 | uCode version major
[    3.515746] iwlwifi 0000:00:14.3: 0xE8F84E98 | uCode version minor
[    3.516308] iwlwifi 0000:00:14.3: 0x00000370 | hw version
[    3.516840] iwlwifi 0000:00:14.3: 0x00C80002 | board version
[    3.517350] iwlwifi 0000:00:14.3: 0x8095F500 | hcmd
[    3.517841] iwlwifi 0000:00:14.3: 0x00020000 | isr0
[    3.518312] iwlwifi 0000:00:14.3: 0x00000000 | isr1
[    3.518765] iwlwifi 0000:00:14.3: 0x48F00002 | isr2
[    3.519202] iwlwifi 0000:00:14.3: 0x00C3000D | isr3
[    3.519635] iwlwifi 0000:00:14.3: 0x00000000 | isr4
[    3.520057] iwlwifi 0000:00:14.3: 0x00180103 | last cmd Id
[    3.520475] iwlwifi 0000:00:14.3: 0x00005FF4 | wait_event
[    3.520905] iwlwifi 0000:00:14.3: 0x000083AB | l2p_control
[    3.521322] iwlwifi 0000:00:14.3: 0x00000020 | l2p_duration
[    3.521754] iwlwifi 0000:00:14.3: 0x00000003 | l2p_mhvalid
[    3.522171] iwlwifi 0000:00:14.3: 0x00000000 | l2p_addr_match
[    3.522588] iwlwifi 0000:00:14.3: 0x00000009 | lmpm_pmg_sel
[    3.523009] iwlwifi 0000:00:14.3: 0x00000000 | timestamp
[    3.523422] iwlwifi 0000:00:14.3: 0x00002044 | flow_handler
[    3.523885] iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
[... 87 lines cut ...]
    - 192.168.0.41/16
    gateway: 192.168.255.254
    method: auto
    nameservers:
    - 192.168.255.253
    ready: true
  ipv6:
    address:
    - 2a0e:e701:104f:0xxxxxxxxxxxx
    - fe80::e144:81bb:6eb4:a9b8/64
    gateway: fe80::de2c:6eff:fe2d:4937
    method: auto
    nameservers: []
    ready: true
  mac: 5C:B4:7E:B8:C9:28
  primary: false
  type: wireless
  vlan: null
  wifi:
    auth: wpa-psk
    mode: infrastructure
    signal: 49
    ssid: MySSID
supervisor_internet: true
```

sairon, 2025-01-31:
@flamme-demon Nice! Thank you for checking. I will make a note to close this again once we have OS 15 on beta then.