home-assistant/operating-system #3555: OS Upgrade breaks Intel bluetooth on proxmox hosted install

titchy, 2024-08-22:
### Describe the issue you are experiencing

After updating to the latest HAOS bluetooth is no longer available.
Here is a screenshot from the console showing bluetooth errors that were not there previously.
[image]

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

13.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Proxmox KVM VM. I have passed the bluetooth device through to the VM and it was working prior to the upgrade.

### Steps to reproduce the issue

1. Upgrade OS
2. Bluetooth stops
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2024.8.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.46-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 31 August 2024 at 00:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-6.ui.nabu.casa
certificate_status | ready
instance_id | ea5ca937aa654fc895d1b480d8cdabe3
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 30.8 GB
disk_used | 9.7 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Z-Wave JS (0.6.2), Mosquitto broker (6.4.1), ESPHome (2024.8.0), Piper (1.5.2), Whisper (2.1.2), JupyterLab (0.15.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 0
views | 11
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 19 August 2024 at 07:45
-- | --
current_recorder_run | 22 August 2024 at 15:36
estimated_db_size | 357.93 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

sairon, 2024-08-26:
It seems the device disappears from the bus and it's possibly not the result of the OS upgrade itself. First of all try rebooting the VM couple of times and if it doesn't help, cold boot the Proxmox host too. If it keeps being broken, check if reverting to the previous version (using `ha os boot-slot other`) makes it reliably stable again - then we'll get a confirmation it's a guest OS/Linux regression.

titchy, 2024-08-27:
Rebooting the VM had no effect.
Removing the USB pass through and re-adding it to the VM has fixed the issue.
I have had this running under the previous OS version with a fair few reboots and no issues. It was only after the OS update that rebooting would not result in the BT device being available. Hence my linking it to the OS upgrade.
I will keep an eye on it and see if I can work out what happened. But I think for now this should be closed as not OS related.

closed 2024-08-27 by titchy