home-assistant/operating-system #3557: HP t520 fails to load kernel on 13.1

sairon, 2024-08-23:
Yesterday had the same issue while upgrading from 12.4 to 13.0 on HP t520 Thin Client with 4GB RAM and SATA SSD. This system had 12.0 installed several months ago and I did at least 3 or 4 updates without any issue until yesterday... 

Tried SLOT A and B, also recovery console - nothing boots. I have backups on Google Drive - didn't try the restore procedure yet. Just in case - booted live linux and backing up drive contents to my NAS. 

Is there anything I could get from the logs that would help figuring out the reason of this issue?..

UPDATE

Spent some time today troubleshooting this issue:
1. Tried a clean install of HAOS 13.1 on the same hardware - doesn't boot past Grub Menu
2. Tried clean install of HAOS 12.4 on the same hardware - works as expected (made a backup of /dev/sda1 at this point)
3. Restored from the backup --> Reboot - works as expected
4. OS update from 12.4 to 13.1 --> Reboot - doesn't boot past Grub Menu
5. Booted Live Linux, replaced /dev/sda1/.../boot*.efi files with the ones from HAOS 12.4 --> Reboot - everything works as expected.

So it seems my issue is related to https://github.com/home-assistant/operating-system/issues/3305 and it turns out it affects AMD CPUs also...

Attaching dmidecode and "smbios --type 4 --get-qword 8" output :
32386119686159359

[dmidecode.txt](https://github.com/user-attachments/files/16717296/dmidecode.txt)

_Originally posted by @a4mkd9 in https://github.com/home-assistant/operating-system/issues/3153#issuecomment-2304018767_

sairon, 2024-08-23:
@a4mkd9 I've started a [test build](https://github.com/home-assistant/operating-system/actions/runs/10524326696) with the CPU ID added to the exclusions list. Once it finishes (~2 hours from now), could you please test it?

a4mkd9, 2024-08-26:
Hello sairon,

Sorry for the delay - had to find original 16GB m2 SATA drive to avoid reinstalling and recovering everything again in case of failure.
16GB m2 SATA SSD had OS Version 12.0 and Home Assistant Core 2024.3.0. Started up system to make sure everything works as expected. Then inserted USB drive named CONFIG with haos_generic-x86-64-13.2.dev1724421726.raucb in the root and rebooted the host. Then system booted up without any issue and now reports it's running OS Version 13.2.dev1724421726.

sairon, 2024-08-26:
@a4mkd9 Thank you, I'll create a PR to include the fix, which means it will be available on the next OS >13.1, starting with the next RC already. You should be safe running the dev version in the meantime, or using the EFI binaries to fix an older stable release.

hectork2, 2024-09-22:
I am experiencing the exact same issue on an HP T630. When I update from version 12.4 to 13.1, after rebooting, I encounter the following error: Slot A (OK=1 Try=3) I have even tried clean installing both versions 13 and 13.1, but I still get the same error. I have also updated my BIOS, but the issue persists.

commit d289f0ae (2024-08-26): Use GRUB legacy loader for HP t520 to fix boot with GRUB 2.12

HP t520 have been reported to have the same issues as Atom boards with bad UEFI
firmware that doesn't work well with the new EFI loader used since GRUB 2.12.
Apply the patch to use legacy loader for its CPU ID as well.

Fixes #3557
PR #3562 (2024-08-26): Use GRUB legacy loader for HP t520 to fix boot with GRUB 2.12
HP t520 have been reported to have the same issues as Atom boards with bad UEFI firmware that doesn't work well with the new EFI loader used since GRUB 2.12. Apply the patch to use legacy loader for its CPU ID as well.

Fixes #3557
commit 5eb7ccb7 (2024-08-26): Use GRUB legacy loader for HP t520 to fix boot with GRUB 2.12 (#3562)

HP t520 have been reported to have the same issues as Atom boards with bad UEFI
firmware that doesn't work well with the new EFI loader used since GRUB 2.12.
Apply the patch to use legacy loader for its CPU ID as well.

Fixes #3557