home-assistant/operating-system #3559: HA OS (generic x86_64) upgraded to OS 13.1 - WiFi (Intel(R) AX210) is gone again

bearhntr, 2024-08-24:
### Describe the issue you are experiencing

I took the plunge today.  Took a FULL Backup of HA OS (on ThinClient T620) and installed the 13.1 upgrade to the OS.  As soon as it came back up, again the WiFi adapter is not seen.  I was running 12.1 which worked fine.  The card in the T620 is m.2 AX210 with WiFi 6E and BT.  The BT is working.  

Only one Interface showing again:
[image]

Take it back to 12.1 and both interfaces are there:
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.20-haos

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

HP T620 ThinClient - 16GB RAM with 120GB SSD (SATA/m.2) and INTEL AX210 m.2 card
USB **Zooz_ZST10_700_Z-Wave_Stick** 
USB **Sonoff Zigbee 3.0 USB Dongle Plus**

### Steps to reproduce the issue

1.  Went to the UPDATES page in HA
2.  Installed OS Update from 12.1 >> 13.1 
3.  Rebooted
4.  No WiFi
5.  Ran **_ha os update --version 12.1_** - to put it back to working version
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-08-24 12:45:14.054 INFO (MainThread) [__main__] Running Supervisor
2024-08-24 12:45:14.067 INFO (MainThread) [supervisor.os.manager] Rauc: B - marked slot kernel.1 as good
2024-08-24 12:45:14.094 INFO (MainThread) [supervisor.addons.manager] Phase 'initialize' starting 0 add-ons
2024-08-24 12:45:14.101 INFO (MainThread) [supervisor.addons.manager] Phase 'system' starting 2 add-ons
2024-08-24 12:45:14.127 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass
2024-08-24 12:45:14.128 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-08-24 12:45:14.128 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-08-24 12:45:14.137 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass
2024-08-24 12:45:14.138 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-08-24 12:45:14.138 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-08-24 12:45:14.707 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-mariadb with version 2.7.1
2024-08-24 12:45:15.359 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/zwave-js-ui/amd64 with version 3.10.0
2024-08-24 12:45:16.329 WARNING (MainThread) [supervisor.services.modules.mysql] Can't remove not exists services
2024-08-24 12:45:16.744 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass
2024-08-24 12:45:16.746 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-08-24 12:45:16.746 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-08-24 12:45:17.501 INFO (MainThread) [supervisor.services.modules.mysql] Set core_mariadb as service provider for MySQL
2024-08-24 12:45:45.458 INFO (MainThread) [supervisor.addons.manager] Phase 'services' starting 3 add-ons
2024-08-24 12:45:45.508 WARNING (MainThread) [supervisor.docker.addon] Advanced SSH & Web Terminal running with disabled protected mode!
2024-08-24 12:45:45.884 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/ssh/amd64 with version 18.0.0
2024-08-24 12:45:46.227 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-samba with version 12.3.2
2024-08-24 12:45:46.788 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_samba
2024-08-24 12:45:46.975 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/vscode/amd64 with version 5.15.0
2024-08-24 12:46:17.317 INFO (MainThread) [supervisor.core] Start Home Assistant Core
2024-08-24 12:46:17.387 INFO (SyncWorker_0) [supervisor.docker.manager] Starting homeassistant
2024-08-24 12:46:17.648 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
2024-08-24 12:46:25.446 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state startup
2024-08-24 12:46:25.446 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-08-24 12:46:25.585 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass
2024-08-24 12:46:25.585 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
[... 534 lines cut ...]
2024-08-24 16:44:51.149 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00000080 | IML/ROM WFPM_AUTH_KEY_0
2024-08-24 16:44:51.154 homeassistant kernel: iwlwifi 0000:01:00.0: Fseq Registers:
2024-08-24 16:44:51.154 homeassistant kernel: iwlwifi 0000:01:00.0: 0x60000100 | FSEQ_ERROR_CODE
2024-08-24 16:44:51.155 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00440007 | FSEQ_TOP_INIT_VERSION
2024-08-24 16:44:51.156 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00080009 | FSEQ_CNVIO_INIT_VERSION
2024-08-24 16:44:51.157 homeassistant kernel: iwlwifi 0000:01:00.0: 0x0000A652 | FSEQ_OTP_VERSION
2024-08-24 16:44:51.158 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00000002 | FSEQ_TOP_CONTENT_VERSION
2024-08-24 16:44:51.159 homeassistant kernel: iwlwifi 0000:01:00.0: 0x4552414E | FSEQ_ALIVE_TOKEN
2024-08-24 16:44:51.160 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00400410 | FSEQ_CNVI_ID
2024-08-24 16:44:51.161 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00400410 | FSEQ_CNVR_ID
2024-08-24 16:44:51.162 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00400410 | CNVI_AUX_MISC_CHIP
2024-08-24 16:44:51.164 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00400410 | CNVR_AUX_MISC_CHIP
2024-08-24 16:44:51.164 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00009061 | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
2024-08-24 16:44:51.165 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00000061 | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
2024-08-24 16:44:51.166 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00080009 | FSEQ_PREV_CNVIO_INIT_VERSION
2024-08-24 16:44:51.168 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00440005 | FSEQ_WIFI_FSEQ_VERSION
2024-08-24 16:44:51.169 homeassistant kernel: iwlwifi 0000:01:00.0: 0x00440007 | FSEQ_BT_FSEQ_VERSION
2024-08-24 16:44:51.171 homeassistant kernel: iwlwifi 0000:01:00.0: 0x000000DC | FSEQ_CLASS_TP_VERSION
2024-08-24 16:44:51.172 homeassistant kernel: iwlwifi 0000:01:00.0: UMAC CURRENT PC: 0x8048d3e4
2024-08-24 16:44:51.173 homeassistant kernel: iwlwifi 0000:01:00.0: LMAC1 CURRENT PC: 0xd0
2024-08-24 16:44:51.174 homeassistant kernel: iwlwifi 0000:01:00.0: WRT: Collecting data: ini trigger 13 fired (delay=0ms).
2024-08-24 16:44:51.934 homeassistant kernel: iwlwifi 0000:01:00.0: Failed to run INIT ucode: -5
2024-08-24 16:44:51.947 homeassistant kernel: iwlwifi 0000:01:00.0: retry init count 2
2024-08-24 16:44:52.757 ha NetworkManager[469]: <info>  [1724517892.7573] Loaded device plugin: NMWifiFactory (/usr/lib/NetworkManager/1.44.2/libnm-device-plugin-wifi.so)
(END)

sairon, 2024-08-25:
See #3556, it should be _really_ fixed in the latest dev.

closed 2024-08-25 by sairon

bearhntr, 2024-08-25:
> See #3556, it should be _really_ fixed in the latest dev.

How do I assure that I can get this working?  I did the 13.1 upgrade.  Rebooted the physical hardware as well - and the WiFi is not seen.  

I find it odd that 12.1 works just fine but 12.2, 12.3, 12.4 did not.  Then 13.1 did not either.

sairon, 2024-08-26:
> I find it odd that 12.1 works just fine but 12.2, 12.3, 12.4 did not. Then 13.1 did not either.

It's not surprising knowing what happened. OS 12.2 brought update of Buildroot, where a downstream change adding .pnvm files required by the driver went missing. I originally thought the regression was caused by the firmware compression that we introduced at the same time, because I misread the filename and assumed it is present in the latest OS, but then I realized there's only one file of very similar name (for the AX211 card), the one for AX210 was missing. Details are in the linked PR.

> How do I assure that I can get this working? I did the 13.1 upgrade. Rebooted the physical hardware as well - and the WiFi is not seen.

In fact, I will also appreciate if you help me test it is working! There are two options:

1. Flashing HAOS to some removable drive and use BIOS boot menu to temporarily boot from that.
2. Switch to latest dev version using `ha os update --version 13.2.dev20240823`. Don't worry it's a dev, there are no changes that should introduce major system instability. You might need to enable beta channel in the [Updates](https://my.home-assistant.io/redirect/updates/) settings first. Once you finish testing, you can roll back to the previous version using `ha os boot-slot other` - but if it fixes your problems, you might as well keep the dev version installed and wait before you get update notification for next rc/stable version.

sairon, 2024-08-27:
> if I do this route - I build a bootable USB - just I like made the original m.2 SATA? Just get the ThinClient to boot from the USB?

Right. HAOS can run from the USB drive alone as well.

> It is going to be like a brand new install, I take it - and none of my stuff will be in there? You just want to see if my WiFi and Bluetooth (from the AX210) is there?

Exactly, but actually it might be good to pull out the M.2 disk out for this test because otherwise it would appear as a new install and would mark your data disk as disabled (something about that [here](https://community.home-assistant.io/t/behind-the-scenes-of-the-move-data-disk-feature/666664/1)). It's not something it can't be undone but I think it's not worth the trouble. Ultimately I think the other options takes less effort and is less risky too.

> I am building a Proxmox installation of HA that I will eventually move to - it has the same AX210 in it

If you're not going to use PCIe passthrough for sharing the card with the VM (which I **do not** recommend, unless you have very strong reason to do so), it will use the drivers, firmware and network stack of the Proxmox host. You can still use it for wireless connection but everything will be configured on the Proxmox host. HAOS kernel isn't involved in any way, so I don't need any information from that system at this point.

> I am not at all familiar with this **boot -slot** thing.

You can read up on that [in docs](https://www.home-assistant.io/common-tasks/os/#about-boot-slots-used-during-the-update). In short, you don't need to do another update/downgrade when some version doesn't work well for you, just use the boot-slot command to switch to the previous installed version.

Also, this thing I said yesterday is no longer true (thanks to [this change](https://github.com/home-assistant/version/pull/389)):

> You might need to enable beta channel in the [Updates](https://my.home-assistant.io/redirect/updates/) settings first. 

So all you need to do is to run `ha os update --version 13.2.dev20240823` in CLI, check if the WiFi card works after updating, and then either stay on that version or run `ha os boot-slot other`, which reboots into the version that was installed before you ran the update command.

bearhntr, 2024-08-27:
> > if I do this route - I build a bootable USB - just I like made the original m.2 SATA? Just get the ThinClient to boot from the USB?
> 
> Right. HAOS can run from the USB drive alone as well.
> 
It took 3 times building the USB (the first was on an 8GB - apparently it was damaged or too small.  I got to the end and styed for hours with black screen (at the top "Waiting for Home Assistant CLI to be ready..."  I finally just turned it off.  When I turned it back on, it gave some kind of error loading something.  I know you have to rebuild the USB at this point.  Which I did, to a 16GB USB.  It failed as well.  I re-downloaded the .xz file and extracted it (please note I am doing these from a Windows 10 Pro PC).  Then I built it on a brand new 32GB USB 3.1 USB (freshly taken out of the packaging).  This time it booted all the way to here (see image).  I let it sit there for 30 mins.  Then rebooted it with  'host reboot' command in the CLI and it came up to the 2nd image.  Notice there is no WiFi Adapter (but - I think I turned too many things OFF in the BIOS).  

In the BIOS I can disable the m.2 drive; and I might have turned OFF the WIFI card too, by mistake.  Going to check.  

[image]

[image]

13.1 and 2024.8.3 --  YAY

bearhntr, 2024-08-27:
OK - I found the setting I disabled - and re-enabled it.  I watched as it booted.  I actually saw the error on the screen and took a picture (attached).  It states there is a PNVM file missing.

[image]

sairon, 2024-08-27:
> Notice there is no WiFi Adapter (but - I think I turned too many things OFF in the BIOS).

If it's visible by the system, typing `login` (to enter the root shell) and then `dmesg | grep iwlwifi` should produce some output.

But the problem is that you're NOT running the development version that's supposed to fix it - you're on 13.1. You'd have to download the development image from [os-artifacts page](https://os-artifacts.home-assistant.io/index.html). Alternatively, with the system running from the USB drive, you can run the `ha os update` command as suggested before, this time not affecting your production instance.

bearhntr, 2024-08-27:
> If it's visible by the system, typing `login` (to enter the root shell) and then `dmesg | grep iwlwifi` should produce some output.
>
> But the problem is that you're NOT running the development [...]

OK - I have downloaded the `haos_generic-x86-64-13.2.dev20240823.img.xz`  and I am making a USB with it.  I will test this one and let you know.

bearhntr, 2024-08-27:
The install failed --- I downloaded the 'developer edition' and made the same USB that I was using for the 13.1 version.  I booted up, and left the room.  Came back about 20 mins later to this:

[image]

I typed 'EXIT' and it looked like it was going to continue - then about 12 mins later...I saw this:

[image]

It has been sitting on that screen now 30+ minutes.  I did nothing more, no reboot or commands.

bearhntr, 2024-08-27:
I HARD BOOTED it again - and this time it appeared to 'complete' what ever it was trying to do.  I saw this:

[image]

Then I went to browser and went through the setup wizard until I got to here:

[image]

Completed the setup of the WiFi in the webpage - performed another reboot...and I now have this:

[image]

...and the DHCP reservations worked too.

sairon, 2024-08-27:
That's great, thank you for the confirmation and for going through all the trouble! I hope it will work for you in the next stable release without any further hiccups. Unfortunately, Intel WiFi cards break as well in other Linux distributions under certain setups from time to time. Just in case you encounter any problems in the future, the cause of it will be probably different, in that case please open a new ticket with all requested information. Thanks!

bearhntr, 2024-08-27:
> That's great, thank you for the confirmation and for going through all the trouble! I hope it will work for you in the next stable release without any further hiccups. Unfortunately, Intel WiFi [...]

Thank You for your help.  I will leave my "regular HA" in place and boot off the internal m.2 for now.  Not usually one to run BETAs for too long.  

It was my pleasure to do the testing.  

Do you know of any add-ons for monitoring or viewing WiFi stuff?

bearhntr, 2024-09-05:
@sairon - Any idea when 13.2 (I am hoping the fix is gonna be in that version) will be released?

bearhntr, 2024-10-17:
@sairon ---   THANK YOU for the fix.  I noticed that 13.2 was out there for install.  I installed it this AM, and I still have my Wi-Fi

[image]

[image]