home-assistant/operating-system #3561: Realtek RTL8821CE kernel panics the system (virtio)

Chicchi7393, 2024-08-25:
### Describe the issue you are experiencing

Hi,
I was interested to do a pci passthrough of my wifi+bt card, the rtl8821ce, to my libvirt HAOS, but on boot it kernel panics spitting out this error on the serial console

```
[    1.996135] BUG: kernel NULL pointer dereference, address: 000000000000003c
[    1.996953] #PF: supervisor read access in kernel mode
[    1.997572] #PF: error_code(0x0000) - not-present page
[    1.998179] PGD 0 P4D 0 
[    1.998471] Oops: 0000 [#1] PREEMPT SMP NOPTI
[    1.998958] CPU: 1 PID: 169 Comm: (udev-worker) Not tainted 6.6.46-haos #1
[    1.999805] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS 0.0.0 02/06/2015
[    2.000764] RIP: 0010:rtw_pci_probe+0x91c/0x1b30 [rtw88_pci]
[    2.001312] Code: 48 c7 c6 48 13 32 c0 e8 42 92 d3 cc 49 8b 7c 24 08 48 c7 c6 98 13 32 c0 e8 31 92 d3 cc 48 89 ef e8 d9 9c 9c cc e9 3d fd ff ff <66> 41 81 7e 3c 86 80 0f 85 bd fd ff ff 41 c6 84 24 18 7b 00 00 01
[    2.003318] RSP: 0018:ffffc900006d7b28 EFLAGS: 00010246
[    2.003966] RAX: 0000000000000000 RBX: ffffffffc0297040 RCX: 0000000000000000
[    2.004887] RDX: ffff88810b51548b RSI: ffff88810b512b62 RDI: ffff88810b512376
[    2.005814] RBP: ffff888100a0b000 R08: ffff88810b512376 R09: 0000000000000003
[    2.006736] R10: ffff88810b51548b R11: ffff88810b512b62 R12: ffff88810b512040
[    2.007639] R13: ffff88810b517778 R14: 0000000000000000 R15: 0000000000000200
[    2.008552] FS:  00007f8ad1fa9840(0000) GS:ffff8881b9c80000(0000) knlGS:0000000000000000
[    2.009612] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    2.010377] CR2: 000000000000003c CR3: 0000000107c60005 CR4: 0000000000370ee0
[    2.011274] Call Trace:
[    2.011614]  <TASK>
[    2.011905]  ? __die+0x1e/0x60
[    2.012308]  ? page_fault_oops+0x178/0x490
[    2.012836]  ? exc_page_fault+0x6d/0x150
[    2.013337]  ? asm_exc_page_fault+0x26/0x30
[    2.013903]  ? rtw_pci_probe+0x91c/0x1b30 [rtw88_pci]
[    2.014574]  ? rtw_pci_probe+0x6cd/0x1b30 [rtw88_pci]
[    2.015226]  local_pci_probe+0x3a/0x90
[    2.015765]  pci_device_probe+0xab/0x180
[    2.016302]  really_probe+0x15a/0x2c0
[    2.016805]  ? __pfx___driver_attach+0x10/0x10
[    2.017377]  __driver_probe_device+0x6e/0x110
[    2.018000]  driver_probe_device+0x1a/0xe0
[    2.018563]  __driver_attach+0x83/0x180
[    2.019109]  bus_for_each_dev+0x80/0xd0
[    2.019631]  bus_add_driver+0xe3/0x210
[    2.020147]  driver_register+0x54/0x100
[    2.020665]  ? __pfx_rtw_8821ce_driver_init+0x10/0x10 [rtw88_8821ce]
[    2.021544]  do_one_initcall+0x55/0x230
[    2.022136]  do_init_module+0x5b/0x200
[    2.022653]  init_module_from_file+0x81/0xc0
[    2.023294]  idempotent_init_module+0x11b/0x310
[    2.023906]  __x64_sys_finit_module+0x51/0x90
[    2.024524]  do_syscall_64+0x38/0x80
[    2.025038]  entry_SYSCALL_64_after_hwframe+0x78/0xe2
[    2.025707] RIP: 0033:0x7f8ad27be6a9
[    2.026142] Code: ff c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 47 97 0c 00 f7 d8 64 89 01 48
[    2.028585] RSP: 002b:00007ffc14e1d758 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    2.029523] RAX: ffffffffffffffda RBX: 0000000000000000 RCX: 00007f8ad27be6a9
[    2.030374] RDX: 0000000000000000 RSI: 00007f8ad28aaafd RDI: 0000000000000013
[    2.031182] RBP: 000055d40f9c5bd0 R08: 0000000000000040 R09: 0000000000000000
[    2.032053] R10: 0000000000000038 R11: 0000000000000246 R12: 00007f8ad28aaafd
[    2.033009] R13: 0000000000020000 R14: 000055d40f961bc0 R15: 0000000000000000
[    2.033939]  </TASK>
[    2.034215] Modules linked in: sch_fq_codel rtw88_8821ce(+) rtw88_8821c rtw88_pci rtw88_core mac80211 libarc4 cfg80211 cdc_acm virtio_balloon dmi_sysfs qemu_fw_cfg
[    2.036212] CR2: 000000000000003c
[    2.036667] ---[ end trace 0000000000000000 ]---
[    2.037276] RIP: 0010:rtw_pci_probe+0x91c/0x1b30 [rtw88_pci]
[    2.038001] Code: 48 c7 c6 48 13 32 c0 e8 42 92 d3 cc 49 8b 7c 24 08 48 c7 c6 98 13 32 c0 e8 31 92 d3 cc 48 89 ef e8 d9 9c 9c cc e9 3d fd ff ff <66> 41 81 7e 3c 86 80 0f 85 bd fd ff ff 41 c6 84 24 18 7b 00 00 01
[    2.040217] RSP: 0018:ffffc900006d7b28 EFLAGS: 00010246
[    2.040972] RAX: 0000000000000000 RBX: ffffffffc0297040 RCX: 0000000000000000
[    2.042034] RDX: ffff88810b51548b RSI: ffff88810b512b62 RDI: ffff88810b512376
[    2.043053] RBP: ffff888100a0b000 R08: ffff88810b512376 R09: 0000000000000003
[    2.044077] R10: ffff88810b51548b R11: ffff88810b512b62 R12: ffff88810b512040
[    2.045096] R13: ffff88810b517778 R14: 0000000000000000 R15: 0000000000000200
[... 350 lines cut ...]
resources | 10
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 16 agosto 2024 alle ore 12:06
-- | --
current_recorder_run | 25 agosto 2024 alle ore 20:49
estimated_db_size | 57.43 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Sonoff</summary>

version | 3.7.3 (e240aaf)
-- | --
cloud_online | 1 / 2
local_online | 1 / 1

</details>

sairon, 2024-08-26:
PCIe passthrough is tricky, even though it ultimately manifests as a stack trace in the guest OS, it may be caused by the host OS or any hardware settings. Can you check if it works with another Linux distribution as the guest?

Chicchi7393, 2024-09-01:
Sorry for the late reply, I've been quite busy this past days and i kinda forgot. Anyways I've managed to boot debian testing with the passtrough, so it isn't an host os error

Chicchi7393, 2024-09-05:
@sairon sorry to mention you, you might have missed the issue

sairon, 2024-09-10:
I did not, I just didn't have an opportunity to look into that - until now. I am able to reproduce the issue with this card too and confirm it really works on Debian-based distros (tested latest stable Debian and Ubuntu). However, it fails the same way on Arch Linux (both with 6.6 linux-lts and 6.10 linux), and when I applied Ubuntu's kernel config to HAOS build (with some minor modification that are required for successful boot), the problem persisted, so it's unclear to me yet why it works on Debian. There are still some more things to try, for example testing of other mainstream distributions, ideally those running the same minor kernel version (i.e. 6.6.y) without many patches. That way we could start "bisecting" the configuration to find what's needed to make it working.

closed 2024-12-18 by github-actions[bot]