home-assistant/operating-system #3564: Host audio not working after 13.0 -> 13.1 upgrade

davidtinker, 2024-08-26:
### Describe the issue you are experiencing

Onboard audio on the host no longer works after 13.0 -> 13.1 upgrade. I verified by downgrading to 13.0. `pactl list sinks` doesn't show sink for it on 13.1. Audio to a bluetooth speaker still works.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

13.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

It is a Beelink S12 Pro Mini PC, Intel N100.

### Steps to reproduce the issue

[core-ssh ~]$ pactl list sinks
(built in audio sink not shown)

Connecting a bluetooth speaker works.

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing interesting
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing interesting
```

### Additional information

## HA OS 13.0

After connecting bluetooth speaker:

```
[core-ssh ~]$ pactl list sinks
Sink #0
	State: IDLE
	Name: alsa_output.pci-0000_00_1f.3.analog-stereo
	Description: Built-in Audio Analog Stereo
	Driver: module-alsa-card.c
	Sample Specification: s16le 2ch 44100Hz
	Channel Map: front-left,front-right
	Owner Module: 6
	Mute: no
	Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
	        balance 0.00
	Base Volume: 65536 / 100% / 0.00 dB
	Monitor Source: alsa_output.pci-0000_00_1f.3.analog-stereo.monitor
	Latency: 1853313 usec, configured 2000000 usec
	Flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY
	Properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "ALC897 Analog"
		alsa.id = "ALC897 Analog"
[... 200 lines cut ...]
snd_hda_codec_generic    98304  1 snd_hda_codec_realtek
snd_hda_codec_hdmi     81920  1
snd_hda_codec_realtek   188416  1
snd_hda_core          126976  9 snd_hda_codec_hdmi,snd_hda_codec_realtek,snd_hda_codec_generic,snd_sof_intel_hda_common,snd_sof_intel_hda,snd_soc_hdac_hda,snd_hda_ext_core,snd_hda_intel,snd_hda_codec
snd_hda_ext_core       28672  4 snd_sof_intel_hda_common,snd_sof_intel_hda_mlink,snd_sof_intel_hda,snd_soc_hdac_hda
snd_hda_intel          49152  0
snd_hwdep              12288  1 snd_hda_codec
snd_intel_dspcfg       32768  3 snd_sof_intel_hda_common,snd_sof,snd_hda_intel
snd_pcm               172032  9 snd_hda_codec_hdmi,snd_sof_intel_hda_common,snd_sof,snd_sof_utils,snd_soc_core,snd_compress,snd_hda_intel,snd_hda_codec,snd_hda_core
snd_soc_acpi           12288  2 snd_sof_intel_hda_common,snd_soc_acpi_intel_match
snd_soc_acpi_intel_match    94208  2 snd_sof_pci_intel_tgl,snd_sof_intel_hda_common
snd_soc_core          348160  3 snd_sof_intel_hda_common,snd_sof,snd_soc_hdac_hda
snd_soc_hdac_hda       20480  1 snd_sof_intel_hda_common
snd_sof               294912  3 snd_sof_intel_hda_common,snd_sof_intel_hda,snd_sof_pci
snd_sof_intel_hda      16384  1 snd_sof_intel_hda_common
snd_sof_intel_hda_common   180224  1 snd_sof_pci_intel_tgl
snd_sof_intel_hda_mlink    28672  1 snd_sof_intel_hda_common
snd_sof_pci            16384  2 snd_sof_pci_intel_tgl,snd_sof_intel_hda_common
snd_sof_pci_intel_tgl    12288  0
snd_sof_utils          12288  1 snd_sof
snd_sof_xtensa_dsp     12288  1 snd_sof_intel_hda_common
ttm                    90112  1 i915
x86_pkg_temp_thermal    16384  0
xfrm_user              57344  1
```

davidtinker, 2024-08-27:
This time it worked fine? Here are the dmesg outputs. Tx for looking into it. I should have done another reboot before filling a bug.

[dmesg_13_0.txt](https://github.com/user-attachments/files/16764360/dmesg_13_0.txt)
[dmesg_13_1.txt](https://github.com/user-attachments/files/16764362/dmesg_13_1.txt)

magicus, 2024-09-01:
I was just about to comment that I had the same problem. I thought I had rebooted (several times), but thought "what the heck, I'll do it once more just to be sure before commenting here", and sure enough, the sink re-appeared!

So something is a bit fishy with this upgrade, but rebooting is obviously a work-around. I'm just worried now that it is a bit random if it works after rebooting or not, so I'll try to avoid rebooting until there is an upgrade of hassOS.

closed 2024-12-09 by github-actions[bot]