home-assistant/operating-system #3565: Please update systemd to v256

IngmarStein, 2024-08-26:
### Describe the issue you are experiencing

https://github.com/systemd/systemd/pull/32359 was merged into systemd v256 (see [NEWS](https://github.com/systemd/systemd/blob/v256-stable/NEWS)) and fixed https://github.com/systemd/systemd/issues/28001.

This bug is triggered when running the OVA with QEMU with [HyperV Enlightenment](https://www.qemu.org/docs/master/system/i386/hyperv.html) enabled. With systemd versions before v256, `systemd-detect-virt` returned `microsoft` (expected: `kvm`) which then prevents the QEMU guest agent from being started (`ConditionVirtualization=|kvm was not met, ConditionVirtualization=|qem was not met`).

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

13.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Synology Virtual Machine Manager 2.6.5-12202

### Steps to reproduce the issue

1. create a VM using the current OVA image
2. enable Hyper-V Enlightenments
3. start the VM
4. run `systemctl status qemu-guest.service` to confirm that QEMU guest agent is not started

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

agners, 2024-08-27:
We usually track systemd as shipped by buildroot. Updating is always associated with a risk. That said, there is a systemd stable repo, maybe this bugfix has been backported? :thinking: 

Reading up on Hyper-V enlightenment, it seems that this is specifically mean for Windows. Can this CPU extensions maybe disabled for the HAOS VM? :thinking:

IngmarStein, 2024-08-27:
Yes, Hyper-V Enlightenments is a feature originally designed by Microsoft for Windows, but it was expanded to Linux guests where it can also be benefitial: https://events19.lfasiallc.com/wp-content/uploads/2017/11/Use-Hyper-V-Enlightenments-to-Increase-KVM-VM-Performance_Density_Chao-Peng.pdf

Of course, it can be disabled for the HAOS VM, in which case the virtualization detection works just find and the QEMU Guest Agent is started.

Looking at https://github.com/systemd/systemd-stable/blob/a2a7968701a4ed5e35bcd9ee2b2b0473ec949773/src/basic/virt.c#L502-L509, it doesn't seem like this patch has been backported to v254. If it was, would it be picked up by HAOS? The diff doesn't seem huge, so maybe there's a chance it would be accepted.

IngmarStein, 2024-09-18:
@agners: systemd 256.4 is now available in buildroot: https://gitlab.com/buildroot.org/buildroot/-/commit/127c42033590e73e3161852acd540a79c98d9f3d

commit 0fdf920b (2024-09-25): Fix guest-agent not starting with HyperV Enlightenments enabled (#3592)

Guest agent doesn't start because if HyperV Enlightenments are enabled, the
virtualization gets detected incorrectly. Backport Systemd patch that fixes the
detection, allowing the guest-agent service to meet its dependencies.

This patch should be no longer needed after update of Systemd to v256, or in
case the patch gets eventually backported to the v254 stable branch.

Fixes #3565
closed 2024-09-25 by sairon