home-assistant/operating-system #3571: Error updating Home Assistant Operating System: Cannot connect to host github.com:443

p-mitana, 2024-09-01:
### Describe the issue you are experiencing

In short – Home Assistant lists an available update, but I cannot install it – it says it cannot connect to GitHub. Running `ha os update` command also fails. Copying the link and downloading the file via `wget` works fine though, the file starts being downloaded:

[image]

There already has been an issue for that – #3525 – but it has been closed and said to apparently be a network error. Also there were suspicions that it could be related but some GitHub problems. However:

- I don't expect it to be a network problem, as I can fetch the exact same file from the same terminal session that I cannot update via from which the `ha os update` command fails
- I don't also expect it to be related with any GitHub failure, as a couple of weeks have passed and the problem persists
- Also, after 12.4 -> 13.0 update failed, the 13.1 came out and I still cannot update from 12.4.
- Also, I have seen other reports from other users in the forum having the same problem – for example in [this topic](https://community.home-assistant.io/t/haos-update-failing/743068).

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

USB devices – 2x SkyConnect. HAOS is installed in a VM. Host is Ubuntu 24.04 on a NUC type x86-64 computer. Hypervisor is KVM I expect.

### Steps to reproduce the issue

1. Enter terminal
2. Run `ha os update`
3. Alternatively, just try to update OS via GUI

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-09-01 11:49:19.304 INFO (MainThread) [supervisor.api.middleware.security] /os/update access from a0d7b954_ssh
2024-09-01 11:49:22.992 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://os-artifacts.home-assistant.io/13.1/haos_ova-13.1.raucb
2024-09-01 11:50:28.257 ERROR (MainThread) [supervisor.os.manager] Can't fetch OTA update from https://os-artifacts.home-assistant.io/13.1/haos_ova-13.1.raucb: Cannot connect to host github.com:443 ssl:default [None]
2024-09-01 11:51:15.845 INFO (MainThread) [supervisor.api.middleware.security] /os/update access from a0d7b954_ssh
2024-09-01 11:51:19.532 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://os-artifacts.home-assistant.io/13.1/haos_ova-13.1.raucb
2024-09-01 11:52:23.235 ERROR (MainThread) [supervisor.os.manager] Can't fetch OTA update from https://os-artifacts.home-assistant.io/13.1/haos_ova-13.1.raucb: Cannot connect to host github.com:443 ssl:default [None]
2024-09-01 12:04:39.779 INFO (MainThread) [supervisor.api.middleware.security] /os/update access from a0d7b954_ssh
2024-09-01 12:04:45.468 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://os-artifacts.home-assistant.io/13.1/haos_ova-13.1.raucb
2024-09-01 12:05:49.202 ERROR (MainThread) [supervisor.os.manager] Can't fetch OTA update from https://os-artifacts.home-assistant.io/13.1/haos_ova-13.1.raucb: Cannot connect to host github.com:443 ssl:default [None]
```

### Anything in the Host logs that might be useful for us?

```txt
I can't see anything related
```

sairon, 2024-09-02:
There's indeed something fishy... I have replied [on the community forums](https://community.home-assistant.io/t/haos-update-failing/743068/24?u=sairon), let's follow up there.

p-mitana, 2024-10-07:
[This comment](https://community.home-assistant.io/t/haos-update-failing/743068/44?u=mipioter) might help.

Actually my issue was solved by replacing a quite old router that I had. Still, I expect the issue is be relevant for some users and specific configurations.

closed 2025-01-14 by github-actions[bot]