home-assistant/operating-system #3572: Waiting for home assistant cli to be ready (every power loss and most upgrades)

CrAzYDr1veR, 2024-09-01:
### Describe the issue you are experiencing

Hi, so every time i have a power loss (even ups switching from grid to battery) or most of the times the system updates
my home assistant pi 3 becomes a brick.
Is there a way to make this more resilient to power losses and less corruption of the file system?

I had a mini powerbank as a ups but it takes too long to switch from charging to discharging and powers off anyway.
This takes quite a while to reinstall everything even from imager and the backup of my slowpi3.

Any information you need i can provide if you want.

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

13

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

pi3b+

### Steps to reproduce the issue

1.upgrade or pull the power cable
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no access
```

### Anything in the Host logs that might be useful for us?

```txt
no access
```

sairon, 2024-09-23:
There's really not much to be done OS-side to be ready for unexpected power losses. Proper reboot during the upgrade should not trigger such behavior, if it leads to FS corruptions too, then there's likely something wrong with the storage you're using (you didn't specify what is the media the OS is running from).

CrAzYDr1veR, 2024-09-23:
sorry just a normal 32gb kingston sdcard

Impact123, 2024-09-23:
Seeing `Waiting for the Home Assistant CLI to be ready...` is not a bug absolutely nothing as that happens every boot.
The orange message is from the OS and the blue one from the CLI once it's started
[image]
The messages originate here:
- https://github.com/home-assistant/operating-system/blob/e4d00de47468c56b1b29797286371f0bb055efa4/buildroot-external/rootfs-overlay/usr/sbin/hassos-cli#L6
- https://github.com/home-assistant/cli/blob/b29c5f6fff06c9eb5195c49431a5c7c68de67a1e/cmd/banner.go#L25

**TLDR**: Fromw what I can tell it's normal to see the orange message.
I recommend getting an UPS if your power is unstable. SD Cards are known to corrupt easily even without brownouts or power losses.

closed 2024-12-31 by github-actions[bot]