home-assistant/operating-system #3573: USB device (Zigbee controller) not appearing on OS versions after 12.2 on Home Assistant Blue

jbytheway, 2024-09-01:
### Describe the issue you are experiencing

(Previously discussed in the comments of #3347 but I think my issue is different enough that I should open a new one)

When I first upgraded to 12.4 in June my Zigbee controller was not recognized by the kernel.  No `/dev/ttyUSB` device appeared, so Zigbee didn't work.  I fixed by downgrading to 12.2 (although, strangely, it's necessary to reboot again after downgrading to fix the issue).

This weekend I tried upgrading to 13.1 in the hope that this might have been fixed in the interim.  But the symptoms remain the same as on 12.4.

I don't see any errors in the host logs.  The first relevant log message that appears in the success case but not the failure case is

`usbcore: registered new interface driver ch341`

It seems like the failure case simply never notices the USB Serial device and registers the driver.

I tried connecting my USB device to a different port; it didn't affect the results.

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

13.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

* Standard Home Assistant Blue
* USB CC2652R1 Zigbee coordinator

### Steps to reproduce the issue

1. Boot HA OS 12.4 or 13.1
2. Observe that USB serial device is absent from `/dev`.

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing relevant.
```

### Anything in the Host logs that might be useful for us?

```txt
I've gathered host logs for three different scenarios:
* 13.1, where finding the USB device always fails (tried 4 times).
* 12.2, first boot after downgrading, where finding the USB device fails (tried twice)
* 12.2 on subsequent boots, where finding the USB device succeeds (tried at least twice)

I'll attach the full logs below, but here are some extracted highlights, after running those full logs through `sed -r 's/^([^ ]* ){3}//' | grep -iE 'usb|hub|ch341'`:

13.1 failure:

kernel: usbcore: registered new interface driver usbfs
kernel: usbcore: registered new interface driver hub
kernel: usbcore: registered new device driver usb
kernel: usbcore: registered new interface driver uas
kernel: usbcore: registered new interface driver usb-storage
kernel: usbcore: registered new interface driver usbserial_generic
kernel: usbserial: USB Serial support registered for generic
kernel: usbcore: registered new interface driver usbhid
kernel: usbhid: USB HID core driver
kernel: dwc3-meson-g12a ffe09000.usb: USB2 ports: 2
kernel: dwc3-meson-g12a ffe09000.usb: USB3 ports: 1
kernel: dwc2 ff400000.usb: supply vusb_d not found, using dummy regulator
kernel: dwc2 ff400000.usb: supply vusb_a not found, using dummy regulator
kernel: dwc2 ff400000.usb: EPs: 7, dedicated fifos, 712 entries in SPRAM
kernel: xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 1
kernel: xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 2
kernel: xhci-hcd xhci-hcd.0.auto: Host supports USB 3.0 SuperSpeed
kernel: hub 1-0:1.0: USB hub found
kernel: hub 1-0:1.0: 2 ports detected
kernel: usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
kernel: hub 2-0:1.0: USB hub found
kernel: hub 2-0:1.0: 1 port detected
kernel: dwc3-meson-g12a ffe09000.usb: switching to Device Mode
systemd[1]: Reached target Hardware activated USB gadget.
kernel: usb 1-1: new high-speed USB device number 3 using xhci-hcd
kernel: hub 1-1:1.0: USB hub found
kernel: hub 1-1:1.0: 4 ports detected
kernel: usb 2-1: new SuperSpeed USB device number 3 using xhci-hcd
kernel: hub 2-1:1.0: USB hub found
kernel: hub 2-1:1.0: 4 ports detected
```

12.2 failure:
```
kernel: usbcore: registered new interface driver usbfs
kernel: usbcore: registered new interface driver hub
kernel: usbcore: registered new device driver usb
kernel: usbcore: registered new interface driver uas
kernel: usbcore: registered new interface driver usb-storage
kernel: usbcore: registered new interface driver usbserial_generic
kernel: usbserial: USB Serial support registered for generic
kernel: usbcore: registered new interface driver usbhid
kernel: usbhid: USB HID core driver
kernel: dwc3-meson-g12a ffe09000.usb: USB2 ports: 2
kernel: dwc3-meson-g12a ffe09000.usb: USB3 ports: 1
kernel: dwc2 ff400000.usb: supply vusb_d not found, using dummy regulator
kernel: dwc2 ff400000.usb: supply vusb_a not found, using dummy regulator
kernel: dwc2 ff400000.usb: EPs: 7, dedicated fifos, 712 entries in SPRAM
kernel: xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 1
kernel: xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 2
kernel: xhci-hcd xhci-hcd.0.auto: Host supports USB 3.0 SuperSpeed
kernel: hub 1-0:1.0: USB hub found
kernel: hub 1-0:1.0: 2 ports detected
kernel: usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
kernel: hub 2-0:1.0: USB hub found
kernel: hub 2-0:1.0: 1 port detected
kernel: dwc3-meson-g12a ffe09000.usb: switching to Device Mode
systemd[1]: Reached target Hardware activated USB gadget.
kernel: usb 1-1: new high-speed USB device number 4 using xhci-hcd
kernel: hub 1-1:1.0: USB hub found
kernel: hub 1-1:1.0: 4 ports detected
kernel: usb 2-1: new SuperSpeed USB device number 4 using xhci-hcd
kernel: hub 2-1:1.0: USB hub found
kernel: hub 2-1:1.0: 4 ports detected
```

12.2 success:
```
kernel: usbcore: registered new interface driver usbfs
kernel: usbcore: registered new interface driver hub
kernel: usbcore: registered new device driver usb
kernel: usbcore: registered new interface driver uas
kernel: usbcore: registered new interface driver usb-storage
kernel: usbcore: registered new interface driver usbserial_generic
kernel: usbserial: USB Serial support registered for generic
kernel: usbcore: registered new interface driver usbhid
kernel: usbhid: USB HID core driver
kernel: dwc3-meson-g12a ffe09000.usb: USB2 ports: 2
kernel: dwc3-meson-g12a ffe09000.usb: USB3 ports: 1
kernel: dwc2 ff400000.usb: supply vusb_d not found, using dummy regulator
kernel: dwc2 ff400000.usb: supply vusb_a not found, using dummy regulator
kernel: dwc2 ff400000.usb: EPs: 7, dedicated fifos, 712 entries in SPRAM
kernel: xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 1
kernel: xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 2
kernel: xhci-hcd xhci-hcd.0.auto: Host supports USB 3.0 SuperSpeed
kernel: hub 1-0:1.0: USB hub found
kernel: hub 1-0:1.0: 2 ports detected
kernel: usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
kernel: hub 2-0:1.0: USB hub found
kernel: hub 2-0:1.0: 1 port detected
kernel: dwc3-meson-g12a ffe09000.usb: switching to Device Mode
kernel: usb 1-1: new high-speed USB device number 2 using xhci-hcd
kernel: hub 1-1:1.0: USB hub found
kernel: hub 1-1:1.0: 4 ports detected
kernel: usb 2-1: new SuperSpeed USB device number 2 using xhci-hcd
kernel: hub 2-1:1.0: USB hub found
kernel: hub 2-1:1.0: 4 ports detected
kernel: usb 1-1.1: new full-speed USB device number 3 using xhci-hcd
kernel: usbcore: registered new interface driver ch341
kernel: usbserial: USB Serial support registered for ch341-uart
kernel: ch341 1-1.1:1.0: ch341-uart converter detected
kernel: usb 1-1.1: ch341-uart converter now attached to ttyUSB0
systemd[1]: Reached target Hardware activated USB gadget.
```

There's not much difference before the USB serial messages, but here are some things I noticed:
* In the 12.2 and 13.1 failure cases, the USB hub is detected much later in the boot process than in the 12.2 success case.  So perhaps there's some kind of race condition where the attached USB devices only get detected properly when the hub is detected early enough?
* In the failure cases the USB hubs get assigned device number 3 or 4, but in the success case they get assigned device number 2.  Even though in the failure cases I don't see any other USB devices getting assigned lower numbers, so I am puzzled as to why they would be getting 3 or 4.  This may relate to the fact that the hubs get detected later.
```

### Additional information

Full host logs for the three scenarios discussed above:

[logs-13.1-failed.txt](https://github.com/user-attachments/files/16829581/logs-13.1-failed.txt)
[logs-12.2-failed.txt](https://github.com/user-attachments/files/16829579/logs-12.2-failed.txt)
[logs-12.2-success.txt](https://github.com/user-attachments/files/16829580/logs-12.2-success.txt)

Toukite, 2024-09-06:
Hello,

I have the same issue since the August upgrades:

Conbee II controller not recognized when performing OS reboot, so Zigbee add-on does not start
+
External Bluetooth USB controller not recognized when performing OS reboot, so defaulting to the default RPI one (range too low so some devices not recognized)

However after rebooting HA only they are detected again, it seems only OS reboot is causing the issue.
In the meantime, after OS reboot, I have to reboot core manually, then start manually zigbee addon, then reconfigure bluetooth integration manually to select the correct controller.

TheRoby, 2024-09-08:
Hello,

Same here, with RPI4 and HA OS 13 (or 13.1)

Today, i restart (electric) the RPI without USB. When HA is ok, i plug USB, and it work ... (Teleinfo and Zigbee)

jbytheway, 2024-09-08:
An update to my symptoms: I've since had to reboot (remaining on 12.2), and when I did so the USB was not detected again.  I had to reboot twice before it worked once more.

So perhaps the success or failure is just random, and I got more unlucky with versions later than 12.2.

I did try booting 13.1 four times before giving up.  I guess I should try more, but it's rather time consuming so I'm not enthused to.

jbytheway, 2024-09-29:
This got annoying enough that I gave up on trying to fix it and instead migrated my Home Assistant installation to a new device.

But in the process of migrating, I was able to confirm that it wasn't just this USB device that was failing to appear.  All USB devices I tried to connect (including a USB keyboard) did not work.  So this definitely feels like it was an issue at the kernel or hardware level.

Jaiuserhomeassistant, 2024-10-18:
Facing same issues with Homme assistant blue. All usb's devices are not reachable. tried to reboot. Tried to downgrade and rebooted many times but no access anymore. Very annoying.

therealcmj, 2024-10-20:
I ran into this issue as well and it turns out that there might be a known hardware issue with the Blue device.

Check these links:

- [the thread I started on HA community](https://community.home-assistant.io/t/did-husbzb-1-stop-working-everyone-else-recently/777531/8)
- [Previous mention of issues with the ODroid device](https://community.home-assistant.io/t/all-usb-ports-died-on-homeassistant-blue-odroid-n2/430594/16)

lsusb shows the in built hub, just not the USB stick.

I'm going to go buy a USB hub and see if the the USB ports are dead or if they just can't push any power.

therealcmj, 2024-10-22:
I wound up finding a "USB on the go" hub with power supply in a drawer from another unrelated project with a Pi Zero. Moving my USB stick to that and plugging it into the MicroUSB other side of the HA Blue device got me back up and running. 

Which strongly suggests we are victim to the bum hardware in the ODroid. Which is disappointing but at least explains the issue.

closed 2025-03-31 by github-actions[bot]