home-assistant/operating-system #3578: Flooded by error ➜ eth0: not whole frame pointed by descriptor

shamasis, 2024-09-06:
### Describe the issue you are experiencing

### The problem

Since I am on `2024.8.x`, I am seeing my ha CLI screen being flooded with error message that looks like the following:
```
macb 1f00100000.ethernet eth0: not whole frame pointed by descriptor
```

I would love to know if this needs attention. I am not seeing any side effect in regular usage so far.

My online exploration tells me that this is connected to the config of the network adapter and if I increase the rx buffer size this should be okay. I wanted to check here before I go down the process of enabling SSH and doing these changes, which may or may not persist through updates.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.6.31-haos-raspi

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Running on Pi5 with wired ethernet to my TP-Link ER605 router.

### Steps to reproduce the issue

-

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### System information

## System Information

version | core-2024.8.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.31-haos-raspi
arch | aarch64
timezone | Asia/Calcutta
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | September 26, 2024 at 5:30 AM
relayer_connected | true
relayer_region | ap-southeast-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | ap-southeast-1-2.ui.nabu.casa
certificate_status | ready
instance_id | 8b1ce8c77a6e4f01a24fbf0a36ba482a
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 228.5 GB
disk_used | 8.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (6.4.2), Terminal & SSH (9.14.0), AdGuard Home (5.1.1), UniFi Network Application (3.2.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 0
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 29, 2024 at 3:13 PM
-- | --
current_recorder_run | September 3, 2024 at 1:47 PM
estimated_db_size | 145.27 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Additional information

I am running only AdGuard as an additional addon that serves DNS.

sairon, 2024-09-23:
First time I see this to be honest. Is this something that happens also when you stop all the add-ons and when HAOS is mostly idling? Isn't there something causing excessive traffic in your network?

Minims, 2024-10-06:
Same issue on PI 5 with HA supervised with last version.

closed 2025-01-12 by github-actions[bot]