home-assistant/operating-system #3579: Add support for Z83 mini-pc to HAOS

jonsenior79, 2024-09-08:
### Describe the issue you are experiencing

The Bluetooth and wifi adapters on the Z83 do not work under HAOS. To get them working under debian, a few non-free firmwares needed to be installed. Looking at the contents of /lib/firmware the broadcom files appear to be there, but there are some missing softlinks.

The following exert from dmesg shows the errors for the sst_acpi firmware and the brcmfmac firmware. Firmware brcmfmac43455-sdio is already on the system but there's no matching softlink. I've not checked for the sst_acpi one as I can't easily test by adding files to a live system.

<pre><font color="#4E9A06">[    4.222663] </font><font color="#C4A000">intel_sst_acpi 808622A8:00: </font><b>Direct firmware load for intel/fw_sst_22a8.bin failed with error -2</b>
<font color="#4E9A06">[    4.222688] </font><font color="#C4A000">intel_sst_acpi 808622A8:00: </font><font color="#CC0000">request fw failed</font>
<font color="#4E9A06">[    4.314048] </font><font color="#C4A000">cfg80211: </font>Loading compiled-in X.509 certificates for regulatory database
<font color="#4E9A06">[    4.340428] </font><font color="#C4A000">Loaded X.509 cert &apos;sforshee: </font>00b28ddf47aef9cea7&apos;
<font color="#4E9A06">[    4.355127] </font><font color="#C4A000">Loaded X.509 cert &apos;wens: </font>61c038651aabdcf94bd0ac7ff06c7248db18c600&apos;
<font color="#4E9A06">[    4.916177] </font><font color="#C4A000">brcmfmac: </font>brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
<font color="#4E9A06">[    4.916854] </font><font color="#C4A000">brcmfmac mmc0:0001:1: </font><b>Direct firmware load for brcm/brcmfmac43455-sdio.AZW-Z83-V.bin failed with error -2</b>
<font color="#4E9A06">[    4.929758] </font><font color="#C4A000">usbcore: </font>registered new interface driver brcmfmac
<font color="#4E9A06">[    4.955938] </font><font color="#C4A000">brcmfmac mmc0:0001:1: </font><b>Direct firmware load for brcm/brcmfmac43455-sdio.txt failed with error -2</b>
<font color="#4E9A06">[    5.994448] </font><font color="#C4A000">brcmfmac: </font><font color="#CC0000">brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50</font>
</pre>

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.46-haos

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Z83-V (I believe it's a chinese-made Intel NUC clone). Intel Atom 4-core processor. The wifi+BT chip is an AP6234.

### Steps to reproduce the issue

1. Install HAOS onto device!
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-09-08 17:02:19.134 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2024-09-08 17:02:19.136 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-09-08 17:02:19.136 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2024-09-08 17:02:19.626 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-09-08 17:02:19.628 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-09-08 17:02:19.628 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-09-08 17:31:44.031 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-09-08 18:01:45.203 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-09-08 18:01:58.534 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
2024-09-08 18:02:19.636 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2024-09-08 18:02:19.636 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2024-09-08 18:02:19.637 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2024-09-08 18:02:19.638 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2024-09-08 18:02:19.640 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2024-09-08 18:02:19.643 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-09-08 18:02:19.643 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2024-09-08 18:02:19.645 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2024-09-08 18:02:19.678 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2024-09-08 18:02:19.679 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2024-09-08 18:02:19.680 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
2024-09-08 18:02:19.681 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2024-09-08 18:02:19.682 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-09-08 18:02:19.683 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2024-09-08 18:02:20.133 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-09-08 18:02:20.135 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-09-08 18:02:20.136 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
[... 145 lines cut ...]
version_api | ok
installed_addons | Terminal & SSH (9.14.0), ESPHome (2024.8.3), File editor (5.8.0), Let's Encrypt (5.1.4), Duck DNS (1.18.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 1
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 6 September 2024 at 20:28
-- | --
current_recorder_run | 7 September 2024 at 03:43
estimated_db_size | 19.79 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

jonsenior79, 2024-09-14:
I'd like to help with producing a PR for this, but I'm not quite sure how to go about testing. I understand that few people probably have these boxes, but it's a little silly to buy a BT adapter when they not only have integrated BT, but it is known to work under a Debian / Supervised HA installation.

sairon, 2024-09-16:
I'm afraid I'll disappoint you - with firmware, we are quite conservative, because the space reserved for the rootfs partition is limited, especially on generic-x86-64, so we can't afford adding all firmware files that are available from packages on other distributions. Since the device you want to support is rather an uncommon one, and it's even [less performant than an RPi 3](https://www.cpubenchmark.net/compare/2774vs4671/Intel-Atom-x5-Z8350-vs-BCM2835), it's not very convincing this is the case where we should prioritize adding the files needed for it. 

I think you've picked the wrong lines from the boot log - the brcmfmac errors are shown because there's no specific firmware files for your board, but in the end I think the card should initialize - but this is just the WiFi driver, not BT. You can compare this to another distribution where more firmware files are available. For the Bluetooth, there's a way to add the files using the method described in #2222. Same should be applicable to the audio firmware, which is also rather an exotic one. Otherwise, I'm sorry but closing this request as a wontfix for the reasons written above.