home-assistant/operating-system #3588: HA o/s breaks network's IPv6 connectivity whenever home-router reboots / HAOS suddenly starts impersonating router static IPv6 address when router does DAD during boots up

aleks-mariusz, 2024-09-16:
### Describe the issue you are experiencing

I've just started playing with HomeAssistant, and so to dip my toes, I created a VM back in July on my NAS (which runs Linux, using libvirtd/kvm - While I haven't done anything with HASS (other than set up a TLS cert, install HACS and the ssh-addon), for the most part I'm pretty new to HA but quite comfortable w/ linux/containers/networking (it's my day job).

I have on my home-network dual-stack (IPv4 + IPv6), and I noticed that my IPv6 connectivity starts to act "funny" ever since around the time i set up HA, which seems was uncoincidental..  I have been having issues for months on my home network related to IPv6 connectivity breaking seemingly "randomly" (i only now just figured out what triggers it). 

So of course I logged into my home-router (running OpenWRT) and I noticed the following log messages on my home-router:

```
Mon Sep 16 16:02:50 2024 kern.info kernel: [   66.633040] IPv6: br-lan.2220: IPv6 duplicate address 2a02:aaaa:bbbb:2220:: used by 52:54:00:c6:a5:c6 detected!
Mon Sep 16 16:02:51 2024 daemon.warn odhcpd[2615]: No default route present, overriding ra_lifetime to 0!
Mon Sep 16 16:02:56 2024 daemon.warn odhcpd[2615]: No default route present, overriding ra_lifetime to 0!
```

What seems to be happening is that the router fails doing its attempt at duplicate-address detection because the mac-address of my HomeAssistant VM is advertising itself as the address the router is trying to use.   I've of course double-and-triple-checked that the home-assistant VM is not configured to use that address.

Seems specifically only-triggered whenever i reboot my home-router (so it's something i do a few times a month to keep up to date with patches).. The router boots back up, it suddenly is unable to assign itself the configured IPv6 address, because for some reason, when doing DAD and it asks if anyone has the address, the home-assistant VM starts responding to the router's neighbor-solicitation requests (this use of neighbor solicitation is standard for duplicate-address-detection for any IPv6 host before it assigns itself an address, it asks if anyone else is using it first).

Now i've made sure the home-assistant has been allocated an entirely different static dhcpv6 assignment, and so it's unclear WHY it starts behaving this way once the router goes down.  But because my home-router (when it boots back up) can no longer assign itself the IPv6 address to the VLAN interface for the network HA is on, all my hosts on the vlan no longer have working IPv6 due to DHCPv6 stopping working

The only fix is to reboot the HA vm guest.

What i'm aiming to find out is, why is HAOS doing this?

Below, I captured a tcpdump during the incident over the weekend, just of all icmp6 packets (the dhcpv6 traffic is not captured) - I am posting two versions, one annotated by myself (with the commentary of what i believe is happening), as well as an unannotated/unabridged version (if you don't want my comments/interpretation of the traffic to bias your investigation).

**Note** i am using a 'diff-like output' to colorize the different "speakers" of each ICMPv6 packet to make to easier to follow:

The hosts involved and appearing in the below captures are color-coded as follows (along with their IPv6 addresses, both link-local and global-unicast):

```diff
+  router-main:bd:10 - provides IPv6 connectivity normally, via SLAAC-enabling router-advertisements proposing the client uses DHCPv6 
+     2a02:aaaa:bbbb:2220:: - my statically assigned address for this VLAN's interface - what the HA vm starts impersonating
+     fe80::c0bf:3704:2f17:8093 - link-local stable-privacy address prior-to--reboot
+     fe80::6c0d:8563:da46:ca9c - link-local stable-privacy address after-reboot
+     fe80::213f:816a:a47b:30e4 - link-local stable-privacy address pre-issue-fixed
+     fe80::33b0:f83f:9c09:f500 - link-local stable-privacy address pre-issue-fixed
+     fe80::9bec:db29:40d6:3f66 - link-local stable-privacy address after-issue-fixed
 
-  server-kvm_:e1:b8 - this is my NAS which is the hypervisor for the home-assistant VM
-      fe80::d250:99ff:fe6f:e1b8 - link-local eui64-based address 
-      2a02:aaaa:bbbb:2220:d250:99ff:fe6f:e1b8 - slaac-assigned eui64-based address

!  vm-homeastn:a5:c6 - this is the actual VM running home-assistant OS
!      fe80::a8b4:7a51:21a4:a9ba - link-local eui64-based address
!      2a02:aaaa:bbbb:2220::11 - dhcpv6-assigned global addresss
!      2a02:aaaa:bbbb:2220:6bf3:b959:29ea:d161 - slaac-assigned address

@@ macbook-pro:0c:20 - this is my macbook which is also impacted by this issue ---@@
@@      fe80::1430:f31f:50e3:e9d0 - link local ---@@
@@      2a02:aaaa:bbbb:2220::1  - DHCPv6 assigned address ---@@
@@      2a02:aaaa:bbbb:2220:7c31:b848:97e:7840 - SLAAC-assigned privacy address  ---@@

# and finally this contains comments w/ my understanding of about what's happening on the following line
```

#### key times in the capture

pre-14:00 - everything working normally
14:00:00.000 - the router has been rebooted
14:01:08.391 - the HA vm begins impersonating the router's assigned IPv6 gua address (answering to another host on network, a macbook)
14:01:13.500 - the router has sent its first ICMPv6 message since booting back up
14:01:52.257 - final HA-vm's impersonation of router, at this point it is rebooted. 
14:25:04.935 - the router performs it's DAD again, successfully - things from then on continue to work normally.
14:25:06.154 - the HA vm starts coming back up 
14:25:26.377 - the HA vm appears to try to impersonate again, but this may have been after the DAD timeout expired

*tcpdump captures in the next comment, otherwise i get a max-content error > 65536 characters long)*

### What operating system image do you use?
[... 382 lines cut ...]
Disk total      30.8 GB
Disk used       8.6 GB
Healthy true
Supported       true
host_connectivity       true
supervisor_connectivity true
ntp_synchronized        true
virtualization  kvm
Board   ova
Supervisor API  ok
Version API     ok
Installed add-ons       Advanced SSH & Web Terminal (18.0.0), Studio Code Server (5.15.0), Cloudflared (5.1.19), Frigate (0.14.0), Mosquitto broker (6.4.1)
### Dashboards
Dashboards      2
Resources       0
Views   0
Mode    storage
### Recorder
Oldest run start time   September 12, 2024 at 1:10 PM
Current run start time  September 14, 2024 at 2:23 PM
Estimated database size (MiB)   2.01 MiB
Database engine sqlite
Database version        3.45.3

</details>

aleks-mariusz, 2024-09-16:
<details><summary>unabridged raw tcpdump capture</summary>

```diff
+    13:49:26.035 router-main:bd:10 > 33:33:00:00:00:01,                fe80::c0bf:3704:2f17:8093 > ff02::1                                    ICMP6, router advertisement, length 160
+         hop limit 64, Flags [managed, other stateful], pref medium, router lifetime 1800s, reachable time 0ms, retrans timer 0ms
+           source link-address option (1), length 8 (1): router-main:bd:10
+           mtu option (5), length 8 (1):  1492
+           prefix info option (3), length 32 (4): 2a02:aaaa:bbbb:2220::/64, Flags [onlink, auto], valid time 75758s, pref. time 75758s
+           route info option (24), length 24 (3):  2a02:aaaa:bbbb::/48, pref=medium, lifetime=1800s
+           rdnss option (25), length 24 (3):  lifetime 1800s, addr: 2a02:aaaa:bbbb:2220::
+           dnssl option (31), length 40 (5):  lifetime 1800s, domain(s): lan.home.domain.com.
+           advertisement interval option (7), length 8 (1):  600000ms
+    13:57:30.667 router-main:bd:10 > 33:33:00:00:00:01,                fe80::c0bf:3704:2f17:8093 > ff02::1                                    ICMP6, router advertisement, length 160
+         hop limit 64, Flags [managed, other stateful], pref medium, router lifetime 1800s, reachable time 0ms, retrans timer 0ms
+           source link-address option (1), length 8 (1): router-main:bd:10
+           mtu option (5), length 8 (1):  1492
+           prefix info option (3), length 32 (4): 2a02:aaaa:bbbb:2220::/64, Flags [onlink, auto], valid time 75273s, pref. time 75273s
+           route info option (24), length 24 (3):  2a02:aaaa:bbbb::/48, pref=medium, lifetime=1800s
+           rdnss option (25), length 24 (3):  lifetime 1800s, addr: 2a02:aaaa:bbbb:2220::
+           dnssl option (31), length 40 (5):  lifetime 1800s, domain(s): lan.home.domain.com.
+           advertisement interval option (7), length 8 (1):  600000ms
+    13:57:35.296 router-main:bd:10 > vm-homeastn:a5:c6,                fe80::c0bf:3704:2f17:8093 > 2a02:aaaa:bbbb:2220::11                    ICMP6, neighbor solicitation, length 32, who has 2a02:aaaa:bbbb:2220::11
!    13:57:35.297 vm-homeastn:a5:c6 > router-main:bd:10,                  2a02:aaaa:bbbb:2220::11 > fe80::c0bf:3704:2f17:8093                  ICMP6, neighbor advertisement, length 24, tgt is 2a02:aaaa:bbbb:2220::11, Flags [router, solicited]
!    13:57:35.361 vm-homeastn:a5:c6 > router-main:bd:10,                fe80::a8b4:7a51:21a4:a9ba > fe80::c0bf:3704:2f17:8093                  ICMP6, neighbor solicitation, length 32, who has fe80::c0bf:3704:2f17:8093
+    13:57:35.361 router-main:bd:10 > vm-homeastn:a5:c6,                fe80::c0bf:3704:2f17:8093 > fe80::a8b4:7a51:21a4:a9ba                  ICMP6, neighbor advertisement, length 24, tgt is fe80::c0bf:3704:2f17:8093, Flags [router, solicited]
+    13:57:40.416 router-main:bd:10 > vm-homeastn:a5:c6,                fe80::c0bf:3704:2f17:8093 > fe80::a8b4:7a51:21a4:a9ba                  ICMP6, neighbor solicitation, length 32, who has fe80::a8b4:7a51:21a4:a9ba
!    13:57:40.417 vm-homeastn:a5:c6 > router-main:bd:10,                fe80::a8b4:7a51:21a4:a9ba > fe80::c0bf:3704:2f17:8093                  ICMP6, neighbor advertisement, length 24, tgt is fe80::a8b4:7a51:21a4:a9ba, Flags [router, solicited]
+    13:58:53.121 router-main:bd:10 > server-kvm_:e1:b8,                fe80::c0bf:3704:2f17:8093 > fe80::d250:99ff:fe6f:e1b8                  ICMP6, router advertisement, length 160
+         hop limit 64, Flags [managed, other stateful], pref medium, router lifetime 1800s, reachable time 0ms, retrans timer 0ms
+           source link-address option (1), length 8 (1): router-main:bd:10
+           mtu option (5), length 8 (1):  1492
+           prefix info option (3), length 32 (4): 2a02:aaaa:bbbb:2220::/64, Flags [onlink, auto], valid time 75191s, pref. time 75191s
+           route info option (24), length 24 (3):  2a02:aaaa:bbbb::/48, pref=medium, lifetime=1800s
+           rdnss option (25), length 24 (3):  lifetime 1800s, addr: 2a02:aaaa:bbbb:2220::
+           dnssl option (31), length 40 (5):  lifetime 1800s, domain(s): lan.home.domain.com.
+           advertisement interval option (7), length 8 (1):  600000ms
+    13:58:58.176 router-main:bd:10 > server-kvm_:e1:b8,                fe80::c0bf:3704:2f17:8093 > fe80::d250:99ff:fe6f:e1b8                  ICMP6, neighbor solicitation, length 32, who has fe80::d250:99ff:fe6f:e1b8
-    13:58:58.176 server-kvm_:e1:b8 > router-main:bd:10,                fe80::d250:99ff:fe6f:e1b8 > fe80::c0bf:3704:2f17:8093                  ICMP6, neighbor advertisement, length 24, tgt is fe80::d250:99ff:fe6f:e1b8, Flags [solicited]
-    13:59:03.360 server-kvm_:e1:b8 > router-main:bd:10,                fe80::d250:99ff:fe6f:e1b8 > fe80::c0bf:3704:2f17:8093                  ICMP6, neighbor solicitation, length 32, who has fe80::c0bf:3704:2f17:8093
+    13:59:03.360 router-main:bd:10 > server-kvm_:e1:b8,                fe80::c0bf:3704:2f17:8093 > fe80::d250:99ff:fe6f:e1b8                  ICMP6, neighbor advertisement, length 24, tgt is fe80::c0bf:3704:2f17:8093, Flags [router, solicited]
@@   14:00:54.816 macbook-pro:0c:20 > router-main:bd:10,                fe80::1430:f31f:50e3:e9d0 > fe80::c0bf:3704:2f17:8093                  ICMP6, neighbor solicitation, length 32, who has fe80::c0bf:3704:2f17:8093 ---@@
[... cut ...]
+    13:57:30.667 router-main:bd:10 > 33:33:00:00:00:01,                fe80::c0bf:3704:2f17:8093 > ff02::1                                    ICMP6, router advertisement, length 160
# router looking for HA vm GUA IPv6 addr (still normal)
+    13:57:35.296 router-main:bd:10 > vm-homeastn:a5:c6,                fe80::c0bf:3704:2f17:8093 > 2a02:aaaa:bbbb:2220::11                    ICMP6, neighbor solicitation, length 32, who has 2a02:aaaa:bbbb:2220::11
!    13:57:35.297 vm-homeastn:a5:c6 > router-main:bd:10,                  2a02:aaaa:bbbb:2220::11 > fe80::c0bf:37

agners, 2024-09-17:
> Ok i feel a bit silly but, i've now tested this with a plain Ubuntu 22.04 VM also on the same KVM bridge acting the same way.. so it may not be HA-related afterall :-(

I was going to say, when it comes to networking, we pretty much run vanilla Linux. Our kernel might have a slight different config, but I would not expect that this changes core IPv6 address assignment behavior. Ontop of that we use NetworkManager 1.44.2.

I've recently noticed that Docker actually blocks multicast forwarding on all bridges on the system. Depending on when the Docker daemon gets started, this can lead to IPv6 working at first, but stops working after a while (see https://github.com/moby/moby/issues/48365). But if you don't use Docker, this is unlikely related.

Maybe you accidentally caused a Layer 2 level loop or something :thinking: ? Make sure to enable STP etc.

Maybe you also want to try disabling multicast snooping. There are some reports of such issues in the Linux bug tracker (see https://bugzilla.kernel.org/show_bug.cgi?id=99081#c14).

aleks-mariusz, 2024-09-17:
> But if you don't use Docker, this is unlikely related

wait, I was under the impression that Docker is always used on HA-os as the supervisor ends up starting all the components (including homeassistant-core) in separate containers.

> Make sure to enable STP etc

It's enabled by default on the bridge automatically

> some reports of such issues in the Linux bug tracker

Thanks, I've looked at these already but their specific symptoms are not exactly like mine, where a random VM start impersonating another IPv6 identity on the network.  I can only get this to happen with the router (e.g. rebooting one or the other VMs doesn't make it impersonate the other).  Also thanks for telling me what entity is controlling the network.  The Ubuntu VM i also saw doing this strangeness is using netplan and not NetworkManager.  I still suspect there's something funky going on at the bridge-level unless it's a linux-kernel level issue (at this point, i'm just grasping at straws trying to find commonality).

Anyway thanks for entertaining my strange issue, even though it's very unlikely to be directly related to this project.  I'll post on reddit or something under the KVM section, see if anyone has an idea there as well.

aleks-mariusz, 2024-09-18:
Ok so I have come to realize the underlying cause of for this weirdness.  _I appreciate this is not an HA related issue at all, but it was quite an interesting lesson in IPv6 networking, so i figure i'll share my findings, and express my sincere thanks to you folks (esp @agners) for letting me use this issues-tracker as forum for venting/sharing the tcpdump at least.  It's been an excellent learning experience:_

Cause: It's because i chose an IPv6 address that ends in just  '::' for the router's static assignment in the first place :man_facepalming: 

Solution: I changed my [ip6ifaceid](https://openwrt.org/docs/guide-user/network/ipv6/configuration#:~:text=interface%20(CIDR%20notation)-,ip6ifaceid,-ipv6%20suffix) (which is the "ipv6-address interface-id" - the part that generates the suffix of the IPv6 address) from the neat-looking '::' to anything else (i chose '::ffff') and the issue stopped.  I had only chose '::' originally because it looked neat, without realizing the "ticking-timebomb" that i had introduced once i had VMs up on the network (and probably it could happen w/ any linux host really, VM or not).

Seems this `[2xxx:aaaa:bbbb::]` i ended up with, is a possibly a "reserved" anycast address, and for whatever reason, both my VMs have regardless (even after the fix) in their routing table this address:

```
user@vmname:~$ ip -6 route show table local | grep anycast | grep 2a02
anycast 2a02:aaaa:bbbb:2220:: dev ens2 proto kernel metric 0 pref medium
```

...at least until my router goes down, when it stays in only one VM's who started impostering the router's address.

I have no idea what mechanism triggers this automated behaviour only once the VM goes down, so if anyone has any idea, i'd be game to understand the underlying behavioural-cause, but at least I have a fix for now.

closed 2024-09-18 by aleks-mariusz

aleks-mariusz, 2024-09-18:
Upon further research, this behaviour might be triggered by HAOS seemingly being configured "out-of-the-box" as a router (`sysctl` shows `net.ipv6.conf.all.forwarding=1`).  I don't remember changing this, is this a default setting?  I think this is what causes this VM to behave in this unexpected way - My other culprit VM i also have configured this way, albeit intentionally.  

@agners any idea why is HAOS set up this way?

agners, 2024-09-18:
> @agners any idea if/why is HAOS set up this way?

Yes, HAOS enables IPv6 forwarding by default. The reason is that it makes it possible to turn Home Assistant OS into a Thread Border router. See also the relevant PR #1832.

FWIW, I think recent Docker version enable IPv6 forwarding as well.

Do you know why/how that influences the address setup exactly? :thinking: 

A while ago I also realized that enabling IPv6 routing affected the IPv6 router readability probe (see #2434). We currently carry a patch which reverts that behavior.

agners, 2024-09-18:
> Cause: It's because i chose for the router's static assignment an IPv6 address that ends in just '::' in the first place, which is [a reserved anycast address](https://www.rfc-editor.org/rfc/rfc4291#section-2.6.1) and seems bad things can happen when you use it for unicast purposes 🤦‍♂️

So that is OpenWrt right? So you used `::` as IPv6 suffix? Given that is a reserved anycast address, maybe OpenWrt should actually warn/disallow that setting? :thinking: :sweat_smile:

aleks-mariusz, 2024-09-18:
> HAOS enables IPv6 forwarding by default. The reason is that it makes it possible to turn Home Assistant OS into a Thread Border router

interesting - i would have personally preferred this was only enabled if you actually need to use thread though? I get that thread relies on IPv6 - but i wager it's _not_ the majority of HA users that actually have thread in their environment that it's configured as on-by-default out of the box, just my two cents tho? I can at least see where this came from though.

> FWIW, I think recent Docker version enable IPv6 forwarding as well.

Truth be told, I've stopped using docker entirely, in favor of [podman](https://betterstack.com/community/guides/scaling-docker/podman-vs-docker/#:~:text=eliminating%20the%20need%20for%20a%20continuous%20daemon%20process%20in%20the%20background) - i don't see a reason to rely on a third-party package with its own daemon when all its doing is relying on built-in linux kernel features (namespaces/cgroups).. I guess podman itself is an extra package to be installed, but at least it doesn't need an extra daemon running as root.  Whether the same IPv6 issue exists with podman, i'm really not sure as i haven't played with IPv6 inside containers - i do know that podman requires a somewhat recent kernel (it mostly worked on el7's 3.10 but only for IPv4 - something newer would be needed for IPv6 i reckon), FWIW.

> ...maybe OpenWrt should actually warn/disallow that setting

Yes, the victim was an OpenWRT-based router - the UI actually has a default of ::1 - i overrode it thinking i was clever, but i guess it should probably warn the user, that's a great idea for a PR !  I have already [updated the openwrt-docs](https://openwrt.org/docs/guide-user/network/ipv6/configuration#:~:text=It%20is%20advised%20to%20NOT%20use%20just%20%27%3A%3A%27%20as%20this%20is%20a) at least about these findings... I'll ask if there is an appetite for a PR to put better guard-rails on the UI at least.