home-assistant/operating-system #3599: Restarting HA from webpage causes VMbox to crash

MichaelMTB, 2024-09-27:
### Describe the issue you are experiencing

I had Home Assistant working OK on my laptop as a test to see if I would get into using it. All went well so I recently bought a new mini pc which I installed VMbox on Windows 11 Pro. Installed Home assistant on the VM and restored from a backup of the previous version of HA.
All seems to be working well on the new mini pc, but when I restart HA from the web page (due to making changes that require a restart) by Setting -> 3 dots in upper right corner -> Restart Home Assistant, the web page does not give the messages to show restarting, loading service etc (in bottome left corner of web UI) and VMbox freezes and becomes unresponsive and requires to be shutdown and restarted.
Is there anything I may have missed or done incorrectly when setting up the new mini pc with VMbox and restoring my previous backup of HA.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.46-haos

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

## System Information

version | core-2024.8.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.46-haos
arch | x86_64
timezone | Australia/Sydney
config_dir | /config

USB Dongle attached -> ZBT1 Zigbee Coordinator

### Steps to reproduce the issue

1. Restart HA from webpage UI (due to making changes that require a restart) by:
2. Left sidebar menu -> Setting -> 3 dots in upper right corner -> Restart Home Assistant
3. the web page does not give the messages to show restarting, loading service etc (in bottome left corner of web UI)
4. VMbox freezes and becomes unresponsive and requires to be shutdown and restarted.

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-09-27 18:31:38.927 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-cli with version 2024.09.0
2024-09-27 18:31:38.948 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
2024-09-27 18:31:38.952 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning hassio_cli application
2024-09-27 18:31:39.204 INFO (MainThread) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/amd64-hassio-cli with version 2024.09.0 - 172.30.32.5
2024-09-27 18:31:39.213 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-dns with version 2024.04.0
2024-09-27 18:31:39.229 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
2024-09-27 18:31:39.233 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning hassio_dns application
2024-09-27 18:31:39.521 INFO (MainThread) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/amd64-hassio-dns with version 2024.04.0 - 172.30.32.3
2024-09-27 18:31:39.522 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
2024-09-27 18:31:39.526 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0
2024-09-27 18:31:39.535 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
2024-09-27 18:31:39.542 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning hassio_audio application
2024-09-27 18:31:39.841 INFO (MainThread) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0 - 172.30.32.4
2024-09-27 18:31:39.843 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-observer with version 2023.06.0
2024-09-27 18:31:39.852 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-multicast with version 2024.03.0
2024-09-27 18:31:39.859 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
2024-09-27 18:31:39.862 INFO (SyncWorker_2) [supervisor.docker.manager] Cleaning hassio_multicast application
2024-09-27 18:31:40.072 INFO (MainThread) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/amd64-hassio-multicast with version 2024.03.0 - Host
2024-09-27 18:31:40.076 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
2024-09-27 18:31:40.078 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/qemux86-64-homeassistant with version 2024.8.2
2024-09-27 18:31:40.108 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 13.1 / BootSlot A
[... 262 lines cut ...]

</details>

<details><summary>Miele</summary>

component_version | 2024.8.1
-- | --
reach_miele_cloud | ok

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 18, 2024 at 5:24 AM
-- | --
current_recorder_run | September 27, 2024 at 6:32 PM
estimated_db_size | 19.31 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Additional information

Nope

sairon, 2024-10-09:
Not sure what you mean by "VMbox" (is that supposed to be VirtualBox or VMware?) but this looks rather like a problem in the hypervisor software. You can check the console of the VM when you click the restart and see where it hangs. If it reaches the point where the kernel shuts down, there's little the OS can do.

MichaelMTB, 2024-10-12:
Yes thats VirtualBox. Just did a reboot from HA while looking at the VitualBox window with HA running inside it for clues. There was no commands at all (unlike when it starts up it hhas lots of commands running). The cursor was blinking at the HA > prompt during the restart of HA but then it stopped blinking at remained on. That is when I need to force shutdown VitualBox and restart it again to get HA running again.
I am going to revert to the earlier version of VirtualBox (that I was running on my laptop before buying the NUC Box which I am currently using) as that was restarting HA OK.

MichaelMTB, 2024-10-13:
OK reverted to earlier version of VirtualBox. 7.0.20. When restarting HA it does the same thing again. VirtualBox freezes after about 20 seconds (cursor stops blinking). No messages inside VirtualBox. HA wont reconnect again on webpage. Need to shutdfown VirtualBox by task manager and then restart again. HA then reconnects OK once VirtualBox is up and running.
Very frustrating. If my memory serves me correct, on my old setup (laptop) I could restart HA from webapge OK and would not require any interaction with VirtualBox. I am at a loss on what to do next....

orgebar, 2024-10-17:
Im not 100% sure but it sounds as if I had a similar problem, where HA failed to reboot every time and the reason was a broken database. I guess The database broke when a HA updated failed unknowingly or I maybe accidetially Unplugged the PC. Just like you, i could not reboot and I did not recognize the broken db for quite a while too - There were only a few cryptic error messages on some data/database errors in the logs that I could not connect to a broken database directly. I did a repair/rebuild of the database and then everything worked fine again.

If that's your Problem too, 
1. a repair/rebuild of home-assistant_v2.db from inside the config folder will help. 
2.Maybe youll find there additional database backup files which were automatically created and are named sth. like "home-assistant_v2.db.xxXXxx_broken" or "home-assistant_v2.db.xxXXxx_fixed". The mentioned file(s) are a backup of the old database 

--> For 1. You can restore/repair/rebuild

--> For 2. You can restore/repair/rebuild the new automatically created db "home-assistant_v2.db" or the older version database "..._failed/broken/" the same way as in 1. 

Please keep in Mind that after a restore/repair/rebuild of the "xxxx_failed"-db, the data from "home-assistant_v2.db" are lost, as you cannot merge the two dbs. In other words: You gave to decide which db you want to restore and you cannot add the data from the new db (which contain infos since the original one broke) to the repaired broken one. At least I didnt get it done.

Try ChatGPT or take a look in the documentations to get the steps necessary for either of both cases mentioned above. There are only a few steps to do but the rebuild can take quite a while depending on its size. Be aware that The console in which you started the rebuild must be open/active during the whole rebuild - if its closed the rebuild will cancel. To avoid this danger, try starting the rebuild as a passive/background process.

MichaelMTB, 2024-10-22:
Thanks for the direction orgebar. Did a bit of a search on this and looked at /homeassistant/home-assistant.log Found the following entry...
2024-10-16 17:45:14.433 WARNING (Recorder) [homeassistant.components.recorder.util] The system could not validate that the sqlite3 database at //config/home-assistant_v2.db was shutdown cleanly
Also there is only one database file so hopefuly just a repair might help.
If I am feeling adventerous one day I will try to repair my database to see if that helps.

closed 2025-01-29 by github-actions[bot]