home-assistant/operating-system #3602: "NetworkManager not correctly configured" + unsupported state warning

bcutter, 2024-09-28:
### Describe the issue you are experiencing

Today while trying to do an addon upgrade I discovered an issue I've seen (and fixed by restarting HA OS/the host) before:

https://github.com/home-assistant/supervisor/issues/5249

Addon upgrade didn't do anything. According to supervisor logs, nothing happened.
I restarted the supervisor and was able to perform the addon updates afterwards, but at the same time a problem was raised:

[image]

[image]

(same for the observer)
[image]

...which redirects to https://www.home-assistant.io/more-info/unsupported/network_manager

At the same time, `/config/network` does not properly (or at all) show network device information anymore:

[image]

Meanwhile I'm very sure this is a HA OS issue. As I don't want to run into this on a monthly basis (first occured at 2024-08-14, now today at 2024-09-28), I created this issue with some host logs attached.

I believe this "always" happens after running HA OS for a longer time:
- 2024-08-14: host up and running for ~ 24 days, RAM usage ~ 40 %, SWAP usage ~ 36 %
- 2024-09-28: host up and running for ~ 34 days, RAM usage ~ 40 %, SWAP usage ~ 23 %

On 2024-08-14 at some point I had to restart the host as the whole system became heavily unusable (unresponsive, CPU at 100 %, load-1 far beyond > 15, everything crazy).

**Hopefully someone will pick this up quickly _(few hours/half a day at best)_ before I need to apply the workaround (I fear the host might run into similiar instability issues again) and reboot the host to regain a supported, fully working system. Currently I can very likely dig for logs and test stuff, after the reboot usually everything's gone (`ha host logs -b-1 -n1000` gives no logs).**

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

4 usb devices attached
- UPS
- ZigBee stick (with extension cable)
- DuoFern stick (with extension cable)
- external NVMe disk (only attached, not used at all yet)

### Steps to reproduce the issue

1. Use HA OS (and HA) for a longer time like 3 to 5 weeks without any restart/reboot
2. See network based issues like supervisor unable to fetch and perform updates etc.

### Anything in the Supervisor logs that might be useful for us?

```txt
No, nothing suspicious at all
```

### Anything in the Host logs that might be useful for us?

```txt
Yes, very interesting (captured with `ha host logs -n1000` as the UI limits the output to a quite short period of time). To what I see, everything before "2024-09-28 05:29:*" is quite normal. Then some kind of "thing" (incident) happened and network related stuff broke, please interpret this on your own:

---------------------------------------------------------------------------------------------------------

(edit: see next comment(s) as GitHub noted "Comment is too long (maximum is 65536 characters)")
```

bcutter, 2024-09-28:
**Host logs (too big for the initial post):**

```
2024-09-28 03:29:36.093 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.Mxf1CW.mount: Deactivated successfully.
2024-09-28 03:30:03.130 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-7bbf8ce9e8d5040c90e20186dc12e688436b3127096898f9d4596bb4633f2a3d-runc.GNEbqz.mount: Deactivated successfully.
2024-09-28 03:31:53.499 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.aljVJC.mount: Deactivated successfully.
2024-09-28 03:32:08.751 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-4cd355a807f5defafc65677c6f7da301b48e1d65c8adca9e8194bc4fcdf6bd4a-runc.7yXMrT.mount: Deactivated successfully.
2024-09-28 03:39:15.091 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-28821694a1f753901c290acdab2d95c603d2858543d7017b4aaaea3e5a696406-runc.XTsDiP.mount: Deactivated successfully.
2024-09-28 03:41:54.574 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-b00ef2d9730430e827a495fd9d70d1e7642347b9148e8aaf52b0d5ead705f0ee-runc.WHPyd0.mount: Deactivated successfully.
2024-09-28 03:42:39.707 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.eZMyzT.mount: Deactivated successfully.
2024-09-28 03:46:35.677 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-7bbf8ce9e8d5040c90e20186dc12e688436b3127096898f9d4596bb4633f2a3d-runc.Vq46TH.mount: Deactivated successfully.
2024-09-28 03:54:08.343 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.jObYON.mount: Deactivated successfully.
2024-09-28 03:54:34.099 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-7bbf8ce9e8d5040c90e20186dc12e688436b3127096898f9d4596bb4633f2a3d-runc.EaB2xj.mount: Deactivated successfully.
2024-09-28 03:55:32.644 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-b00ef2d9730430e827a495fd9d70d1e7642347b9148e8aaf52b0d5ead705f0ee-runc.kKilsy.mount: Deactivated successfully.
2024-09-28 03:56:13.915 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.YhALlx.mount: Deactivated successfully.
2024-09-28 03:57:22.042 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-7bbf8ce9e8d5040c90e20186dc12e688436b3127096898f9d4596bb4633f2a3d-runc.NS0Nlc.mount: Deactivated successfully.
2024-09-28 03:58:41.596 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.i5Dj32.mount: Deactivated successfully.
2024-09-28 03:59:03.859 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-7bbf8ce9e8d5040c90e20186dc12e688436b3127096898f9d4596bb4633f2a3d-runc.jouhr4.mount: Deactivated successfully.
2024-09-28 04:00:43.759 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-7bbf8ce9e8d5040c90e20186dc12e688436b3127096898f9d4596bb4633f2a3d-runc.PR3teA.mount: Deactivated successfully.
2024-09-28 04:02:10.341 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.JUct41.mount: Deactivated successfully.
2024-09-28 04:02:38.095 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-02ffabd5c01a055bb14bc525209c674b802b2baa702f55556a2aae88360ba96f-runc.SGP7I8.mount: Deactivated successfully.
2024-09-28 04:05:23.693 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.1xGsL8.mount: Deactivated successfully.
2024-09-28 04:05:54.264 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.G4CATW.mount: Deactivated successfully.
2024-09-28 04:16:20.264 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.pOflPg.mount: Deactivated successfully.
2024-09-28 04:20:36.746 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-7bbf8ce9e8d5040c90e20186dc12e688436b3127096898f9d4596bb4633f2a3d-runc.SQOvcZ.mount: Deactivated successfully.
2024-09-28 04:23:05.491 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-286806ad7e51382948bba4449ab585670a53d680df64e7e1dd4f960a4952f983-runc.uySL9M.mount: Deactivated successfully.
2024-09-28 04:25:13.986 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-286806ad7e51382948bba4449ab585670a53d680df64e7e1dd4f960a4952f983-runc.rEg4KC.mount: Deactivated successfully.
2024-09-28 04:30:12.012 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-7bbf8ce9e8d5040c90e20186dc12e688436b3127096898f9d4596bb4633f2a3d-runc.lYnV5v.mount: Deactivated successfully.
2024-09-28 04:31:05.943 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.bC4vyJ.mount: Deactivated successfully.
2024-09-28 04:43:51.348 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-7bbf8ce9e8d5040c90e20186dc12e688436b3127096898f9d4596bb4633f2a3d-runc.PTkDYF.mount: Deactivated successfully.
2024-09-28 04:45:56.450 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.R46OFu.mount: Deactivated successfully.
[... cut ...]
2024-09-28 15:07:06.114 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-333e156904669b9406be0f42681b1fe7f844b30c97b3e44ffaa53593e8eff865-runc.Wllm99.mount: Deactivated successfully.
2024-09-28 15:07:11.211 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-603118b470842614d03cbe9569acc5e1cd66e49dbe13a11db90358286681aff5-runc.eu9aBq.mount: Deactivated successfully.
2024-09-28 15:08:46.662 MyHostName systemd[1]: run-docker-runtime\x2drunc-moby-286806ad7e51382948bba4449ab585670a53d680df64e7e1dd4f960a4952f983-runc.SbPob4.mount: Deactivated successfully.
```

bcutter, 2024-09-30:
Some day, soon? :-)

sairon, 2024-09-30:
There is no obvious trigger for the issues in the logs but it appears that something caused the dbus-broker exhausted some of the quotas. If you have access to the host console (via the developer SSH on port 22222), then it would be interesting to check `systemctl status dbus-broker` output. Otherwise `ha host logs -b0 -t dbus-broker -n10000 | grep -v "resources to perform"` (which doesn't need the dev SSH) could also show what happened prior to the point where it became broken. I think the rest is just collateral of that problem, the only other suspicious thing that stands out is the `Consumed 7h 37min 8.500s CPU time` for the Bluetooth service, which is quite a lot. What is the Bluetooth being used for on your instance?

I notice you're running HAOS 12.4, there was a [similar report](https://github.com/home-assistant/supervisor/issues/5187) prior to OS 13.0 release in the Supervisor tracker that manifested differently but the dbus-broker was also in that state - the user reported upgrade resolved after the updates, so I'd suggest to update to the latest OS as well - with the latest version the logs wouldn't be cluttered with the "Deactivated successfully" lines too. And on a side note, to get around the issue with the message length, you can attach a text file to the report - for very long logs it better to use that.

bcutter, 2024-09-30:
Thanks. In short:

- no dev SSH via 22222 available
- And `ha host logs -b0 -t dbus-broker -n10000 | grep -v "resources to perform"` gives `Logs only contain 0 boots` which I mentioned at various places already (`-b0` is useless).
- But here's an output, hopefully of interest?
```
  ~ ha host logs -t dbus-broker -n10000 | grep -v "resources to perform" 
Peer :1.1488 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1489 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1490 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1491 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1604 is being disconnected as it does not have the resources to receive a reply it requested.
Peer :1.1766 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1767 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.71 is being disconnected as it does not have the resources to receive a signal it subscribed to.
Peer :1.1492 is being disconnected as it does not have the resources to receive a signal it subscribed to.
Peer :1.1769 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1772 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1773 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1774 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1775 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1776 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1777 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.27 is being disconnected as it does not have the resources to receive a reply it requested.
Peer :1.1779 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
Peer :1.1781 is being disconnected as it does not have the resources to receive a reply or unicast signal it expects.
```
- I agree, a lot of "collateral" symptoms
- Regarding
> What is the Bluetooth being used for on your instance?
Well, looking at what uses Bluetooth at all, basically gives the following integrations:
  - SwitchBot
  - Private BLE Device
  - BTHome
  - BodyMiScale
  - iBeacon Tracker

Now that you mention bluetooth: yesterday I was trying to send a command to a SwitchBot (really almost next to the HA server/Pi 4) several times, which took like forever to process. Literally, minutes (instead of usually a few seconds). Tested via smartphone app (no HA involved) which triggered the SwitchBot immediately. **So yes it looks like Bluetooth in HA has some serious issues - the question is is this a collateral damage (symptom) now or is it the cause?** It's the native Pi 4 chip, no dedicated other bluetooth stick (only bluetooth proxies in ESPHome or Shelly).

Bluetooth has been a thing in HA OS as I saw issues regularly in the issue tracker bout that.

Yes, now that 13.1 is out (I always skip the .0 and forgot about the 13.1 being available for more than 4 weeks now which I consider quite stable, nothing particular found in the open issues section for 13.0/13.1) I think I will first reboot the host with 12.4 to check everything is working smoothly again, create a backup as always and then update to HA OS 13.1 - and hope to never see this again.

Unfortunately https://github.com/home-assistant/supervisor/issues/5187 does not specify which board is used. But I agree, the same symptom is one I suffer from too (one network share not accessible). It's completely irrelevant which HA Core version is used. Also my initial speculation in https://github.com/home-assistant/supervisor/issues/5249 the Supervisor might be the bad guy here is wrong, it's clearly a OS/hardware related thing.

**So do you see anything relevant in the logs?** Otherwise I'd now proceed with (the preparations for) the HA OS 13.1 update.

bcutter, 2024-11-02:
Please note HA OS has been updated to 13.1 at 2024-09-30 and is running since then. Until now the same Networkmanager issue did not occur (yet), but the I think underlying Bluetooth issues now unfold in the way so many other Pi users experienced them. See https://github.com/home-assistant/core/issues/92379#issuecomment-2453189833 for details on this, I hope this helps shedding some light on the Bluetooth issue.

closed 2025-02-09 by github-actions[bot]

bcutter, 2025-02-16:
Roughly 1.5 days after updating HA OS to 14.2 (and Supervisor from 2025.2.0 to 2025.2.1), this old ~~friend~~ enemy is back:

[image]

[image]

[image]

[image]

[image]

https://www.home-assistant.io/more-info/unsupported/network_manager

[image]

[image]

I'm going to reboot the host now (and migrate the system to another machine later on). Supervisor logs without anything special

```
2025-02-16 10:58:59.649 INFO (MainThread) [supervisor.mounts.manager] Reloading mount: Media_RPi1
2025-02-16 11:01:20.321 INFO (MainThread) [supervisor.host.info] Updating local host information
2025-02-16 11:01:21.586 INFO (MainThread) [supervisor.host.services] Updating service information
2025-02-16 11:01:21.593 WARNING (MainThread) [supervisor.host.services] Can't update host service information!
2025-02-16 11:01:21.721 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2025-02-16 11:01:21.735 INFO (MainThread) [supervisor.host.manager] Host information reload completed
```