home-assistant/operating-system #3607: HAOS may create not properly aligned partitions, causing the system to fail to boot

this-user, 2024-10-01:
### Describe the issue you are experiencing

I did a fresh installation of HAOS 13.1 on an x86_64 System with a 128 GB NVMe disk by writing the `generic_x86-64` image to disk by using `dd`. After restarting the system, it only boots partially, receiving a correct network configuration via DHCP and responding to pings, but failing to start fully or even just open any ports.

Upon tracking this issue down, I found that `/mnt/data/` wasn't properly mounted, causing services like Docker to fail, blocking proper start-up. As it turns it out, the issue was that, unlike `hassos-boot` and `hassos-overlay`, `hassos-data` was not present under `/dev/disk/by-label/` and was consequently not being mounted. Mounting it manually worked fine, and `fsck` on the partition returned no issue.

However, the partition was not being listed when using `blkid`, and the `verify` functionality in `gdisk`  gave a warning that the end of last partition (p8) was not properly aligned as a multiple of 2048. It seems this was the issue that caused the partition to not be fully recognised, causing the label to be missing, and causing it not be mounted, thereby prevented the start of HAOS.

I copied the data to a temporary location, deleted the partition in question, then recreated it with proper FS and labels, then restored its contents. This fixed the issue, and HAOS subsequently launched properly.

I was also able to recreate the same issue with HAOS 12.4.

So it seems there is a possibility that HAOS can create these partitions that are not properly aligned that will cause an issue that needs to be fixed manually. The same problem did not occur on other hardware.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

13.1 / 12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

`smpboot: CPU0: Intel(R) Pentium(R) Silver N6000 @ 1.10GHz (family: 0x6, model: 0x9c, stepping: 0x0)
`

This is the re-aligned partition table.

```
# gdisk -l /dev/nvme0n1
GPT fdisk (gdisk) version 1.0.10

Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: present

Found valid GPT with protective MBR; using GPT.
Disk /dev/nvme0n1: 250069680 sectors, 119.2 GiB
Model: kimtigo SSD 128GB                       
Sector size (logical/physical): 512/512 bytes
Disk identifier (GUID): 72AE4EEF-00F3-4B1A-93E4-AFC4E8079C2E
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 250069646
Partitions will be aligned on 2048-sector boundaries
Total free space is 2669 sectors (1.3 MiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048           67583   32.0 MiB    EF00  hassos-boot
   2           67584          116735   24.0 MiB    8300  hassos-kernel0
   3          116736          641023   256.0 MiB   8300  hassos-system0
   4          641024          690175   24.0 MiB    8300  hassos-kernel1
   5          690176         1214463   256.0 MiB   8300  hassos-system1
   6         1214464         1230847   8.0 MiB     8300  hassos-bootstate
   7         1230848         1427455   96.0 MiB    8300  hassos-overlay
   8         1427456       250068991   118.6 GiB   8300  hassos-data
```

### Steps to reproduce the issue

1. Write the HAOS image to disk
2. Reboot
3. Problem occurs
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Unable to copy anything due to local access
```

### Anything in the Host logs that might be useful for us?

```txt
Unable to copy anything due to local access
```

m1dec, 2024-11-28:
Same exact problem.

pcmike, 2024-12-04:
So, I literally just did the same procedure I used to successfully update from 13.1 to 13.2 and this time the update succeeded. Essentially, all I did was.. "host reboot" from the CLI and allow the VM to reboot successfully (it did) and then immediately take a snapshot (just incase) and then attempt an update to HAOS 14 and this time the update completed and rebooted successfully. Obviously there is entirely no way to capture everything that's printed in the console during the reboot, but I did make a screen recording of the console while that was doing its thing.

closed 2025-03-13 by github-actions[bot]

emilhem, 2025-11-07:
I have this issue as well, although I can't even boot due to the partition table being messed up.

It could be because the SSD I have has 4096-byte sectors and I assume that the disk image assumes that you have 512-byte sectors. I'm experimenting with this currently `dd bs=512 ...`