home-assistant/operating-system #3609: Restore from backup never completes and silently fails if attempting to restore from newer supervisor version.

bharvey88, 2024-10-05:
### Describe the issue you are experiencing

i think i found a bug or at the very least non-ideal-behavior from haos. i tried restoring from backup on a brand new install on the landing page after flashing an ssd - > try to backup -> just spins forever -> i eventually reload the gui, and it acts like i never even started a restore.

I finally set it up as a new device and then restore from backup and see an error when scrolling down about my supervisor version being too new (i'm running beta haos). I realize this is expected behavior ultimately, but from an onboarding perspective with restoring from backup this seems like an issue? it should at least pop up with an error and not just spin forever acting like it's doing something right?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

13.2rc1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

beelink n100 mini pc s12 pro bare metal haos install

### Steps to reproduce the issue

1.attempt to restore from backup on a brand new haos install using newer supervisor version than you flashed to the ssd
2.it just loads forever and if you reload the browser page it acts like it never started the restore.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

## System Information

version | core-2024.10.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.52-haos
arch | x86_64
timezone | America/Chicago
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1435
Downloaded Repositories | 50

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.2.rc1
-- | --
update_channel | beta
supervisor_version | supervisor-2024.10.0
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 457.7 GB
disk_used | 52.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (6.5.1), Samba share (12.3.2), APC UPS Daemon (No USB) (1.9), AdGuard Home (5.1.4), Advanced SSH & Web Terminal (19.0.0), AirCast (4.2.2), Dashstream (1.0.1), DbStats (0.5.11), ESPHome (2024.9.2), Frigate (0.14.1), Frigate Beta (0.14.1), Hyperion.NG (2.0.16), Mosquitto broker (6.4.1), Music Assistant Server (2.2.7), Nextcloud Backup (0.17.7), Piper (1.5.2), Studio Code Server (5.17.1), Tailscale (0.22.1), Tailscale with features (0.22.1.0), TubesZB TI CC2652 FW Flasher (0.3.5.2), Whisper (2.1.2), Z-Wave JS UI (3.13.2), Zigbee2MQTT (1.40.2-1), openWakeWord (1.10.0)

</details>

<details><summary>keymaster</summary>

zwave_integration | zwave_js
-- | --
network_status | on

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 17
views | 15
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 6, 2024 at 8:08 PM
-- | --
current_recorder_run | October 4, 2024 at 1:59 PM
estimated_db_size | 5151.56 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

rafikowicz, 2024-11-13:
I had the same problem today when I wanted to restore a backup created on supervisor version 2024.11.3, and the latest supervisor version on the new instance is currently 2024.11.2 (global supervisor issue).
I lost a lot of time waiting on the restore backup view, while the backup actually failed at the very beginning without any information. I only discovered this error after creating a clean HA account and trying to restore the backup from the Settings -> System view.

bharvey88, 2025-02-12:
lol

closed 2025-05-21 by github-actions[bot]