home-assistant/operating-system #3616: Wifi setup

IgorDancik, 2024-10-09:
### Describe the issue you are experiencing

when trying to connect to wifi from HA:
- all available AP visible
- selecting any AP + option (automatic, static) 
- produces mesage: home assistant connect wifi Object does not exist at path “/org/freedesktop/NetworkManager/ActiveConnection/6”

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

13.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

RPi 4
Core2024.10.1
Supervisor2024.10.0
Operating System13.1
Frontend20241002.2

### Steps to reproduce the issue

1. install new HA on LAN
2. select AP + connect options 
3. HP refuses to connect + produces mesage: home assistant connect wifi Object does not exist at path “/org/freedesktop/NetworkManager/ActiveConnection/6”
I manage to connect to wifi in previous versions under similar conditions...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-10-09 05:07:13.478 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-10-09 05:07:13.478 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-10-09 05:07:13.478 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-10-09 05:10:20.142 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
2024-10-09 05:10:21.423 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-10-09 05:23:38.375 INFO (MainThread) [supervisor.host.info] Updating local host information
2024-10-09 05:23:39.004 INFO (MainThread) [supervisor.host.services] Updating service information
2024-10-09 05:23:39.018 INFO (MainThread) [supervisor.host.network] Updating local network information
2024-10-09 05:23:39.225 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2024-10-09 05:23:39.236 INFO (MainThread) [supervisor.host.manager] Host information reload completed
2024-10-09 05:40:21.737 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-10-09 06:07:13.498 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2024-10-09 06:07:13.499 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2024-10-09 06:07:13.499 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-10-09 06:07:13.500 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2024-10-09 06:07:13.504 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2024-10-09 06:07:13.505 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2024-10-09 06:07:13.665 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2024-10-09 06:07:13.679 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2024-10-09 06:07:13.679 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2024-10-09 06:07:13.680 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2024-10-09 06:07:13.681 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2024-10-09 06:07:13.681 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-10-09 06:07:13.681 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2024-10-09 06:07:13.845 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-10-09 06:07:13.845 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-10-09 06:07:13.845 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-10-09 06:10:22.060 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-10-09 06:40:22.375 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-10-09 07:07:13.883 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2024-10-09 07:07:13.884 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2024-10-09 07:07:13.885 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-10-09 07:07:13.885 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
[... 101 lines cut ...]
2024-10-09 01:16:57.435 NewHA systemd[1]: Starting Hostname Service...
2024-10-09 01:16:57.704 NewHA systemd[1]: Started Hostname Service.
2024-10-09 01:16:57.718 NewHA kernel: audit: type=1334 audit(1728436617.715:234): prog-id=65 op=LOAD
2024-10-09 01:16:57.718 NewHA kernel: audit: type=1334 audit(1728436617.715:235): prog-id=66 op=LOAD
2024-10-09 01:16:57.719 NewHA kernel: audit: type=1334 audit(1728436617.715:236): prog-id=67 op=LOAD
2024-10-09 01:16:57.744 NewHA systemd[1]: Starting Time & Date Service...
2024-10-09 01:16:57.978 NewHA systemd[1]: Started Time & Date Service.
2024-10-09 01:17:27.744 NewHA systemd[1]: systemd-hostnamed.service: Deactivated successfully.
2024-10-09 01:17:29.472 NewHA kernel: audit: type=1334 audit(1728436647.846:237): prog-id=64 op=UNLOAD
2024-10-09 01:17:29.472 NewHA kernel: audit: type=1334 audit(1728436647.846:238): prog-id=63 op=UNLOAD
2024-10-09 01:17:29.472 NewHA kernel: audit: type=1334 audit(1728436647.846:239): prog-id=62 op=UNLOAD
2024-10-09 01:17:29.473 NewHA systemd[1]: systemd-timedated.service: Deactivated successfully.
2024-10-09 01:17:29.476 NewHA kernel: audit: type=1334 audit(1728436649.470:240): prog-id=67 op=UNLOAD
2024-10-09 01:17:29.477 NewHA kernel: audit: type=1334 audit(1728436649.470:241): prog-id=66 op=UNLOAD
2024-10-09 01:17:29.477 NewHA kernel: audit: type=1334 audit(1728436649.470:242): prog-id=65 op=UNLOAD
2024-10-09 03:23:38.396 NewHA kernel: audit: type=1334 audit(1728444218.390:243): prog-id=68 op=LOAD
2024-10-09 03:23:38.397 NewHA kernel: audit: type=1334 audit(1728444218.390:244): prog-id=69 op=LOAD
2024-10-09 03:23:38.400 NewHA kernel: audit: type=1334 audit(1728444218.394:245): prog-id=70 op=LOAD
2024-10-09 03:23:38.426 NewHA systemd[1]: Starting Hostname Service...
2024-10-09 03:23:38.691 NewHA systemd[1]: Started Hostname Service.
```

### System information

-

agners, 2024-10-09:
When is the error appearing exactly, when you press Save?

In a quick test on a Raspberry Pi 4 here, setting WiFi configuration did work. Maybe it is something specific with your WiFi? What security mode are you using? What WiFi version and channel are you using?

IgorDancik, 2024-10-09:
Thank you,
Ig

Yes

WiFi is OK, have another HA configured in the past, works fine

WiFi configuration does not work

ROUTER: WPA2-PSK[AES], ha: wpa2 + Password

2.4GHz channel 3, 802.11n

5GHz channel 36, 802.11ax

Auto configure

      other - functioning - ha: Detected: wlan0

      currently problematic: Detected: end0

I tried to see the setting over  ssh + network + ...:

      can see all AP

      wlan0 is inactive

      end0 active + set up

patricksadowski, 2024-10-11:
Same for me. Yesterday I tried to setup a new Home Assistant instance using the Web UI also on rpi4-64. Each retry increases the number in the message:
```
/org/freedesktop/NetworkManager/ActiveConnection/6
/org/freedesktop/NetworkManager/ActiveConnection/7
/org/freedesktop/NetworkManager/ActiveConnection/8
...
```
Happily, I had a backup to restore.

hystrix1, 2024-10-11:
Same problem for me on a Pi4. I previously had HA on a Pi3 working with wifi ok.

IgorDancik, 2024-10-11:
The problem is somewhere in Linux and Network manager - I suppose

JohannesHilgert, 2024-10-12:
Same Problem for me (RPI4, HA 2024.10.1) two days ago, so I am now still using LAN connection as workaround.

I had a similar Installation prior to HA 2024.10 3 weeks ago, which had no major issues configuring wifi.

Chi-Yu, 2024-10-12:
I just got myself a brand new Raspberry Pi 4 to finally try and get into Home Assistant, and the first thing I encounter is a somewhat abysmal user interface for configuring WiFi which spews these unprocessed internal error messages into my face.

It took me a while to find the correct GitHub repository and this issue. I now realize that there is a "Bug reports" link hidden in the "About" section which isn't easy to find and it doesn't event point at this repository in particular.

The first thing I found using my preferred search engine was the "Home Assistant Community" where WiFi issues seem to be dealt with in a very condescending manner of telling you how WiFi is unreliable and making you look like an idiot for not using a cable connection in the first place.

I'm well aware that this is somewhat off-topic, but I hope the people voluntarily working on this project are aware that this isn't the type of pleasant user experience that is going to convince people to switch over from proprietary solutions to something like HA OS.

At least give an error message that tells the average user what to do now and keep the confusing technical details to the logs or a report function.

Do anything but a modal with a cryptic error message and an OK button. Please.

nightfox1e, 2024-10-12:
> ### Describe the issue you are experiencing
> 
> when trying to connect to wifi from HA:
> 
>     * all available AP visible
> 
>     * selecting any AP + option (automatic, static)
> 
>     * produces mesage: home assistant connect wifi Object does not exist at path “/org/freedesktop/NetworkManager/ActiveConnection/6”
> 
> 
> ### What operating system image do you use?
> 
> rpi4-64 (Raspberry Pi 4/400 64-bit OS)
> ### What version of Home Assistant Operating System is installed?
> 
> 13.1
> ### Did the problem occur after upgrading the Operating System?
> 
> No
> ### Hardware details
> 
> RPi 4 Core2024.10.1 Supervisor2024.10.0 Operating System13.1 Frontend20241002.2
> ### Steps to reproduce the issue
> 
>     1. install new HA on LAN
> 
>     2. select AP + connect options
> 
>     3. HP refuses to connect + produces mesage: home assistant connect wifi Object does not exist at path “/org/freedesktop/NetworkManager/ActiveConnection/6”
>        I manage to connect to wifi in previous versions under similar conditions...
> 
> 
> ### Anything in the Supervisor logs that might be useful for us?
> 
> ```
> 2024-10-09 05:07:13.478 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
> 2024-10-09 05:07:13.478 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
> 2024-10-09 05:07:13.478 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
> 2024-10-09 05:10:20.142 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
> 2024-10-09 05:10:21.423 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
> 2024-10-09 05:23:38.375 INFO (MainThread) [supervisor.host.info] Updating local host information
> 2024-10-09 05:23:39.004 INFO (MainThread) [supervisor.host.services] Updating service information
> 2024-10-09 05:23:39.018 INFO (MainThread) [supervisor.host.network] Updating local network information
> 2024-10-09 05:23:39.225 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
> 2024-10-09 05:23:39.236 INFO (MainThread) [supervisor.host.manager] Host information reload completed
> 2024-10-09 05:40:21.737 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
> 2024-10-09 06:07:13.498 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
> 2024-10-09 06:07:13.499 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
> 2024-10-09 06:07:13.499 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
> 2024-10-09 06:07:13.500 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
> 2024-10-09 06:07:13.504 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
> 2024-10-09 06:07:13.505 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
> 2024-10-09 06:07:13.665 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
> 2024-10-09 06:07:13.679 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
> 2024-10-09 06:07:13.679 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
> 2024-10-09 06:07:13.680 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
> 2024-10-09 06:07:13.681 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
> 2024-10-09 06:07:13.681 INFO (MainThread) [supervisor.resolution.check] System checks complete
> 2024-10-09 06:07:13.681 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
> 2024-10-09 06:07:13.845 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
> 2024-10-09 06:07:13.845 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
> 2024-10-09 06:07:13.845 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
> 2024-10-09 06:10:22.060 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
> 2024-10-09 06:40:22.375 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
> 2024-10-09 07:07:13.883 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
> 2024-10-09 07:07:13.884 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
> 2024-10-09 07:07:13.885 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
> 2024-10-09 07:07:13.885 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
> 2024-10-09 07:07:13.886 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
[... 105 lines cut ...]
> 2024-10-09 01:16:57.744 NewHA systemd[1]: Starting Time & Date Service...
> 2024-10-09 01:16:57.978 NewHA systemd[1]: Started Time & Date Service.
> 2024-10-09 01:17:27.744 NewHA systemd[1]: systemd-hostnamed.service: Deactivated successfully.
> 2024-10-09 01:17:29.472 NewHA kernel: audit: type=1334 audit(1728436647.846:237): prog-id=64 op=UNLOAD
> 2024-10-09 01:17:29.472 NewHA kernel: audit: type=1334 audit(1728436647.846:238): prog-id=63 op=UNLOAD
[... cut ...]
> 2024-10-09 03:23:38.396 NewHA kernel: audit: type=1334 audit(1728444218.390:243): prog-id=68 op=LOAD
> 2024-10-09 03:23:38.397 NewHA kernel: audit: type=1334 audit(1728444218.390:244): prog-id=69 op=LOAD
> 2024-10-09 03:23:38.400 NewHA kernel: audit: type=1334 audit(1728444218.394:245): prog-id=70 op=LOAD
> 2024-10-09 03:23:38.426 NewHA systemd[1]: Starting Hostname Service...
> 2024-10-09 03:23:38.691 NewHA systemd[1]: Started Hostname Service.
> ```
> 
> ### System information
> 
> 
> ### Additional information
> 
> _No response_

try to install HAOS normally, on your SD-card or USB stick.

Chi-Yu, 2024-10-12:
> try to install HAOS normally, on your SD-card or USB stick.

I installed HA OS onto an SD-card using Raspberry Pi Imager as described in the official documentation and ran into this same issue. This **is** a normal setup, unless it is not, in which case it either shouldn't be in the official documentation or marked as not officially supported.

Can we skip the not helpful "user must have done something wrong" comments and get to actually fixing the problem?

IgorDancik, 2024-10-12:
Well, my installation was not my first, could not be more "normal" - following manual, using official images documentation....
BUT  Bug is 3 days old so I hope it will be addressed soon... :-(

fiskbil, 2024-10-12:
This bug does not seem to be Raspberry Pi specific, I'm seeing the same error message running 2024.10.2 on x86-64.

rp2019433, 2024-10-13:
Same problem here running HA on x86-64 architecture.

Core 2024.10.2
Supervisor 2024.10.0
Operating System 13.1

[image]