home-assistant/operating-system #3627: Boot Loop Failure - Can't set block device, o valid slot found

Gelob, 2024-10-15:
### Describe the issue you are experiencing

After an AC power loss my home assistant installation will not boot. It boot loops with the following information and goes so fast I had to record a video and single out the frame. This goes on indefinitely. I've had this happen to me twice now, each time I can mount the USB SSD onto another machine but the systemd journals are corrupted. Why does this corruption keep happening? Surely with SSD the installation should survive an AC power loss unclean. The file system itself doesn't appear to be corrupted as it mounts fine.  I have to re-image the drive each time and restore from backup. The system info I'm including is after a new installation.

Googling this error seems its not encountered often in my specific case
[image]
[image]

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.6.31-haos-raspi

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Raspberry Pi 4 with Kingston 120GB A400 Internal SSD SA400S37/120G connected via StarTech.com USB 3.1 to 2.5" SATA Hard Drive Adapter (USB312SAT3CB) with SONOFF Zigbee 3.0 USB Dongle Plus Gateway

### Steps to reproduce the issue

1. remove ac power
2. boot loop
3.

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a cant get logs from original system
```

### Anything in the Host logs that might be useful for us?

```txt
n/a cant get logs from original system
```

### System information

## System Information

version | core-2024.10.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.31-haos-raspi
arch | aarch64
timezone | America/Chicago
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | November 5, 2024 at 6:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-east-1-0.ui.nabu.casa
certificate_status | ready
instance_id | e0dec0695b664930aea1d332a2c28d94
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.10.2
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 109.3 GB
disk_used | 3.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | 

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 0
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 23, 2024 at 7:20 PM
-- | --
current_recorder_run | October 15, 2024 at 7:54 AM
estimated_db_size | 26.04 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

flip8522, 2024-10-22:
Unfortunately I have the same problem, 
Raspberry Pi 4B - Booting from a NVME SSD (SanDisk SSD PLUS M.2 NVMe SSD 250 GB (M.2 2280, PCIe Gen 3.0) with a UGREEN NVMe M.2 USB 3.2 SSD enclosure adapter. I also use a SONOFF Zigbee Gateway, ZBDongle-E 3.0 USB Dongle Plus. 
After reinstalling HassOS and importing the backup, it worked again for a short time, but after about 1 hour the same problem occurred.

gooner86, 2024-11-09:
Same issue here. Any fix?

dominikhoebert, 2024-11-12:
Just updated from 12.3 to 13.2 (I think), It does not boot up afterward, same messages.
Rpi 4, USB3.0 SSD.

eakoning, 2025-01-31:
Is it possible there's a problem with the powersupply? The SSD not getting enough power through the USB? I've been wrestling with an unstable HA installation that seems to stop for no reason (found nothing in the logs afterwards that was helpful) but giving me headaches. I may not be having the same bootloop as @Gelob but I cannot be sure yet. I cannot find my micro HDMI cable to hook it up to a monitor. I've been running the machine headless for a few years already. 
Using a Startech usb312sat3cb in conjunction with Samsung EVO 840 SSD

flip8522, 2025-02-15:
Yes, it was the power supply, after I used a new one the error was gone - just restore HA again and that was it. 
Has been running very stable for a long time now

closed 2025-05-24 by github-actions[bot]