home-assistant/operating-system #3630: HASS OS 13.2 does not install

hugobloem, 2024-10-16:
### Describe the issue you are experiencing

When I click install from the UI the update starts, the system reboots. However, after the reboot the update is still available and the reported OS version is still 13.1. 

I checked the logs but there are no entries that indicate a fault with the installation.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

13.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

- RPi 5 8 GB
- Sonoff Zigbee 3.0 USB Dongle P
- USB SSD for datadisk

### Steps to reproduce the issue

1. Click update
2. System reboots
3. Update not installed, still on 13.1

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-10-16 10:02:12.077 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/raspberrypi5-64-homeassistant with version 2024.10.2
2024-10-16 10:02:12.189 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 13.1 / BootSlot A
2024-10-16 10:02:13.048 WARNING (SyncWorker_1) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Signalk Server
2024-10-16 10:02:13.548 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/e85eeb1c repository
2024-10-16 10:02:13.548 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/c863ef44 repository
2024-10-16 10:02:13.549 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
2024-10-16 10:02:13.550 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/45df7312 repository
2024-10-16 10:02:13.551 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
2024-10-16 10:02:13.553 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/d5369777 repository
2024-10-16 10:02:13.554 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
2024-10-16 10:02:13.556 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/db21ed7f repository
2024-10-16 10:02:14.832 WARNING (SyncWorker_6) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Signalk Server
2024-10-16 10:02:14.954 INFO (MainThread) [supervisor.store] Loading add-ons from store: 183 all - 183 new - 0 remove
2024-10-16 10:02:15.000 INFO (MainThread) [supervisor.addons.manager] Found 13 installed add-ons
2024-10-16 10:02:15.040 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/aarch64 with version 5.17.1
2024-10-16 10:02:15.048 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 9.15.0
2024-10-16 10:02:15.058 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/esphome/esphome-hassio with version 2024.10.0
2024-10-16 10:02:15.065 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mosquitto with version 6.4.1
2024-10-16 10:02:15.075 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/nginxproxymanager/aarch64 with version 1.0.1
2024-10-16 10:02:15.082 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-whisper with version 2.2.0
2024-10-16 10:02:15.088 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-openwakeword with version 1.10.0
2024-10-16 10:02:15.097 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-piper with version 1.5.2
2024-10-16 10:02:15.101 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hugobloem/aarch64-wyoming-microsoft-stt with version 1.0.7
2024-10-16 10:02:15.108 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/wireguard/aarch64 with version 0.10.2
2024-10-16 10:02:15.116 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/music-assistant/server with version 2.2.7
2024-10-16 10:02:15.123 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hugobloem/aarch64-wyoming-microsoft-tts with version 1.0.8
2024-10-16 10:02:15.217 INFO (MainThread) [supervisor.docker.interface] Attaching to zigbee2mqtt/zigbee2mqtt-aarch64 with version 1.40.2-1
2024-10-16 10:02:15.341 INFO (MainThread) [supervisor.backups.manager] Found 49 backup files
2024-10-16 10:02:15.388 INFO (MainThread) [supervisor.discovery] Loaded 7 messages
2024-10-16 10:02:15.388 INFO (MainThread) [supervisor.ingress] Loaded 2 ingress sessions
2024-10-16 10:02:15.388 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup
2024-10-16 10:02:15.388 INFO (MainThread) [supervisor.resolution.checks.base] Run check for detached_addon_missing/addon
2024-10-16 10:02:15.388 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2024-10-16 10:02:15.388 INFO (MainThread) [supervisor.resolution.checks.base] Run check for detached_addon_removed/addon
2024-10-16 10:02:15.389 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-10-16 10:02:15.389 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-10-16 10:02:15.389 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
[... 144 lines cut ...]
2024-10-16 09:03:58.191 homeassistant kernel: audit: type=1334 audit(1729069438.183:397): prog-id=99 op=LOAD
2024-10-16 09:03:58.204 homeassistant systemd[1]: Starting Hostname Service...
2024-10-16 09:03:58.383 homeassistant systemd[1]: Started Hostname Service.
2024-10-16 09:03:58.531 homeassistant kernel: audit: type=1334 audit(1729069438.519:398): prog-id=100 op=LOAD
2024-10-16 09:03:58.531 homeassistant kernel: audit: type=1334 audit(1729069438.519:399): prog-id=101 op=LOAD
2024-10-16 09:03:58.531 homeassistant kernel: audit: type=1334 audit(1729069438.519:400): prog-id=102 op=LOAD
2024-10-16 09:03:58.551 homeassistant systemd[1]: Starting Time & Date Service...
2024-10-16 09:03:58.720 homeassistant systemd[1]: Started Time & Date Service.
2024-10-16 09:03:58.731 homeassistant rauc[6974]: tryboot get-state A
2024-10-16 09:03:58.731 homeassistant rauc[6974]: returning good
2024-10-16 09:03:58.734 homeassistant rauc[6976]: tryboot get-state B
2024-10-16 09:03:58.734 homeassistant rauc[6976]: returning bad
2024-10-16 09:04:28.412 homeassistant systemd[1]: systemd-hostnamed.service: Deactivated successfully.
2024-10-16 09:04:28.543 homeassistant kernel: audit: type=1334 audit(1729069468.538:401): prog-id=99 op=UNLOAD
2024-10-16 09:04:28.543 homeassistant kernel: audit: type=1334 audit(1729069468.538:402): prog-id=98 op=UNLOAD
2024-10-16 09:04:28.543 homeassistant kernel: audit: type=1334 audit(1729069468.538:403): prog-id=97 op=UNLOAD
2024-10-16 09:04:28.749 homeassistant systemd[1]: systemd-timedated.service: Deactivated successfully.
2024-10-16 09:04:28.767 homeassistant kernel: audit: type=1334 audit(1729069468.762:404): prog-id=102 op=UNLOAD
2024-10-16 09:04:28.767 homeassistant kernel: audit: type=1334 audit(1729069468.762:405): prog-id=101 op=UNLOAD
2024-10-16 09:04:28.767 homeassistant kernel: audit: type=1334 audit(1729069468.762:406): prog-id=100 op=UNLOAD
2024-10-16 09:08:41.747 homeassistant kernel: audit: type=1334 audit(1729069721.740:407): prog-id=103 op=LOAD
2024-10-16 09:08:41.748 homeassistant systemd-timesyncd[618]: Network configuration changed, trying to establish connection.
2024-10-16 09:08:41.764 homeassistant systemd[1]: Started Journal Gateway Service.
2024-10-16 09:08:41.767 homeassistant systemd-timesyncd[618]: Contacted time server 162.159.200.123:123 (time.cloudflare.com).
```

agners, 2024-10-16:
Hm, it seems the boot slot B is just not considered good for some reason :thinking: 

Can you follow the logs (using `ha supervisor logs --follow` and `ha host logs --follow` e.g. from the web terminal) while updating and see if the update system reports a problem?

In general, the Raspberry Pi 5 upgrade worked here, so this seems something specific to your instance :thinking:

hugobloem, 2024-10-17:
There does seem to be a problem with the boot slot.

Any suggestions on how to fix this?

```text
2024-10-17 07:21:46.032 homeassistant systemd[1]: Starting Time & Date Service...
2024-10-17 07:21:46.204 homeassistant systemd[1]: Started Time & Date Service.
2024-10-17 07:21:46.213 homeassistant rauc[7398]: tryboot get-state A
2024-10-17 07:21:46.213 homeassistant rauc[7398]: returning good
2024-10-17 07:21:46.215 homeassistant rauc[7400]: tryboot get-state B
2024-10-17 07:21:46.215 homeassistant rauc[7400]: returning bad
```

hugobloem, 2024-10-18:
I am not sure how, but this seems to have resolved itself? :man_shrugging:

closed 2024-10-18 by hugobloem