home-assistant/operating-system #3631: Cant update from 12.4 kernel panic error

Sergmasharov, 2024-10-16:
### Describe the issue you are experiencing

I stack in 12.4. When try to update it reports kernel panic when boots and loading 12.4. I wait for 13.2 OS and get the same error as when I try to update to 13.1
[Video of booting](https://youtu.be/pcK9s4fKM_w)
In for screens is two boot try.
[image]
[image]
[image]
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

In cmdline.txt file (located in /mnt/boot/) appended nolapic and noapic paremeter. (needed to my HW)
Intel Atom(TM) CPU D2000 family CPU
Grub version 2.12

### Steps to reproduce the issue

1. click update
2. rebooting
3. It try to boot slot B. Then error while booting: kernel panic - not syncing. After 3 try, boots 12.4 version from slot A.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

## System Information

version | core-2024.10.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.33-haos
arch | x86_64
timezone | Europe/Samara
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1442
Downloaded Repositories | 22

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.10.2
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 292.7 GB
disk_used | 20.0 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), RPC Shutdown (2.5), MariaDB (2.7.1), Home Assistant Google Drive Backup (0.112.1), Duck DNS (1.18.0), Samba share (12.3.2), Advanced SSH & Web Terminal (19.0.0), FTP (5.1.1), Mosquitto broker (6.4.1), Zigbee2MQTT (1.40.2-1), ESPHome (2024.9.2), HassOS SSH port 22222 Configurator (0.9.3), Terminal & SSH (9.15.0), go2rtc (1.9.4), HDD Tools (1.1.0), Matter Server (6.6.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 9
views | 10
mode | yaml

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 9 октября 2024 г. в 14:00
-- | --
current_recorder_run | 16 октября 2024 г. в 21:25
estimated_db_size | 340.02 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Sonoff</summary>

version | 3.8.1 (ffa7e22)
-- | --
cloud_online | 3 / 6
local_online | 1 / 1

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.7.20
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 1
total_devices | 57

</details>

### Additional information

[core-ssh ~]$ cat /proc/cpuinfo
processor : 0
vendor_id : GenuineIntel
cpu family : 6
model : 54
model name : Genuine Intel(R) CPU @ 1.86GHz
stepping : 1
microcode : 0x10d
cpu MHz : 1861.949
cache size : 512 KB
physical id : 0
siblings : 1
core id : 0
cpu cores : 1
apicid : 0
initial apicid : 0
fpu : yes
fpu_exception : yes
cpuid level : 10
wp : yes
flags : fpu vme de pse tsc msr pae mce cx8 sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx lm constant_tsc arch_perfmon pebs bts nopl nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl est tm2 ssse3 cx16 xtpr pdcm movbe lahf_lm dtherm arat
bugs :
bogomips : 3723.77
clflush size : 64
cache_alignment : 64
address sizes : 36 bits physical, 48 bits virtual

Sergmasharov, 2024-10-16:
Adding hostlog:
`ha host logs -n2000 -b0`
[hostlog0.txt](https://github.com/user-attachments/files/17401239/hostlog0.txt)

Sergmasharov, 2024-10-17:
Maybe it is related to [this](https://github.com/home-assistant/operating-system/pull/3526)
`grub> smbios --type 4 --get-qword 8`
result: 13829424153406670433
[image]

Sergmasharov, 2024-10-17:
Adding info about boot slots:

[image]

agners, 2024-10-21:
> Maybe it is related to [this](https://github.com/home-assistant/operating-system/pull/3526)

Even though you are using an Atom, I don't think this is related as 12.4 already uses the new GRUB2 :thinking: 

It does look like there is an issue reading from the root file system. What type/make of disk are you using?

Sergmasharov, 2024-10-21:
> > Maybe it is related to [this](https://github.com/home-assistant/operating-system/pull/3526)
> 
> Even though you are using an Atom, I don't think this is related as 12.4 already uses the new GRUB2 🤔
> 
> It does look like there is an issue reading from the root file system. What type/make of disk are you using?
You may look in my shot TOSHIBA-MK327 6GSX-442IC2MFT HDD disk.

Sergmasharov, 2024-11-13:
@agners Do you have some ideas?

Sergmasharov, 2024-12-07:
Hi! I tryed to update os to new 14 release and it no boots too... Something changed in releases after 12.4
I downgraded slots A/B without any problems (ha os update --version 12.3), but I cant update to any newer release after 12.4 I stuck on it. System not boot.
What is the problem, maybe some ideas? @agners @sairon I can provide any additional info if needed.
[image]

Patrice3122, 2024-12-19:
> Hi! I tryed to update os to new 14 release and it no boots too... Something changed in releases after 12.4 I downgraded slots A/B without any problems (ha os update --version 12.3), but I cant update to any newer release after 12.4 I stuck on it. System not boot. What is the problem, maybe some ideas? @agners @sairon I can provide any additional info if needed. ![1](https://private-user-images.githubusercontent.com/69318865/393506701-2a47f1b4-86d4-4e49-88e0-7e4e8483cdf0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2NDUxMjMsIm5iZiI6MTczNDY0NDgyMywicGF0aCI6Ii82OTMxODg2NS8zOTM1MDY3MDEtMmE0N2YxYjQtODZkNC00ZTQ5LTg4ZTAtN2U0ZTg0ODNjZGYwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE5VDIxNDcwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM2ODVjZWUzYTFmZjcwY2JmNTZjYjY3NWU0NGYwNjEyNjg0NmJkMjBkMjhhNzgyYmY2NDlhMDZlNzZlYTJmMDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.KdlApZW5jt5txwnoNBFKpbmmJjFNVL5cyu8loFVRLU4)

[Sergmasharov](https://github.com/Sergmasharov), try the following resolution : https://github.com/home-assistant/operating-system/issues/3566#issuecomment-2425199152

Sergmasharov, 2024-12-19:
> > Hi! I tryed to update os to new 14 release and it no boots too... Something changed in releases after 12.4 I downgraded slots A/B without any problems (ha os update --version 12.3), but I cant update to any newer release after 12.4 I stuck on it. System not boot. What is the problem, maybe some ideas? @agners @sairon I can provide any additional info if needed. ![1](https://private-user-images.githubusercontent.com/69318865/393506701-2a47f1b4-86d4-4e49-88e0-7e4e8483cdf0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2NDUxMjMsIm5iZiI6MTczNDY0NDgyMywicGF0aCI6Ii82OTMxODg2NS8zOTM1MDY3MDEtMmE0N2YxYjQtODZkNC00ZTQ5LTg4ZTAtN2U0ZTg0ODNjZGYwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE5VDIxNDcwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM2ODVjZWUzYTFmZjcwY2JmNTZjYjY3NWU0NGYwNjEyNjg0NmJkMjBkMjhhNzgyYmY2NDlhMDZlNzZlYTJmMDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.KdlApZW5jt5txwnoNBFKpbmmJjFNVL5cyu8loFVRLU4)
> 
> [Sergmasharov](https://github.com/Sergmasharov), try the following resolution : [#3566 (comment)](https://github.com/home-assistant/operating-system/issues/3566#issuecomment-2425199152)

Already tried. No luck. Something wrong with drivers in releases after 12.4 I have a kernel panic errors.

Sergmasharov, 2025-01-09:
Still cant update. After updating to 14.1 OS stucks. Still work on 12.4
[image]

Sergmasharov, 2025-03-17:
since 15.0rc1 i have no problems.

closed 2025-03-17 by Sergmasharov