home-assistant/operating-system #3632: Unable to upgrade to 13.2

jfmoots, 2024-10-17:
### Describe the issue you are experiencing

Clicking the Install link yields the following in the Supervisor log

2024-10-16 20:48:25.295 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Destination device '/dev/disk/by-partlabel/hassos-boot' not found

My installation doesn't have a /dev/disk/by-partlabel it has a /dev/disk/by-label directory or /dev/disk/by-partuuid. 

I don't believe anything has changed. This installation has been stable and kept current for a couple years now. 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

6.6.46-haos

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

It's just a VM running on a ProxMox host. AMD processor. 

### Steps to reproduce the issue

1. Log in to Home Assistant
2. Click Settings
3. Click the Home Assistant Operation System 13.2 update
4. Click Install
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-10-16 20:44:34.347 INFO (MainThread) [supervisor.auth] Successful login for 'admin'
2024-10-16 20:48:21.181 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://os-artifacts.home-assistant.io/13.2/haos_ova-13.2.raucb
2024-10-16 20:48:25.135 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-13.2.raucb
2024-10-16 20:48:25.295 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Destination device '/dev/disk/by-partlabel/hassos-boot' not found
```

### Anything in the Host logs that might be useful for us?

```txt
I don't think there's anything here.
```

### System information

## System Information

version | core-2024.10.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.46-haos
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1441
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | August 24, 2025 at 8:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | us-east-1-11.ui.nabu.casa
certificate_status | ready
instance_id | 3397814f31ee476283522e1edeee979e
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.10.2
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 38.7 GB
disk_used | 16.0 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Terminal & SSH (9.15.0), ESPHome (2024.10.0), SQLite Web (4.2.2), Mosquitto broker (6.4.1), Z-Wave JS UI (3.14.0), Z-Wave JS (0.8.0)

</details>

<details><summary>Universal Devices ISY/IoX</summary>

host_reachable | ok
-- | --
device_connected | true
last_heartbeat | October 16, 2024 at 8:59 PM
websocket_status | connected

</details>

<details><summary>keymaster</summary>

zwave_integration | zwave_js
-- | --
network_status | on

</details>

<details><summary>Dashboards</summary>

dashboards | 8
-- | --
resources | 4
views | 15
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | October 11, 2024 at 1:59 PM
-- | --
current_recorder_run | October 16, 2024 at 11:35 AM
estimated_db_size | 1104.55 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

COCottonwood, 2024-10-17:
I am seeing the same issue... I click on the update link and it fails to update.

Running ProxMox on a Beelink NUC N100.

COCottonwood, 2024-10-21:
Update to OS 13.2 was successful when the new CORE update (2024.10.3) came out...  

Currently on  
Core 2024.10.3
Supervisor 2024.10.2
Operating System 13.2
Frontend 20241002.3

jfmoots, 2024-10-22:
I was able to update the Core, no problem. But my OS still won't update. Stuck on 13.1. 

2024-10-21 20:57:35.157 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Destination device '/dev/disk/by-partlabel/hassos-boot' not found

I don't have a /dev/disk/by-partlabel/hassos-boot, I have a /dev/disk/by-label/hassos-boot.

hakkytak, 2024-10-29:
Having the same issue. 
2024-10-29 09:51:28.197 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Destination device '/dev/disk/by-partlabel/hassos-system0' not found

Core 2024.10.4
Supervisor 2024.10.3
Operating System 13.0
Frontend 20241002.4

gky008, 2024-11-24:
Unable to upgrade to 13.2
[image]
[image]

p0058781, 2024-12-05:
I had similar issue, for some reason partlabel (gpt partiton name) was missing on some partitions.

I used sgdisk from ha os root shell to change partition names
For example: (**DONT COPY/PASTE commands below, writing from memory, google correct command or use sgdisk --help**)

`sgdisk /dev/mmcblk0 -c 1:hassos-boot`

This is how it looks like after i added correct partlabels (i have data partition on different drive, ignore that)
```
# lsblk -o name,size,label,partlabel,fstype,mountpoints
NAME           SIZE LABEL          PARTLABEL        FSTYPE   MOUNTPOINTS
mmcblk0        7.3G                                          
|-mmcblk0p1     32M hassos-boot    hassos-boot      vfat     /mnt/boot
|-mmcblk0p2     24M                hassos-kernel0   squashfs 
|-mmcblk0p3    256M                hassos-system0   erofs    
|-mmcblk0p4     24M                hassos-kernel1   squashfs 
|-mmcblk0p5    256M                hassos-system1   erofs    /
|-mmcblk0p6      8M                hassos-bootstate          
`-mmcblk0p7     96M hassos-overlay                  ext4     /var/lib/systemd
                                                             /var/lib/bluetooth
                                                             /var/lib/NetworkManager
                                                             /etc/hostname
                                                             /etc/systemd/timesyncd.conf
                                                             /etc/hosts
                                                             /etc/NetworkManager/system-connections
                                                             /etc/udev/rules.d
                                                             /root/.ssh
                                                             /root/.docker
                                                             /etc/modules-load.d
                                                             /etc/modprobe.d
                                                             /etc/dropbear
                                                             /mnt/overlay
mmcblk0boot0     4M                                          
mmcblk0boot1     4M                                          
zram0            0B                                          
zram1           32M                                          
zram2           16M                                          /tmp
nvme0n1      931.5G                                          
`-nvme0n1p8  931.5G hassos-data                     ext4     /var/log/journal
                                                             /var/lib/docker
                                                             /mnt/data
```

After that first i rebooted and then `ha os update` worked

closed 2025-03-14 by github-actions[bot]