home-assistant/operating-system #3633: OS stuck on 12.3 - unable to update OS

mirkocampisi, 2024-10-17:
### Describe the issue you are experiencing

My system is unable to update OS from 12.3, I tried with every update from 12.4 to 13.2. I try to update from command line and also from Lovelace, but the system reboots always on 12.3. There are no errors on the logs. I searched Home Assistant Log and Supervisor log. I looked into journalctl and found anything. My system is RPi 4 32bit, with OS/boot on SD and data on external SSD. I found that with ha os info I have both boot slots in "good" state, slot A is active with version 12.3 and slot B is inactive with 13.2. I tried the command 'ha os boot-slot B' or 'ha os boot-slot other' but after the reboot the active slot is always A with OS 12.3. 'rauc status' is similar to 'ha os info': kernel.0 with bootname A is good and booted, kernel.1 with bootname B is good and inactive.  

### What operating system image do you use?

rpi4  (Raspberry Pi 4/400 32-bit OS)

### What version of Home Assistant Operating System is installed?

12.3

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

RPi 4 32bit, with OS/boot on SD and data on external SSD. UPS and SDR dongle attached via USB, already tried to update without those connected.

### Steps to reproduce the issue

a. From Lovelace, go to Settings and update OS 
b. from HA CLI -> ha os update OS ha os boot-slot B OR ha os boot-slot other

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m2024-10-17 10:23:04.595 INFO (MainThread) [supervisor.api.middleware.security] /backups/0e9968d1/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.621 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on smeagolworms4/mqtt-explorer with version browser-1.0.3[0m
[32m2024-10-17 10:23:04.623 INFO (MainThread) [supervisor.api.middleware.security] /backups/1ec94983/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.641 INFO (MainThread) [supervisor.api.middleware.security] /backups/b4061e76/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.651 INFO (MainThread) [supervisor.api.middleware.security] /backups/19f21fb8/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.662 INFO (MainThread) [supervisor.api.middleware.security] /backups/856b529d/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.668 INFO (MainThread) [supervisor.api.middleware.security] /backups/7c5950f7/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.674 INFO (MainThread) [supervisor.api.middleware.security] /backups/a9fa098e/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.684 INFO (MainThread) [supervisor.api.middleware.security] /backups/e37ab74e/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.693 INFO (MainThread) [supervisor.api.middleware.security] /backups/5f73e799/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.702 INFO (MainThread) [supervisor.api.middleware.security] /backups/45680b9e/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.710 INFO (MainThread) [supervisor.api.middleware.security] /backups/0be39ed1/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.719 INFO (MainThread) [supervisor.api.middleware.security] /backups/43ba0e47/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.726 INFO (MainThread) [supervisor.api.middleware.security] /backups/17ce873f/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.767 INFO (MainThread) [supervisor.api.middleware.security] /backups/7b36fae8/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.824 INFO (MainThread) [supervisor.api.middleware.security] /backups/49f43310/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.875 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/input/event0 - None[0m
[32m2024-10-17 10:23:04.913 INFO (MainThread) [supervisor.api.middleware.security] /backups/51bdbd90/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.935 INFO (MainThread) [supervisor.api.middleware.security] /backups/2d877a3f/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.944 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/lirc0 - None[0m
[32m2024-10-17 10:23:04.945 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/dvb/adapter0/net0 - None[0m
[32m2024-10-17 10:23:04.946 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/dvb/adapter0/demux0 - None[0m
[32m2024-10-17 10:23:04.963 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/dvb/adapter0/dvr0 - None[0m
[32m2024-10-17 10:23:04.964 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/dvb/adapter0/frontend0 - None[0m
[32m2024-10-17 10:23:04.965 INFO (MainThread) [supervisor.api.middleware.security] /backups/bff30d1b/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.967 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/swradio0 - None[0m
[32m2024-10-17 10:23:04.973 INFO (MainThread) [supervisor.api.middleware.security] /backups/5cee4a73/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.978 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/media4 - None[0m
[32m2024-10-17 10:23:04.989 INFO (MainThread) [supervisor.api.middleware.security] /backups/c68f6bdd/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:04.996 INFO (MainThread) [supervisor.api.middleware.security] /backups/50390a1f/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:05.004 INFO (MainThread) [supervisor.api.middleware.security] /backups/a7ec833d/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:05.016 INFO (MainThread) [supervisor.api.middleware.security] /backups/51e58841/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:05.026 INFO (MainThread) [supervisor.api.middleware.security] /backups/929a24eb/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:05.037 INFO (MainThread) [supervisor.api.middleware.security] /backups/7abdccf9/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:05.046 INFO (MainThread) [supervisor.api.middleware.security] /backups/83a4e65f/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:05.055 INFO (MainThread) [supervisor.api.middleware.security] /backups/7017b5b2/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:05.064 INFO (MainThread) [supervisor.api.middleware.security] /backups/f537ca34/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:05.072 INFO (MainThread) [supervisor.api.middleware.security] /backups/d3e0bff5/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:05.081 INFO (MainThread) [supervisor.api.middleware.security] /backups/c7a927ce/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-10-17 10:23:05.088 INFO (MainThread) [supervisor.api.middleware.security] /backups/016beddc/info access from cebe7a76_hassio_google_drive_backup[0m
[33m2024-10-17 10:23:05.124 WARNING (MainThread) [supervisor.auth] Unauthorized login for 'addons'[0m
[32m2024-10-17 10:23:27.076 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled[0m
[... 292 lines cut ...]
Oct 17 08:18:08 homeassistant rauc[502]: Using deprecated 'Install' D-Bus Method (replaced by 'InstallBundle')
Oct 17 08:18:08 homeassistant rauc[502]: Active slot bootname: A
Oct 17 08:18:08 homeassistant rauc[502]: input bundle: /mnt/data/supervisor/tmp/hassos-13.2.raucb
Oct 17 08:18:08 homeassistant rauc[502]: Installation ab446511 started
Oct 17 08:18:08 homeassistant rauc[502]: installing /mnt/data/supervisor/tmp/hassos-13.2.raucb: started
Oct 17 08:18:08 homeassistant rauc[502]: Reading bundle: /mnt/data/supervisor/tmp/hassos-13.2.raucb
[... cut ...]
Oct 17 08:18:09 homeassistant rauc[502]: installing /mnt/data/supervisor/tmp/hassos-13.2.raucb: Updating slots...
Oct 17 08:18:09 homeassistant rauc[502]: installing /mnt/data/supervisor/tmp/hassos-13.2.raucb: Checking slot boot.0
Oct 17 08:18:09 homeassistant rauc[502]: Updating slot boot.0
Oct 17 08:18:09 homeassistant rauc[502]: Updating /dev/disk/by-partlabel/hassos-boot with 'install' slot hook
Oct 17 08:18:09 homeassistant rauc[502]: installing /mnt/data/supervisor/tmp/hassos-13.2.raucb: Updating slot boot.0

Sergmasharov, 2024-10-17:
Try to upgrade to 12.4 I stuck on it.

KristianMuc, 2024-10-17:
Same issue with rpi5 stuck on version and not able to update since Home operation system 12.3

tescophil, 2024-10-17:
I'm stuck on 12.4, all subsequent version causes a reboot after about 1 minute

mirkocampisi, 2024-10-17:
> Try to upgrade to 12.4 I stuck on it.

Already tried, boot slot B reports versione 12.4 but still impossibile to boot from that slot/version :(

Patrice3122, 2024-10-20:
> > Try to upgrade to 12.4 I stuck on it.
> 
> Already tried, boot slot B reports versione 12.4 but still impossibile to boot from that slot/version :(

I was experiencing a similar issue and managed to resolve it. My resolution has been posted , [https://github.com/home-assistant/operating-system/issues/3566](URL)

mclardass, 2024-10-21:
I've been at 11.2 (or 11.5) for a year with similar log output to OP. If I update through the GUI the OS downgrades to 11.2 and running ha os update --version from the CLI won't go beyond 11.5 on a RPi CM4. Most of the advice I've received on reddit and the community forum is to wipe and start from scratch (not optimal).

mirkocampisi, 2024-10-21:
> > > Try to upgrade to 12.4 I stuck on it.
> > 
> > 
> > Already tried, boot slot B reports versione 12.4 but still impossibile to boot from that slot/version :(
> 
> I was experiencing a similar issue and managed to resolve it.
> 
> 1. Could you try to downgrade your OS version store on Boot B with the command: ha os update --version 13.1
> 2. Restart on **boot B**
> 3. Update the OS with the command : ha os update --version 13.2
> 
> My resolution has been posted [here](https://github.com/home-assistant/operating-system/issues/3566).

I can't restart on boot slot B. As I wrote in the first message:  "_I tried the command 'ha os boot-slot B' or 'ha os boot-slot other' but after the reboot the active slot is always A with OS 12.3._"

mclardass, 2024-10-21:
Made progress after nearly breaking my install. Using the below commands I was able to kill the system but once a new image was downloaded (repair?) the system came up under 11.2. I was then able to update to 11.5 via the GUI and after an hour it prompted me to update to 13.2 (the latest stable). I'm now at the same point as others in that I can see 13.2 (or some 12.x rev) in the alternate slot but cannot boot to it (even with keyboard and monitor attached). 

ha su repair
ha core rebuild
ha host reboot

mirkocampisi, 2024-10-22:
> Made progress after nearly breaking my install. Using the below commands I was able to kill the system but once a new image was downloaded (core rebuild?) the system came up under 11.2. I was then able to update to 11.5 via the GUI and after an hour it prompted me to update to 13.2 (the latest stable). I'm now at the same point as others in that I can see 13.2 (or some 12.x rev) in the alternate slot but cannot boot to it (even with keyboard and monitor attached).
> 
> ha su repair ha core rebuild ha host reboot
> 
> Doing a non-specific 'ha os update' from the CLI gets me to the output below showing 13.2 running but the current version is 11.2 (which is also reflected in the GUI):
> 
> [core-ssh ~]$ ha os info board: rpi4-64 boot: B boot_slots: A: state: inactive status: good version: "11.5" B: state: booted status: good version: "13.2" data_disk: BJTD4R-0 update_available: true version: "11.2" version_latest: "13.2"
> 
> So I have no idea what version is actually running now. Is there some lock file that contains the actual OS version (thinking cat /proc/version or something)?

Thank you for the suggestion, but I tried without success...

mirkocampisi, 2024-10-28:
@sairon @agners any ideas? thanks!

KristianMuc, 2024-12-12:
Still have the problem. But figured out that ha os on boot slot A is working with Version 12.3. Boot Slot B has Status Bad and inactive. Have been able to up or downgrade differentt ha os Versions but the status keeps always bad and inactive. Any idea how to repair boot Slot B?
[image]

I think this is the issue why a general os update is not working.

mirkocampisi, 2024-12-12:
My case is different... The picture is self-explanatory...

[image]

Sergmasharov, 2024-12-12:
I have a little similar case [#3631](https://github.com/home-assistant/operating-system/issues/3631)

mauricecyril, 2024-12-14:
I have a similar issue when trying to update to version 14

`
board: generic-x86-64
boot: A
boot_slots:
    A:
        state: booted
        status: good
        version: "13.1"
    B:
        state: inactive
        status: good
        version: "14.0"
`

Update appears to works but unfortunately installs on Boot Slot B and doesn't boot from it.

mclardass, 2025-01-15:
Since I'm now having issues updating core I thought I'd try the OS update again. 

# ha os info
board: rpi4-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: "13.2"
data_disk: BJTD4R-0xd7f9f43c
update_available: true
version: "11.5"
version_latest: "14.1"

[image]

So, what OS version am I running?!?!

mclardass, 2025-01-15:
Here's the log of the "upgrade", says everything went swimmingly yet the GUI is still showing 11.5 and I don't know other ways to check what the hell us running at this point. I'm trying to clear off another box to migrate services over to before wiping this one out and starting fresh. 

2025-01-14 20:03:07.514 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-01-14 20:03:40.082 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-14.1.raucb
2025-01-14 20:04:05.938 INFO (MainThread) [supervisor.os.manager] Install of Home Assistant Operating System 14.1 success
2025-01-14 20:04:05.940 INFO (MainThread) [supervisor.host.control] Initialize host reboot using logind
2025-01-14 20:04:05.945 INFO (MainThread) [supervisor.addons.manager] Phase 'application' stopping 4 add-ons
2025-01-14 20:04:05.996 INFO (SyncWorker_6) [supervisor.docker.manager] Stopping addon_15d21743_samba_backup application
2025-01-14 20:04:11.516 INFO (SyncWorker_6) [supervisor.docker.manager] Cleaning addon_15d21743_samba_backup application
2025-01-14 20:04:11.627 INFO (SyncWorker_4) [supervisor.docker.manager] Stopping addon_45df7312_zigbee2mqtt application
2025-01-14 20:04:23.400 INFO (SyncWorker_4) [supervisor.docker.manager] Cleaning addon_45df7312_zigbee2mqtt application
2025-01-14 20:04:23.522 INFO (SyncWorker_0) [supervisor.docker.manager] Stopping addon_a0d7b954_appdaemon application
2025-01-14 20:04:23.670 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API for a0d7b954_appdaemon closed
2025-01-14 20:04:28.720 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning addon_a0d7b954_appdaemon application
2025-01-14 20:04:28.828 INFO (SyncWorker_3) [supervisor.docker.manager] Stopping addon_db21ed7f_portainer application
2025-01-14 20:04:32.601 INFO (SyncWorker_3) [supervisor.docker.manager] Cleaning addon_db21ed7f_portainer application
2025-01-14 20:04:32.704 INFO (SyncWorker_7) [supervisor.docker.manager] Stopping homeassistant application
2025-01-14 20:04:47.556 INFO (MainThread) [supervisor.addons.manager] Phase 'services' stopping 4 add-ons
2025-01-14 20:04:47.587 INFO (SyncWorker_1) [supervisor.docker.manager] Stopping addon_core_ssh application

mclardass, 2025-01-17:
Okay, back to the basics.. Downstream provider issue?

traceroute to ghcr.io (140.82.114.34), 30 hops max, 46 byte packets
 1  a889bffc-go2rtc.local.hass.io (172.30.32.1)  0.026 ms  0.024 ms  0.017 ms
 2  192.168.xx.xx (192.168.xx.xx)  1.079 ms  0.905 ms  1.191 ms
 3  10.4.xx.xx (10.4.xx.xx)  25.470 ms  25.394 ms  25.207 ms
 4  P0-1-1-2.BLTMMD-LCR-22.verizon-gni.net (130.81.135.100)  26.372 ms  27.902 ms  27.660 ms
 5  *  *  *
 6  *  *  *
 7  rest-bb1-link.ip.twelve99.net (62.115.121.156)  31.934 ms  31.728 ms  31.416 ms
 8  rest-b2-link.ip.twelve99.net (62.115.123.41)  32.225 ms  33.645 ms  32.602 ms
 9  github-ic-368832.ip.twelve99-cust.net (213.248.67.47)  39.204 ms  40.443 ms  70.630 ms
10  *  *  *
11  *  *  *
12  *  *  *

I had a quicker trace on my side with an extra Verizon hop but after it gets to twelve99 it still dies.

sairon, 2025-01-17:
Usual cause of the versions in the slot info not corresponding with the actual version booted is that there are "two installs" on a same machine - it can happen e.g. when you install the OS to a USB drive and keep the SD card with the full installation in. I don't find any details about your hardware but isn't it the case? Running `lsblk` in the host shell would shed some light on it.

sairon, 2025-01-17:
@mclardass Usual cause of the versions in the slot info not corresponding with the actual version booted is that there are "two installs" on a same machine - it can happen e.g. when you install the OS to a USB drive and keep the SD card with the full installation in. I don't find any details about your hardware but isn't it the case? Running `lsblk` in the host shell would shed some light on it.

mclardass, 2025-01-17:
> [@mclardass](https://github.com/mclardass) Usual cause of the versions in the slot info not corresponding with the actual version booted is that there are "two installs" on a same machine

Hardware is a CM4 on an official I/O board and, until yesterday, everything was installed on the local eMMC. While troubleshooting yesterday I moved the data disk to a USB SSD drive but otherwise the system has always been self-contained with only one drive.

```
lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda            8:0    0 232.9G  0 disk 
`-sda1         8:1    0 232.9G  0 part /var/log/journal
                                       /var/lib/docker
                                       /mnt/data
mmcblk0      179:0    0  29.1G  0 disk 
|-mmcblk0p1  179:1    0    32M  0 part /mnt/boot
|-mmcblk0p2  179:2    0    24M  0 part 
|-mmcblk0p3  179:3    0   256M  0 part /
|-mmcblk0p4  179:4    0    24M  0 part 
|-mmcblk0p5  179:5    0   256M  0 part 
|-mmcblk0p6  179:6    0     8M  0 part 
|-mmcblk0p7  179:7    0    96M  0 part /var/lib/systemd
|                                      /var/lib/bluetooth
|                                      /var/lib/NetworkManager
|                                      /etc/hosts
|                                      /etc/systemd/timesyncd.conf
|                                      /etc/NetworkManager/system-connections
|                                      /etc/hostname
|                                      /root/.ssh
|                                      /root/.docker
|                                      /etc/modules-load.d
|                                      /etc/udev/rules.d
|                                      /etc/modprobe.d
|                                      /etc/dropbear
|                                      /mnt/overlay
`-mmcblk0p8  179:8    0  28.4G  0 part 
mmcblk0boot0 179:32   0     4M  1 disk 
mmcblk0boot1 179:64   0     4M  1 disk 
zram0        254:0    0     0B  0 disk 
zram1        254:1    0    32M  0 disk 
zram2        254:2    0    16M  0 disk /tmp

```
Thanks!

mclardass, 2025-01-18:
Tried another round of setting boot-slot and updating (I had done this many times before):

```
# ha os info
board: rpi4-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: "13.2"
data_disk: External-USB3.0-201703310007F
update_available: true
version: "11.5"     <-----  Current version
version_latest: "14.1"

# ha os boot-slot B

# ha os info
board: rpi4-64
boot: B
boot_slots:
  A:
    state: inactive
    status: good
    version: "14.1"
  B:
    state: booted
    status: good
    version: "13.2"
data_disk: External-USB3.0-201703310007F
update_available: true
version: "11.2"     <----- Older version that system boots to instead of 11.5 for some bizarre reason
version_latest: "14.1"

# ha os update --version 14.1

Command completed successfully.

# ha os info
board: rpi4-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "14.1"
  B:
    state: inactive
    status: good
    version: "13.2"
data_disk: CT250BX100SSD1-1536F00BEF7B
update_available: false
version: "14.1"     <----- And damn, it worked finally
version_latest: "11.5"
```
And I have no idea why it's working now.. Guess I'll see how thing go when 14.2 drops.

mclardass, 2025-01-28:
Applied the 14.2 OS update through the GUI and it appears to have worked normally. So, yay. 
```
[core-ssh ~]$ ha os info
board: rpi4-64
boot: B
boot_slots:
  A:
    state: inactive
    status: good
    version: "14.1"
  B:
    state: booted
    status: good
    version: "14.2"
data_disk: CT250BX100SSD1-1536F00BEF7B
update_available: false
version: "14.2"
version_latest: "14.2"

```

guidorobben, 2025-03-19:
[image]

Version 15.0 doesnt solve it for me.

samsmooth, 2025-04-01:
`board: ova
boot: B
boot_slots:
  A:
    state: inactive
    status: bad
    version: "15.1"
  B:
    state: booted
    status: good
    version: "14.2"
data_disk: QEMU-QEMU-HARDDISK-drive-scsi0
update_available: true
version: "14.2"
version_latest: "15.1"
`

15.1 neither does.

KristianMuc, 2025-07-06:
For me 15.2 doesn’t solve the problem. System still stucks on 13.2 on boot slot A. It looks like 15.2 is installed on Boot Slot B but status still bad and inactive.

[image]

guidorobben, 2025-10-05:
Still hasn't been fixed.