home-assistant/operating-system #3636: Home Assistant OS crashes in Raspberry Pi 4

tyron, 2024-10-18:
### Describe the issue you are experiencing

Recently I started seeing many more crashes on my Home Assistant installation running on a Raspberry Pi 4. It wasn't an issue until recent months, so I'd say it got worse around 2024.08 release. 

From a hunch, it seems to happen more when I'm interacting with Visual Studio add-on. 

Additional details:
* I'm running HA from a SSD attached to my RPi
* I'm using the official Power Supply for RPi4.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

13.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Raspberry Pi 4
2GB RAM
256Gb SSD
Increase swap addon (to add 4GB SWAP to my SSD https://github.com/TazzerMAN/increase_swap_addon)

### Steps to reproduce the issue

1. Use HA regularly
2. Wait randomly until it breaks (not only when I interact with Visual Studio add-on, but very likely to happen when I interact with the add-on)

### Anything in the Supervisor logs that might be useful for us?

```txt
No, logs rotate after the crash/reboot
```

### Anything in the Host logs that might be useful for us?

```txt
No, logs rotate after the crash/reboot
```

### System information

## System Information

version | core-2024.10.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.31-haos-raspi
arch | aarch64
timezone | America/Toronto
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
[... 118 lines cut ...]
Oct 18 00:36:41 homeassistant addon_core_configurator[592]: INFO:2024-10-17 20:36:41,281:hass_configurator.configurator:127.0.0.1 - "GET / HTTP/1.1" 200 -
Oct 18 00:36:41 homeassistant addon_a0d7b954_vscode[592]: File not found: /usr/local/lib/code-server/lib/vscode/out/vsda_bg.wasm
Oct 18 00:36:41 homeassistant addon_a0d7b954_vscode[592]: [20:36:41] [172.30.32.2][abcb3741][ManagementConnection] New connection established.
Oct 18 00:36:42 homeassistant addon_a0d7b954_vscode[592]: [20:36:42] [172.30.32.2][694a7696][ExtensionHostConnection] New connection established.
Oct 18 00:36:42 homeassistant homeassistant[592]: 2024-10-17 20:36:42.285 WARNING (MainThread) [homeassistant.components.sensor] Updating rest sensor took longer than the scheduled update interval 0:00:01
Oct 18 00:36:42 homeassistant addon_a0d7b954_vscode[592]: [20:36:42] [172.30.32.2][694a7696][ExtensionHostConnection] <49901> Launched Extension Host Process.
Oct 18 00:36:42 homeassistant homeassistant[592]: 2024-10-17 20:36:42.421 ERROR (MainThread) [homeassistant.components.rest.data] Error fetching data: http://crealityv3ke.internal:7125/printer/objects/query?heater_bed&extruder&print_stats&toolhead&display_status&virtual_sdcard&gcode_move&webhooks&temperature_sensor mcu_temp&filament_switch_sensor filament_sensor&output_pin fan0&output_pin MainBoardFan failed with All connection attempts failed
Oct 18 00:36:42 homeassistant systemd[1]: docker-4423f9809e45d89bd4cc1a35296fe507a3646f7c386da63720dd6514aadbfe22.scope: Deactivated successfully.
Oct 18 00:36:42 homeassistant homeassistant[592]: 2024-10-17 20:36:42.422 WARNING (MainThread) [homeassistant.components.rest.util] Empty reply found when expecting JSON data
Oct 18 00:36:42 homeassistant kernel: audit: type=1334 audit(1729211802.473:1188): prog-id=276 op=UNLOAD
Oct 18 00:36:42 homeassistant audit: BPF prog-id=276 op=UNLOAD
Oct 18 00:36:42 homeassistant systemd[1]: docker-4423f9809e45d89bd4cc1a35296fe507a3646f7c386da63720dd6514aadbfe22.scope: Consumed 31.591s CPU time.
Oct 18 00:36:42 homeassistant audit: BPF prog-id=279 op=UNLOAD
Oct 18 00:36:42 homeassistant kernel: audit: type=1334 audit(1729211802.493:1189): prog-id=279 op=UNLOAD
Oct 18 00:36:42 homeassistant homeassistant[592]: 2024-10-17 20:36:42.549 ERROR (MainThread) [moonraker_api.websockets.websocketclient] Websocket connection error: Cannot connect to host crealityv3ke.internal:7125 ssl:default [Connect call failed ('192.168.20.121', 7125)]
Oct 18 00:36:42 homeassistant dockerd[592]: time="2024-10-18T00:36:42.612374879Z" level=info msg="ignoring event" container=4423f9809e45d89bd4cc1a35296fe507a3646f7c386da63720dd6514aadbfe22 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
Oct 18 00:36:42 homeassistant containerd[549]: time="2024-10-18T00:36:42.637676235Z" level=info msg="shim disconnected" id=4423f9809e45d89bd4cc1a35296fe507a3646f7c386da63720dd6514aadbfe22 namespace=moby
Oct 18 00:36:42 homeassistant containerd[549]: time="2024-10-18T00:36:42.652806286Z" level=warning msg="cleaning up after shim disconnected" id=4423f9809e45d89bd4cc1a35296fe507a3646f7c386da63720dd6514aadbfe22 namespace=moby
Oct 18 00:36:42 homeassistant containerd[549]: time="2024-10-18T00:36:42.652947358Z" level=info msg="cleaning up dead shim" namespace=moby
Oct 18 00:36:44 homeassistant homeassistant[592]: 2024-10-17 20:36:44.286 WARNING (MainThread) [homeassistant.components.sensor] Updating rest sensor took longer than the scheduled update interval 0:00:01
Oct 18 00:36:46 homeassistant dockerd[592]: time="2024-10-18T00:36:46.649498025Z" level=info msg="Container failed to exit within 10s of signal 15 - using the force" container=4423f9809e45d89bd4cc1a35296fe507a3646f7c386da63720dd6514aadbfe22

Oct 18 00:37:20 homeassistant kernel: brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout

```

ZorroBoyMan, 2024-10-18:
I can confirm the problems with Studio Code Server Addon. The RAM usage of the Addon goes above 15% (4GB on RPI4). After stopping the Addon it's running smooth again. I'm now using File Editor Addon.

tyron, 2024-10-18:
I found [this bug report](https://github.com/hassio-addons/addon-vscode/issues/884) in the VSCode add-on, and I'm in the process of fixing all my automations to the new syntax (based on [this comment](https://github.com/hassio-addons/addon-vscode/issues/884#issuecomment-2416409869)).
I'll update here if that solves my issue

tyron, 2024-10-18:
Initial testing tells me this was the problem indeed. The syntax check is very resource hungry, and the new syntaxes introduced in 2024.08 and expanded in 2024.10 seem to have caused the crash on RPi4. I'll file a bug report in the home-assistant/core repo, to raise attention and asking if there are plans on doing an automated migration of these things.

closed 2024-10-18 by tyron