home-assistant/operating-system #3637: Serial exception after OS upgrade to 13.2

marceldeklerk, 2024-10-18:
### Describe the issue you are experiencing

Hi. Today I upgraded my Home Assistant operating system to version 13.2
Immediately, 15 seconds after it booted, it crashed. This is what i found in the home-assistant.log file

`[bellows.uart] Lost serial connection: SerialException('device reports readiness to read but returned no data (device disconnected or multiple access on port?)')`

Every new attempt to boot my Intel NUC device ends up in HA crashing a few seconds after having booted up. The only way to boot my device is to disconnect my Sonoff Zigbee 3.0 USB dongle from my device. 

Is this making sense to anyone? Especially since this first happened immediately after the upgrade, I think it will be related to it.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.54-haos

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Sonoff zigbee 3.0 USB dongle

## System Information

version | core-2024.10.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.54-haos
arch | x86_64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1442
Downloaded Repositories | 25

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.10.2
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 219.4 GB
disk_used | 23.0 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Let's Encrypt (5.2.2), NGINX Home Assistant SSL proxy (3.11.0), Samba share (12.3.2), TasmoAdmin (0.30.6), Grafana (10.1.2), AirCast (4.2.2), File editor (5.8.0), Mosquitto broker (6.4.1), AdGuard Home (5.1.4), UniFi Network Application (3.4.0), MariaDB (2.7.1), Advanced SSH & Web Terminal (19.0.0), AppDaemon (0.16.7), MQTT Explorer (browser-1.0.1), Glances (0.21.1), Studio Code Server (5.17.2), Log Viewer (0.17.1), go2rtc (1.9.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 14
views | 16
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 18, 2024 at 6:23 PM
-- | --
current_recorder_run | October 18, 2024 at 2:58 PM
estimated_db_size | 2391.89 MiB
database_engine | mysql
database_version | 10.11.6

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Steps to reproduce the issue

1. Reboot the HA device
2. Wait around 15 seconds after home assistant is up
3. Find the error in the home-assistant.log

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-10-18 14:53:19.394 ERROR (bellows.thread_0) [bellows.uart] Lost serial connection: SerialException('device reports readiness to read but returned no data (device disconnected or multiple access on port?)')
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing else than the error already shared.
```

agners, 2024-10-21:
Can you try if downgrading to the previous release using `ha os boot-slot other` helps?

marceldeklerk, 2024-10-22:
Unfortunately that didn’t help. 
The Zigbee controller is fine when I connect it to a new HA installation.

closed 2025-01-29 by github-actions[bot]