home-assistant/operating-system #3641: HomeAssistant OS on Synology VM error at first start

BZH-Taz, 2024-10-21:
### Describe the issue you are experiencing

I tried to install the latest version of Home Assistant OS (haos_ova-13.2.ova) on a virtual machine on my NAS Synology but it failed each time at the first start of Home Assistant. When I connect to the ipadress:8123 I get the error message : "An error occured while installing Home Assistant, check the logs below for more information."

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

n

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

NAS Synology DS920+ with 20Go RAM.

### Steps to reproduce the issue

1. Create a VM using Virtual Machine Manager and import the OVA file (haos_ova-13.2.ova) - 2 core / 4GO RAM / 32GO disk space / UEFI bios... - same as https://community.home-assistant.io/t/installation-on-synology-virtual-machine-managager/281608
2. Start the VM and connect to it until the console show the home assistant start page with IP adress and URL to conect to it
3. Connect to Home Assistant in Firefox with Ipadress:8123 and getting a message saying the HA is installing and it can take 20min
4. After several min I have the error message with the following log...
...

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
[__main__] Initializing Supervisor setup
[supervisor.docker.network] Can't find Supervisor network, creating a new network
[supervisor.bootstrap] Setting up coresys for machine: qemux86-64
[supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/amd64-hassio-supervisor with version 2024.10.2
[supervisor.docker.supervisor] Connecting Supervisor to hassio-network
[supervisor.resolution.evaluate] Starting system evaluation with state initialize
[supervisor.resolution.evaluate] System evaluation complete
[__main__] Setting up Supervisor
[supervisor.api] Starting API on 172.30.32.2
[supervisor.hardware.monitor] Started Supervisor hardware monitor
[supervisor.dbus.manager] Connected to system D-Bus.
[supervisor.dbus.agent] Load dbus interface io.hass.os
[supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
[supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
[supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
[supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
[supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
[supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
[supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
[supervisor.host.services] Updating service information
[supervisor.host.sound] Updating PulseAudio information
[supervisor.host.sound] Can't update PulseAudio data: Failed to connect to pulseaudio server
[supervisor.host.network] Updating local network information
[supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
[supervisor.docker.monitor] Started docker events monitor
[... 244 lines cut ...]
[supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3: 500 Server Error for http+docker://localhost/v1.47/images/create?tag=2024.10.3&fromImage=ghcr.io%2Fhome-assistant%2Fqemux86-64-homeassistant&platform=linux%2Famd64: Internal Server Error ("Head "https://ghcr.io/v2/home-assistant/qemux86-64-homeassistant/manifests/2024.10.3": Get "https://ghcr.io/token?scope=repository%3Ahome-assistant%2Fqemux86-64-homeassistant%3Apull&service=ghcr.io": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")
[supervisor.homeassistant.core] Error on Home Assistant installation. Retrying in 30sec
[supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:landingpage to ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3
[supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2024.10.3.
[supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3: 500 Server Error for http+docker://localhost/v1.47/images/create?tag=2024.10.3&fromImage=ghcr.io%2Fhome-assistant%2Fqemux86-64-homeassistant&platform=linux%2Famd64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")
[supervisor.homeassistant.core] Error on Home Assistant installation. Retrying in 30sec
[supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:landingpage to ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3
[supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2024.10.3.
[supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3: 500 Server Error for http+docker://localhost/v1.47/images/create?tag=2024.10.3&fromImage=ghcr.io%2Fhome-assistant%2Fqemux86-64-homeassistant&platform=linux%2Famd64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")
[supervisor.homeassistant.core] Error on Home Assistant installation. Retrying in 30sec
[supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:landingpage to ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3
[supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2024.10.3.
[supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3: 500 Server Error for http+docker://localhost/v1.47/images/create?tag=2024.10.3&fromImage=ghcr.io%2Fhome-assistant%2Fqemux86-64-homeassistant&platform=linux%2Famd64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")
[supervisor.homeassistant.core] Error on Home Assistant installation. Retrying in 30sec
[supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:landingpage to ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3
[supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2024.10.3.
[supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3: 500 Server Error for http+docker://localhost/v1.47/images/create?tag=2024.10.3&fromImage=ghcr.io%2Fhome-assistant%2Fqemux86-64-homeassistant&platform=linux%2Famd64: Internal Server Error ("Get "https://ghcr.io/v2/": context deadline exceeded")
[supervisor.homeassistant.core] Error on Home Assistant installation. Retrying in 30sec
[supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:landingpage to ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3
[supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2024.10.3.
[supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3: 500 Server Error for http+docker://localhost/v1.47/images/create?tag=2024.10.3&fromImage=ghcr.io%2Fhome-assistant%2Fqemux86-64-homeassistant&platform=linux%2Famd64: Internal Server Error ("Get "https://ghcr.io/v2/": context deadline exceeded (Client.Timeout exceeded while awaiting headers)")
[supervisor.homeassistant.core] Error on Home Assistant installation. Retrying in 30sec
[supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:landingpage to ghcr.io/home-assistant/qemux86-64-homeassistant:2024.10.3
[supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2024.10.3.
```

BZH-Taz, 2024-10-23:
Hello,
No one have an idea on how to solve this issue that block the update of the core of Home Assistant at first start when HAOS has been installed on the VM on a Synology NAS ?

eaglerbits, 2024-10-30:
Hello,
I have the exact same setup and same Issue.
To update the core you can go on terminal and execute: 
`ha core info
ha core update`

Just-Zuul, 2024-10-30:
Excause my bad english - 
I see in your Logs a lot of Errors http 500 at the end of the Logfiles

Do you set all User-Rights in the Synology correctly for the VM? 
Firewall OK? 
You can connect over the Terminal in VM to your HA Installation (CLI)???
Do you download the File to your NAS and install then? or you install from the Net?
I also have the HA in a SynologyNAS DS420+ in a virtual Maschine running (last 9 Month).. 

Give your VM more Space (80 GB) for HA , 2 Core, 2 GB RAM, 

For me it was easy to install with help from this site: https://home-assistant-blog.de/homeassistant-synology-installieren/ 

Maybe it helps  - if you not understand use the translator.. good luck

BZH-Taz, 2024-10-30:
Hello,

Thanks for your return.
I manage to solve my issue but only temporary I'm afraid. I remove my NETGEAR router that manage my home network and connect my NAS directly to the internet box (Orange France - Livebox 5) and with that the installation of HomeAssistant OS goes to the end without issue on a virtual machine on my Synology NAS. After that I put back in place my NETGEAR router and connect all my home network to it and everything is working well, I manage to install many extention to my HomeAssistant without issue so the internet connection of the VM through the NETGEAR router is working well.

I check all the network configuration from my NETGEAR router and I don't see anything wierd and my HomeAssistant conect to internet through it now without issue. So I don't understand why it didn't manage to connect for the update of HA core at the first start.

Regarding your questions, yes I can connect to the Terminal through the VM without issue. About the space available now everything is installed and running it only take 6Go of the 32Go available for this VM. I also try to install it on a 100GO VM without succes at first (same issue at HA core update) so I don't think this was a problem of space available.

Until I discover what was the blocking point at the origin, I'm afraid I will have the same problem next time HAOS need to update the core as the Synology NAS and the VM are back on my home network on the NETGEAR router.

Just-Zuul, 2024-10-30:
well, maybe a netgear own special problem? But there I'm out. ;-)
But the next Updates you make directly over the HA Frontend. you don't have to use the VM anymore for updating.

Make allways enought Backups - you can restore also over the Terminal in the VM if needed, but normaly there are no Problems (just the Problems what comes with the Updates from the Updates ;-)  )...

The Diskspace for HA (if I understand you right) you can only change in the VM when HA is OFF

I also starts with 30 GB but after 4 Month I have to change the Space to 80 because the VM was nearly full.  All Backups also use this space - so if you have 5 Backups you need ~ 10-12 GB after some Times.

Sometimes it is helpfull to restart the complete SYN - 

ah - once - .. If you have a point in your netgear to set for the IP of your Synology to static and that the NAS can open nessesary Ports by itself maybe this you can try to find out when you have Problems.

Anyway - when it runs now - have fun and success

eaglerbits, 2024-10-30:
Hello,
Like I said, I had the same issue I’m my setup.
I’m running HA in a VM in a DS902+.
The VM has 2 cores, 6GB or RAM and 100GB of space. Lots of resources in every dimension.
Outside the DS920+ I have a different setup since I’m using a Asus Router to connect all network devices.

The errors looks to be cause by network communication issues. Not only from VM to Internet but also inside, I mean, between docker containers like supervisor, core and addons.

I rolled back my VM to a snapshot before the HA update and didn’t solved.

Before start getting the errors I updated also the DSM, I think I updated all in the same day so I don’t have sure the origin of the problem but I think is something in between HA and DSM updates.

Hope I can fix this soon because life without HomeAssistant automations is not the same! 😅

eaglerbits, 2024-10-30:
I found a solution today and that’s crazy!!
In the HA config I added 2 new DNS servers to the one already existing (the router IP) and rebooted.
All the crazy logs gone and I have my HA back!

@BZH-Taz maybe your problem was the same and the direct router connection “solved” that issue but if it comes back you can try to set additional DNS servers directly on HA config and it’s done.

Cheers

BZH-Taz, 2024-10-31:
Hello,
How do you set those additional DNS servers in Home Assistant ? I might try it in another VM quickly just to confirm if solve my problem for the future.

eaglerbits, 2024-10-31:
Hello,

In HA web interface under System > Network > Network Interfaces > IPV4 > DNS Servers
I had my router IP and added ,1.1.1.1,8.8.8.8

If you cant access the web interface at all, I think you can also do it using the console.

closed 2025-02-06 by github-actions[bot]