home-assistant/operating-system #3643: unable to load firmware for wlan adapter

antst, 2024-10-22:
### Describe the issue you are experiencing

I am trying to get Intel BE200 working. Out of the box it does not work, as HAOS packs outdated firmware. Trying to update firmware using workaround #2225 does not work either, as it seems that `/mnt/data/supervisor/firmware` is mounted way too late, after `iwlwifi` is already loaded with an error. 

```
# dmesg | grep iwl
[    1.842538] iwlwifi 0000:01:00.0: Detected crf-id 0x2001910, cnv-id 0x2001910 wfpm id 0x80000000
[    1.842553] iwlwifi 0000:01:00.0: PCI dev 272b/00f4, rev=0x472, rfid=0x112200
[    1.844152] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-83.ucode failed with error -2
[    1.845194] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-82.ucode failed with error -2
[    1.845731] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-81.ucode failed with error -2
[    1.847859] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-80.ucode failed with error -2
[    1.848489] iwlwifi 0000:01:00.0: no suitable firmware found!
[    1.848490] iwlwifi 0000:01:00.0: minimum version required: iwlwifi-gl-c0-fm-c0-80
[    1.848491] iwlwifi 0000:01:00.0: maximum version supported: iwlwifi-gl-c0-fm-c0-83
[    1.848492] iwlwifi 0000:01:00.0: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
# 
```

if after boot finished, in console I do
```
#rmmod iwlwifi
#modprobe iwlwifi
```
then module will load firmware and I get working interface. And then dmesg will show proper load.
```
# dmesg | grep iwl
[    1.842538] iwlwifi 0000:01:00.0: Detected crf-id 0x2001910, cnv-id 0x2001910 wfpm id 0x80000000
[    1.842553] iwlwifi 0000:01:00.0: PCI dev 272b/00f4, rev=0x472, rfid=0x112200
[    1.844152] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-83.ucode failed with error -2
[    1.845194] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-82.ucode failed with error -2
[    1.845731] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-81.ucode failed with error -2
[    1.847859] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-80.ucode failed with error -2
[    1.848489] iwlwifi 0000:01:00.0: no suitable firmware found!
[    1.848490] iwlwifi 0000:01:00.0: minimum version required: iwlwifi-gl-c0-fm-c0-80
[    1.848491] iwlwifi 0000:01:00.0: maximum version supported: iwlwifi-gl-c0-fm-c0-83
[    1.848492] iwlwifi 0000:01:00.0: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
....
[  658.746199] iwlwifi 0000:01:00.0: Detected crf-id 0x2001910, cnv-id 0x2001910 wfpm id 0x80000000
[  658.746213] iwlwifi 0000:01:00.0: PCI dev 272b/00f4, rev=0x472, rfid=0x112200
[  658.748633] iwlwifi 0000:01:00.0: api flags index 2 larger than supported by driver
[  658.748645] iwlwifi 0000:01:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.0.4.125
[  658.749163] iwlwifi 0000:01:00.0: loaded firmware version 83.ec13314b.0 gl-c0-fm-c0-83.ucode op_mode iwlmvm
[  658.780865] iwlwifi 0000:01:00.0: Detected Intel(R) TBD Bz device, REV=0x472
[  659.055543] iwlwifi 0000:01:00.0: loaded PNVM version 35c04ca8
[  659.164316] iwlwifi 0000:01:00.0: base HW address: 44:38:e8:db:99:85
[  659.236730] iwlwifi 0000:01:00.0 wlp1s0: renamed from wlan0
[  659.649011] iwlwifi 0000:01:00.0: Registered PHC clock: iwlwifi-PTP, with index: 1
# 
```

so, workaround #2225 is only half-working. Suitable only for firmware loading at later stages. 

so, we need either to mount `/mnt/data` very early in boot (but still unlikely to be good enough) or find a way to inject firmware to initrd or, ideally, have consistency between kernel modules and firmware versions :)

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

13.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

QEMU/KVM 9.0.2
PCI (WLAN) and USB(BT) passthrough for BE200 

### Steps to reproduce the issue

1. Install Intel BE200 adapter
2. Create (if missing) /mnt/data/supervisor/firmware
3. Put iwlwifi-gl-c0-fm-c0* files from fresh linux-firmware git
4.  Boot
...

### Anything in the Supervisor logs that might be useful for us?

```txt
irrelevant
```

### Anything in the Host logs that might be useful for us?

```txt
relevant part of dmesg

# dmesg | grep iwl
[    1.842538] iwlwifi 0000:01:00.0: Detected crf-id 0x2001910, cnv-id 0x2001910 wfpm id 0x80000000
[    1.842553] iwlwifi 0000:01:00.0: PCI dev 272b/00f4, rev=0x472, rfid=0x112200
[    1.844152] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-83.ucode failed with error -2
[    1.845194] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-82.ucode failed with error -2
[    1.845731] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-81.ucode failed with error -2
[    1.847859] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-80.ucode failed with error -2
[    1.848489] iwlwifi 0000:01:00.0: no suitable firmware found!
[    1.848490] iwlwifi 0000:01:00.0: minimum version required: iwlwifi-gl-c0-fm-c0-80
[    1.848491] iwlwifi 0000:01:00.0: maximum version supported: iwlwifi-gl-c0-fm-c0-83
[    1.848492] iwlwifi 0000:01:00.0: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
```
```

### System information

## System Information

version | core-2024.10.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.54-haos
arch | x86_64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1446
Downloaded Repositories | 24

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.10.2
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 30.8 GB
disk_used | 12.7 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Terminal & SSH (9.15.0), EMQX (0.7.1), Zigbee2MQTT Edge (edge), Hass.io Access Point (0.5.2.1), Music Assistant Server (2.2.9)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 13
views | 13
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | October 21, 2024 at 3:03 PM
-- | --
current_recorder_run | October 22, 2024 at 4:53 PM
estimated_db_size | 211.03 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

sairon, 2024-11-04:
> so, workaround #2225 is only half-working. Suitable only for firmware loading at later stages.
>
> so, we need either to mount `/mnt/data` very early in boot (but still unlikely to be good enough) or [...]

Your conclusion is (almost) correct, WiFi is indeed initialized too early and adding the firmware to the symlinked directory doesn't work. However, the problem isn't the inconsistency between firmware versions and kernel - the firmware for BE200 is simply not installed by the [`linux-firmware` package](https://github.com/home-assistant/buildroot/blob/a2c10a16a02a3c665cba045a04fb0625aef215d1/package/linux-firmware/linux-firmware.mk). Also note that HAOS doesn't use initrd.

As a remedy for your problem, the solution would be to add the firmware to the list of installed firmwares for Intel WiFi cards. Unlike other distributions we don't to that pre-emptively, as we need to cope with rootfs space constraints, but given that this WiFi cards doesn't seem to be an obscure one, adding support for it makes sense to me.

antst, 2024-11-12:
``

> Your conclusion is (almost) correct, WiFi is indeed initialized too early and adding the firmware to the symlinked directory doesn't work. However, the problem isn't the inconsistency between [...]

Hm...strange, my memory tells me that I spotted somewhere firmware, but wrong versions, bellow iwlwifi-gl-c0-fm-c0-80 or above iwlwifi-gl-c0-fm-c0-83. But maybe my memory didn't serve well :)
Because right now I am checking and don't se any IWL firmware which would match `iwlwifi-gl-c0*`.

antst, 2024-11-12:
so, question is...any workaround? before firmware find its way into hassos?

antst, 2024-11-12:
OK, to simplify life for everyone, I added PRs to two repos
#3668
and 
https://github.com/home-assistant/buildroot/pull/59

antst, 2024-11-13:
and, as temporary workaround, I built image myself with provided patches, and updated with rauc. So, patches are also checked to work.

commit b28dd8c4 (2024-11-14): Add iwlwifi-gl firmware needed for Intel BE200 Wi-Fi 7 card (#3668)

Add iwlwifi-gl firmware, which is required for Intel BE200 card. Targets are
generic_aarch64, generic_x86_64 and ova.

* buildroot 19027bc796...1d7407c66b (1):
  > package/linux-firmware: add iwlwifi gl firmware

Closes #3643.
closed 2024-11-14 by sairon