home-assistant/operating-system #3654: vCenter 8 ova import

Jeremy-Boyle, 2024-11-01:
### Describe the issue you are experiencing

Error: Failed to deploy OVF package. Cause: A specified parameter was not correct: The provided manifest file is invalid: Invalid OVF checksum algorithm: SHA2-256.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

13.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

NA

### Steps to reproduce the issue

1. Upload OVA

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

N/A

### Additional information

To fix
```bash
tar -xvf haos_ova-13.2.ova
home-assistant.ovf
home-assistant.vmdk
home-assistant.mf
sha1sum home-assistant.vmdk > new_manifest.mf
sha1sum home-assistant.ovf >> new_manifest.mf

# Manually edit format in new_manifest.mf
cat new_manifest.mf
SHA1(home-assistant.vmdk)=390fb0e2f4fc6bf8fc9b7dcc0050bba417f5edee
SHA1(home-assistant.ovf)=d4d00cef71a3b735a4309bbbbbfcd322c01cd318
mv home-assistant.mf home-assistant-2.mf
mv new_manifest.mf home-assistant.mf
tar -cvf haos_ova-13.2.ova home-assistant.ovf home-assistant.mf home-assistant.vmdk
```

jlusky, 2024-12-12:
one liner for recreating the manifest (doesn't include the tarring/untarring):
```
for i in *.ovf *.vmdk ; do echo "SHA1($i)= `sha1sum $i` | awk '{print $1}" ; done > home-assistant.mf
```

Hossy, 2025-01-14:
+1 on this.  OVA was generated 12/19/2024.  vCenter 8

rigorm, 2025-03-16:
+1 the workaround did work( thanks @Jeremy-Boyle ) but this should be fixed when releasing the haos OVA image. haos_14.2 (downloaded on 2025-03-16).

commit 2f0ffbe4 (2025-04-11): Change algorithm name in OVA manifest from SHA2-256 to SHA256 (#3981)

Update of OpenSSL in OS 12.2 from 1.1.1 to 3.2 changed the output of `openssl
sha256` command. It seems that some hypervisors don't like this and fail if
it's not plain "SHA256".

Fixes #3654