home-assistant/operating-system #3656: HAOS fails if virtual hardware switches from Intel 440FX to Q35

wojciechczyz, 2024-11-04:
### Describe the issue you are experiencing

I want to upgrade the VM configuration of HAOS. I switched from Intel 440FX to Q35.
HAOS boots but does not connect network interfaces nor console and stays with 1 CPU on 100% usage.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

13.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

VM on Proxmox host

### Steps to reproduce the issue

1. Change VM hardware from Intel 440FX to Q35.
2. Reboot
3.

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

Impact123, 2024-11-04:
I can reproduce the network issue but not the 100% CPU utilization. I recommend you run `top -o %CPU` to check why that is. I'm guessing it's the DNS plugin.

Anyways, here's the output of `ha network info`.
```yaml
# ha network info
docker:
  address: 172.30.32.0/23
  dns: 172.30.32.3
  gateway: 172.30.32.1
  interface: hassio
host_internet: null
interfaces:
- connected: false
  enabled: false
  interface: enp6s18
  ipv4:
    address: []
    gateway: null
    method: disabled
    nameservers: []
    ready: false
  ipv6:
    address: []
    gateway: null
    method: disabled
    nameservers: []
    ready: false
  mac: XX
  primary: false
  type: ethernet
  vlan: null
  wifi: null
supervisor_internet: false
```
And `nmcli con`.
```bash
Supervisor enp0s18  ca1fcaf2-f250-306f-9989-afcfe8a38b27  ethernet  --
```
See the difference in the device name? `/mnt/overlay/etc/NetworkManager/system-connections/` still has the old name.
Manually triggering `ha supervisor repair` and rebooting does not seem to fix it.
I used this to fix it: https://github.com/home-assistant/operating-system/blob/dev/Documentation/network.md#reset-network.
I suppose you can also use `nmcli`. `nmtui` should probably be made part of the OS too.

Impact123, 2024-11-04:
Please don't needlessly quote my whole message. It prevents me from editing it to fix things and clutters the issue.
I used noVNC and xterm via a serial port.

wojciechczyz, 2024-11-04:
What is the type of the display that you set in the Hardware/Display? Tried Default, Standard VGA, Serial - yet result it is the same, does not connect to the console

Impact123, 2024-11-04:
Just the default. I also added a serial port for xterm. If you come to discord I can help you with that. Let's try to keep this issue focuses on the actual bug. Remember that lots of peope get notified when a post is made here.

closed 2025-02-10 by github-actions[bot]