home-assistant/operating-system #3660: Home Assistant Green wipe (reset to factory settings) failed after using external SSD as data disk

schellevis, 2024-11-05:
### Describe the issue you are experiencing

The external SSD my Home Assistant Green used as data disk got corrupted because the cable got a little bit loose. When trying the new factory reset introduced in HAOS 13.1, the device seemed to still initiate the usual booting sequence. The device was pingable, but observer and HA Core were down. When connected a screen, it kept showing the "Waiting for the Home Assistant CLI to be ready" line.

One log file survived the wipe I eventually did using the SD card procedure, and although the date and time don't match, the attached USB keyboard and the errors reflect the situation yesterday. On screen, there was a message date and time were not reliable. The log shows the reset was not initiated:

`mei 27 23:48:37 homeassistant systemd[1]: Home Assistant OS Device Wipe Service was skipped because of an unmet condition check (ConditionKernelCommandLine=haos.wipe=1).`

I had HAOS 13.2 installed and followed the [reset to factory settings guide](https://green.home-assistant.io/guides/reset/). During the reset procedure, the orange led turned on solid as shown in the GIF. After that, there seemed to be no difference in behaviour between booting the HA Green with or without the reset procedure. 

When booting the device without the external SSD, it went into the emergency console. This happened both with and without the reset button pressed at power on.

### What operating system image do you use?

green (Home Assistant Green)

### What version of Home Assistant Operating System is installed?

13.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Home Assistant Green with 250GB external SSD attached.

The Fan-X usb dongle (for controlling ventilation) was also attached during use: https://github.com/Marcelh1/fanx/wiki

### Steps to reproduce the issue

1. move HA data to external data disk 
2. corrupt data disk
3. try to initiate firmware reset / wipe

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
mmc errors:

mei 27 23:48:36 homeassistant kernel: mmc0: new HS200 MMC card at address 0001
mei 27 23:48:36 homeassistant kernel: mmcblk0: mmc0:0001 032G01 29.1 GiB
mei 27 23:48:36 homeassistant kernel: GPT:Primary header thinks Alt. header is not at the end of the disk.
mei 27 23:48:36 homeassistant kernel: GPT:4194303 != 61071359
mei 27 23:48:36 homeassistant kernel: GPT:Alternate GPT header not at the end of the disk.
mei 27 23:48:36 homeassistant kernel: GPT:4194303 != 61071359
mei 27 23:48:36 homeassistant kernel: GPT: Use GNU Parted to correct GPT errors.
mei 27 23:48:36 homeassistant kernel:  mmcblk0: p1 p2 p3 p4 p5 p6 p7 p8
mei 27 23:48:36 homeassistant kernel: mmcblk0boot0: mmc0:0001 032G01 8.00 MiB
mei 27 23:48:36 homeassistant kernel: mmcblk0boot1: mmc0:0001 032G01 8.00 MiB
mei 27 23:48:36 homeassistant kernel: mmcblk0rpmb: mmc0:0001 032G01 4.00 MiB, chardev (237:0)
mei 27 23:48:36 homeassistant kernel: printk: console [netcon0] enabled
mei 27 23:48:36 homeassistant kernel: netconsole: network logging started
mei 27 23:48:36 homeassistant kernel: clk: Disabling unused clocks
mei 27 23:48:36 homeassistant kernel: md: Waiting for all devices to be available before autodetect
mei 27 23:48:36 homeassistant kernel: md: If you don't use raid, use raid=noautodetect
mei 27 23:48:36 homeassistant kernel: md: Autodetecting RAID arrays.
mei 27 23:48:36 homeassistant kernel: md: autorun ...
mei 27 23:48:36 homeassistant kernel: md: ... autorun DONE.
mei 27 23:48:36 homeassistant kernel: F2FS-fs (mmcblk0p3): Magic Mismatch, valid(0xf2f52010) - read(0xe0f5e1e2)
mei 27 23:48:36 homeassistant kernel: F2FS-fs (mmcblk0p3): Can't find valid F2FS filesystem in 1th superblock
mei 27 23:48:36 homeassistant kernel: F2FS-fs (mmcblk0p3): Magic Mismatch, valid(0xf2f52010) - read(0xc0d100a8)
mei 27 23:48:36 homeassistant kernel: F2FS-fs (mmcblk0p3): Can't find valid F2FS filesystem in 2th superblock
[... 47 lines cut ...]
mei 27 23:48:37 homeassistant kernel: panfrost fde60000.gpu: features: 00000000,00000cf7, issues: 00000000,00000400
mei 27 23:48:37 homeassistant kernel: hantro-vpu fdee0000.video-codec: Adding to iommu group 1
mei 27 23:48:37 homeassistant kernel: panfrost fde60000.gpu: Features: L2:0x07110206 Shader:0x00000002 Tiler:0x00000209 Mem:0x1 MMU:0x00002823 AS:0xff JS:0x7
mei 27 23:48:37 homeassistant kernel: panfrost fde60000.gpu: shader_present=0x1 l2_present=0x1
mei 27 23:48:37 homeassistant kernel: hantro-vpu fdee0000.video-codec: registered rockchip,rk3568-vepu-enc as /dev/video2
mei 27 23:48:37 homeassistant kernel: [drm] Initialized panfrost 1.2.0 20180908 for fde60000.gpu on minor 0
mei 27 23:48:37 homeassistant systemd[1]: Mounting HassOS overlay partition...
mei 27 23:48:37 homeassistant kernel: rockchip-drm display-subsystem: bound fe040000.vop (ops vop2_component_ops [rockchipdrm])
mei 27 23:48:37 homeassistant kernel: EXT4-fs (mmcblk0p7): mounted filesystem da75077a-dcfa-4c55-bcdc-8128f3e546d3 r/w with ordered data mode. Quota mode: none.
mei 27 23:48:37 homeassistant systemd[1]: Mounted HassOS overlay partition.
mei 27 23:48:37 homeassistant systemd[1]: Mounting Dropbear persistent keys...
mei 27 23:48:37 homeassistant systemd[1]: etc-modprobe.d.mount: Directory /etc/modprobe.d to mount over is not empty, mounting anyway.
mei 27 23:48:37 homeassistant systemd[1]: Mounting Kernel persistent modprobe.d...
mei 27 23:48:37 homeassistant systemd[1]: Mounting Kernel persistent modules-load.d...
mei 27 23:48:37 homeassistant systemd[1]: Mounting Udev persistent rules.d...
mei 27 23:48:37 homeassistant systemd[1]: Mounting Docker persistent user...
mei 27 23:48:37 homeassistant kernel: dwhdmi-rockchip fe0a0000.hdmi: Detected HDMI TX controller v2.11a with HDCP (DWC HDMI 2.0 TX PHY)
mei 27 23:48:37 homeassistant kernel: dwhdmi-rockchip fe0a0000.hdmi: registered DesignWare HDMI I2C bus driver
mei 27 23:48:37 homeassistant kernel: rockchip-drm display-subsystem: bound fe0a0000.hdmi (ops dw_hdmi_rockchip_ops [rockchipdrm])
mei 27 23:48:37 homeassistant systemd[1]: Mounting Dropbear persistent user...
mei 27 23:48:37 homeassistant systemd[1]: Home Assistant OS data partition migration was skipped because of an unmet condition check (ConditionPathExists=/mnt/overlay/move-data).
mei 27 23:48:37 homeassistant systemd[1]: Starting HassOS data resizing...
mei 27 23:48:37 homeassistant systemd[1]: Starting HassOS overlay setup...
<end of file>
```

gky008, 2024-12-05:
[image]
[image]

siegou, 2025-01-03:
Hi there, 

I got the same issue.
Home assistant green setup with data moved to an external nvme SSD through USB. 
Unable to reset the device :
When nvme is attached
When nvme is not attached
With nothing plugged except network.
Through sdcard reset procedure as well.

With the nvme attached and without pushing the reset button, the device boot normaly as no reset had been done.

siegou, 2025-01-22:
up

siegou, 2025-01-22:
I finally manage to reset the device.
I think there is a bug when data is on an external storage, reset procedure do not rename hassos-data-old in hassos-data in order to have the reset working.
I had to connect with a screen + keyboard to have a shell from where i've been able to rename labelling.
I followed the how-to here : https://community.home-assistant.io/t/waiting-for-home-assistant-cli-to-be-ready/635234

in the how-to data mounting point is on /dev/sda8 but for a home assistant green, device is /dev/mmcblk0p8. Basically, i think it's the 8th partition of the disk.

Changing the label and tadaaaaaaaaaaa, reset work like a charm.
I've found as well why it's not working with sd-card : I open the device and it seems an sd-card has been inserted in the wrong face and broken a pin of the sd-card reader. box was not able to read the card. As I've a way to reset, i will not ask for a device exchange (and i'm not sure the warranty is still ongoing). I've found a procedure to reset from an usb device and playing with mount / dd and other linux command... I hope i will never fall in this very last option to recover.

Hope this post will help

Nzara, 2026-02-08:
I encountered a problem with Home Assistant OS 17.0. Nothing is attached to the system. 
It was probably caused by a power interruption during a factory reset.