home-assistant/operating-system #3670: ZHA will randomly disconnect and reconnect in a loop

NightHawkATL, 2024-11-13:
### Describe the issue you are experiencing

I have noticed since the latest update (2024.11.1) that ZHA will disconnect from the Zigbee radio I am using and constantly reconnect/disconnect until I reboot the radio (TubesZB cc2652-poe-2023). I have tried to narrow it down as to what could be causing it but the only error I get in the logs is this:
[image]

I have recently moved the radio to get better performance and reliability but even moving the device hasn't helped with the error I am seeing.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

13.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Home Assistant is currently running as a VM on Proxmox 8.2.7.
[image]
[image]

### Steps to reproduce the issue

1. I haven't figured out the trigger yet other than the error message provided.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-11-13 10:10:45.507 INFO (MainThread) [supervisor.store] Loading add-ons from store: 96 all - 0 new - 0 remove
2024-11-13 10:10:45.507 INFO (MainThread) [supervisor.store] Loading add-ons from store: 96 all - 0 new - 0 remove
2024-11-13 10:11:46.124 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
2024-11-13 10:12:09.457 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
2024-11-13 10:12:09.459 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/wez/govee2mqtt repository
2024-11-13 10:12:09.462 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/ESPresense/hassio-addons repository
2024-11-13 10:12:09.464 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/mercenaruss/zigstar_addons repository
2024-11-13 10:12:09.470 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/tube0013/tubeszb_addons repository
2024-11-13 10:12:09.482 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/mib1185/ha-addon-syslog repository
2024-11-13 10:12:09.483 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/brenner-tobias/ha-addons repository
2024-11-13 10:12:09.486 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository
2024-11-13 10:12:09.487 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/music-assistant/home-assistant-addon repository
2024-11-13 10:12:09.492 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
2024-11-13 10:12:09.495 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
2024-11-13 10:12:09.497 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
2024-11-13 10:12:09.498 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/koush/scrypted repository
2024-11-13 10:12:10.322 WARNING (SyncWorker_0) [supervisor.store.data] Can't read /data/addons/git/09e60fb6/external/ring-client-api/packages/homebridge-ring/config.schema.json: Invalid Add-on config!. Got {'pluginAlias': 'Ring', 'pluginType': 'platform', 'singular': True, 'customUi': True, 'customUiPath': './lib/homebridge-ui', 'customUiDevServer': 'http://localhost:3000/api/plugins/settings-ui/homebridge-ring', 'headerDisplay': 'See [the wiki](https://github.com/dgreif/ring/wiki/Refresh-Tokens) for details on generating a refresh token', 'accountLinkingHeader': 'Please supply your Ring credentials.  These will be used to contact the Ring servers and generate a unique access token for your acc...
2024-11-13 10:12:10.589 WARNING (SyncWorker_5) [supervisor.store.data] Can't read /data/addons/git/09e60fb6/external/ring-client-api/packages/homebridge-ring/config.schema.json: Invalid Add-on config!. Got {'pluginAlias': 'Ring', 'pluginType': 'platform', 'singular': True, 'customUi': True, 'customUiPath': './lib/homebridge-ui', 'customUiDevServer': 'http://localhost:3000/api/plugins/settings-ui/homebridge-ring', 'headerDisplay': 'See [the wiki](https://github.com/dgreif/ring/wiki/Refresh-Tokens) for details on generating a refresh token', 'accountLinkingHeader': 'Please supply your Ring credentials.  These will be used to contact the Ring servers and generate a unique access token for your acc...
2024-11-13 10:12:10.745 INFO (MainThread) [supervisor.store] Loading add-ons from store: 96 all - 0 new - 0 remove
2024-11-13 10:12:10.745 INFO (MainThread) [supervisor.store] Loading add-ons from store: 96 all - 0 new - 0 remove

Not sure why the "Ring" stuff is in there because I am not even using that integration.
```

### Anything in the Host logs that might be useful for us?

```txt
2024-11-13 11:30:18.607 ha-vm systemd[1]: Starting Time & Date Service...
2024-11-13 11:30:18.851 ha-vm systemd[1]: Started Time & Date Service.
2024-11-13 11:30:48.621 ha-vm systemd[1]: systemd-hostnamed.service: Deactivated successfully.
2024-11-13 11:30:48.651 ha-vm kernel: audit: type=1334 audit(1731497448.650:731): prog-id=279 op=UNLOAD
2024-11-13 11:30:48.651 ha-vm kernel: audit: type=1334 audit(1731497448.650:732): prog-id=278 op=UNLOAD
2024-11-13 11:30:48.651 ha-vm kernel: audit: type=1334 audit(1731497448.650:733): prog-id=277 op=UNLOAD
2024-11-13 11:30:48.870 ha-vm systemd[1]: systemd-timedated.service: Deactivated successfully.
2024-11-13 11:30:48.885 ha-vm kernel: audit: type=1334 audit(1731497448.884:734): prog-id=282 op=UNLOAD
[... 113 lines cut ...]
dashboards | 13
-- | --
resources | 13
views | 78
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | November 4, 2024 at 2:28 PM
-- | --
current_recorder_run | November 13, 2024 at 2:08 AM
estimated_db_size | 7941.47 MiB
database_engine | mysql
database_version | 10.11.6

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

closed 2024-11-13 by github-actions[bot]

puddly, 2024-11-13:
@NightHawkATL When opening a Core issue, can you enable ZHA debug logging and capture the time period surrounding ZHA disconnecting from the radio and trying to re-connect?

NightHawkATL, 2024-11-13:
> @NightHawkATL When opening a Core issue, can you enable ZHA debug logging and capture the time period surrounding ZHA disconnecting from the radio and trying to re-connect?

Where would I enable this? I looked through ZHA and don't see that option to enable.

puddly, 2024-11-13:
Click "Configure" on the ZHA integration and it'll be in the column of buttons, at the bottom. It'll only download a log after you disable it.

[image]

NightHawkATL, 2024-11-13:
Found it just as you posted. sorry.

NightHawkATL, 2024-11-13:
I will share what I get the next time it gets triggered. Thanks for keeping this ticket open!

puddly, 2024-11-13:
Not a problem! When uploading the log, make sure to ZIP it up, it'll shrink the file to a size GitHub will accept.

NightHawkATL, 2024-11-14:
[home-assistant_zha_2024-11-14T14-56-57.533Z.log](https://github.com/user-attachments/files/17750551/home-assistant_zha_2024-11-14T14-56-57.533Z.log)

I noticed it happen overnight (6 hours ago per my alerts so I ran the debug in the hopes that there would be data to be looked over that may help. I did see a few entries for zigpy while looking over the errors myself.

The first error starts at this line:
"*2024-11-13 02:10:10.979 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140145905083168] Unexpected exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy/device.py", line 381, in request
    return await req.result"*

puddly, 2024-11-14:
@NightHawkATL I'm seeing in the logs no references to ZHA actually connecting or disconnecting from anything. There are, however, enormous flurries of network errors with every integration:

```python
2024-11-13 18:35:02.995 ERROR (MainThread) [msmart.base_device] Network error 10.0.3.81:6444: Transport is closing or closed.
2024-11-13 18:36:07.156 WARNING (MainThread) [homeassistant.components.websocket_api.http.connection] [140145020126112] from 127.0.0.1 (Home Assistant/2024.9.4 (io.robbie.HomeAssistant; build:2024.905; iOS 16.7.10)): Disconnected: Did not receive auth message within 10 seconds
2024-11-13 18:39:43.936 WARNING (MainThread) [alexapy.helpers] alexaapi.get_customer_history_records((<alexapy.alexalogin.AlexaLogin object at 0x7f762c5f9700>,), {'max_record_size': 10}): Timeout error occurred accessing AlexaAPI: An exception of type CancelledError occurred. Arguments: ()
```

Since you're using a PoE coordinator, system-wide network outages will result in ZHA disconnects too.

NightHawkATL, 2024-11-14:
I am not seeing any network outages on my network. I have a Meraki network and I have alerts setup to let me know if there are network issues or outages. I have my network on a UPS so power isn't an issue. I have not seen any full network errors on my end.

NightHawkATL, 2024-11-15:
Looks like it hasn't happened since, so far. Guess it was an odd network fluke.