home-assistant/operating-system #3676: Bluetooth Integration Does Not Recognise MediaTek MT7925

jelliuk, 2024-11-17:
### Describe the issue you are experiencing

Newly installed system based on Proxmox utilising Intel Core Ultra 265k and Gigabyte Z890 EAGLE (REV1.0)

https://www.gigabyte.com/Motherboard/Z890-EAGLE-WIFI7/sp#sp

This is therefore reliant on the MT7925 WiFi/BT Chipset

Utilising HA OS as a VM within Proxmox 8.2 with latest kernel enabled (6.11), passing the BT USB Adapater (extrapolation of the inbuilt WiFi Module) or as a separate test, the PCI device related to the WiFi/BT Module in its entirety to the HA OS VM, the Bluetooth device integration fails to recognise there is a Bluetooth Adapter present.

I've confirmed the module is working correctly via passing the PCI Device to a Windows 11 VM on the same host and can access the functions, I did need to load drivers for Windows 11 to recognise correctly, again due to the relative newness of the hardwarae.

I believe the support for MT7925 wasa introduced in [Kernel 6.7](https://www.phoronix.com/news/GNU-Linux-Libre-6.7).  Noted [here](https://kernelnewbies.org/Linux_6.7) with reference to relevant [commit ](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=c948b5da6bbec742b433138e3e3f9537a85af2e5)hence the lack of support/capability within HAOS despite appropriate declaration in the VM configuration and demonstratable function in alternate VMs.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

13.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Intel Core Ultra 265K
Gigabyte Z890 EAGLE with integrated WiFi/BT Module - MediaTek MT7925
https://www.gigabyte.com/Motherboard/Z890-EAGLE-WIFI7/sp#sp

Proxmox Version: pve-manager/8.2.8/a577cfa684c7476d
Kernel Version: Linux 6.11.0-1-pve (2024-10-23T15:32Z)

```
root@proxmox01:~# uname -r
6.11.0-1-pve

root@proxmox01:~# ls -l /usr/lib/firmware/mediatek/mt7925
total 1754
-rw-r--r-- 1 root root  454959 Oct 24 15:58 BT_RAM_CODE_MT7925_1_1_hdr.bin
-rw-r--r-- 1 root root  211040 Oct 24 15:58 WIFI_MT7925_PATCH_MCU_1_1_hdr.bin
-rw-r--r-- 1 root root 1110312 Oct 24 15:58 WIFI_RAM_CODE_MT7925_1_1.bin

root@proxmox01:~# lsusb
Bus 002 Device 002: ID 174c:1351 ASMedia Technology Inc. USB3.1 Device
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 1a86:55d4 QinHeng Electronics SONOFF Zigbee 3.0 USB Dongle Plus V2
Bus 001 Device 007: ID 0489:e124 Foxconn / Hon Hai Wireless_Device
Bus 001 Device 005: ID 048d:5711 Integrated Technology Express, Inc. GIGABYTE Device
Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@proxmox01:~# udevadm info --query=all --name=/dev/bus/usb/001/007
P: /devices/pci0000:80/0000:80:14.0/usb1/1-14
M: 1-14
R: 14
U: usb
T: usb_device
D: c 189:6
N: bus/usb/001/007
L: 0
V: usb
E: DEVPATH=/devices/pci0000:80/0000:80:14.0/usb1/1-14
E: DEVNAME=/dev/bus/usb/001/007
E: DEVTYPE=usb_device
E: DRIVER=usb
E: PRODUCT=489/e124/100
E: TYPE=239/2/1
E: BUSNUM=001
E: DEVNUM=007
E: MAJOR=189
E: MINOR=6
E: SUBSYSTEM=usb
E: USEC_INITIALIZED=11459509
E: ID_BUS=usb
E: ID_MODEL=Wireless_Device
E: ID_MODEL_ENC=Wireless_Device
E: ID_MODEL_ID=e124
E: ID_SERIAL=MediaTek_Inc._Wireless_Device_000000000
E: ID_SERIAL_SHORT=000000000
E: ID_VENDOR=MediaTek_Inc.
E: ID_VENDOR_ENC=MediaTek\x20Inc.
E: ID_VENDOR_ID=0489
E: ID_REVISION=0100
E: ID_USB_MODEL=Wireless_Device
E: ID_USB_MODEL_ENC=Wireless_Device
E: ID_USB_MODEL_ID=e124
E: ID_USB_SERIAL=MediaTek_Inc._Wireless_Device_000000000
E: ID_USB_SERIAL_SHORT=000000000
E: ID_USB_VENDOR=MediaTek_Inc.
E: ID_USB_VENDOR_ENC=MediaTek\x20Inc.
E: ID_USB_VENDOR_ID=0489
E: ID_USB_REVISION=0100
E: ID_USB_INTERFACES=:e00101:
E: ID_VENDOR_FROM_DATABASE=Foxconn / Hon Hai
E: ID_PATH=pci-0000:80:14.0-usb-0:14
E: ID_PATH_TAG=pci-0000_80_14_0-usb-0_14
```

### Steps to reproduce the issue

N/A - New System - No Bluetooth adapter identified

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A - Nothing appears relevant, happy to provide if requested.
```

### Anything in the Host logs that might be useful for us?

```txt
N/A - Nothing appears relevant, happy to provide if requested.
```

### System information

## System Information

version | core-2024.11.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.54-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1461
Downloaded Repositories | 19

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.11.2
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 30.8 GB
disk_used | 21.7 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | ESPHome (2024.10.3), EMQX (0.7.2), SQLite Web (4.2.2), Studio Code Server (5.17.3), Terminal & SSH (9.15.0), Home Assistant Google Drive Backup (0.112.1), Zigbee2MQTT (1.41.0-1), Zigbee2MQTT Edge (edge), Music Assistant Server (2.3.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 11
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 4 November 2024 at 09:32
-- | --
current_recorder_run | 17 November 2024 at 09:20
estimated_db_size | 667.35 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

Rongronggg9, 2024-11-18:
> Bus 001 Device 007: ID 0489:e124 Foxconn / Hon Hai Wireless_Device

It was not until 3 days ago that the recognition for `0489:e124` was merged into [linux-next](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/bluetooth/btusb.c?id=679cb60fd60774798719c3e449874a168642a8e6). That is to say, until now, every release upstream version lacks the support for your card. 6.13 should be the first release with the support of `0489:e124`. Before that, you may compile the kernel with the above patch applied.

jelliuk, 2024-11-18:
> > Bus 001 Device 007: ID 0489:e124 Foxconn / Hon Hai Wireless_Device
> 
> It was not until 3 days ago that the recognition for `0489:e124` was merged into [linux-next](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/bluetooth/btusb.c?id=679cb60fd60774798719c3e449874a168642a8e6). That is to say, until now, every released upstream version lacks the support for your card. 6.13 should be the first release with the support of `0489:e124`. Before that, you may compile the kernel with the above patch applied.

Thanks for the advice, my searches had indicated previously, I'll see if I can try and work out an approach to handle the above.

The joys of leading edge hardware 😅

closed 2024-11-18 by jelliuk