home-assistant/operating-system #3681: HomeAssistantCore.update blocked from execution, no host internet connection

Jordan-Szwejda, 2024-11-18:
### Describe the issue you are experiencing

I have been doing HA updates regularly. Unfortunately I cannot perform Home Assistant Core update to version 2024.11.2 from 2024.11.2.
I get information that :

"Failed to perform the action update/install. Error updating Home Assistant Core: 'HomeAssistantCore.update' blocked from execution, no host internet connection"

I would be grateful for any help to get updates working again. Everything else works as usual.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 13.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Installed a few years ago on Proxmox. Nothing has been touched recently apart from installing HA updates.

### Steps to reproduce the issue

After clicking on web gui: Setting -> Home Assistant Core Update -> Update
I get "Failed to perform the action update/install. Error updating Home Assistant Core: 'HomeAssistantCore.update' blocked from execution, no host internet connection" message.

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-11-18 22:51:43.662 INFO (MainThread) [__main__] Running Supervisor
2024-11-18 22:51:43.676 INFO (MainThread) [supervisor.os.manager] Rauc: slot A - activated slot kernel.0, marked slot kernel.0 as good
2024-11-18 22:51:43.681 INFO (MainThread) [supervisor.addons.manager] Phase 'initialize' starting 0 add-ons
2024-11-18 22:51:43.686 INFO (MainThread) [supervisor.addons.manager] Phase 'system' starting 1 add-ons
2024-11-18 22:51:44.625 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-mosquitto with version 6.4.1
2024-11-18 22:51:46.708 INFO (MainThread) [supervisor.services.modules.mqtt] Set core_mosquitto as service provider for mqtt
2024-11-18 22:51:49.627 INFO (MainThread) [supervisor.addons.manager] Phase 'services' starting 4 add-ons
2024-11-18 22:51:49.796 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-samba with version 12.3.2
2024-11-18 22:51:50.049 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/influxdb/amd64 with version 5.0.1
2024-11-18 22:51:50.097 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_samba
2024-11-18 22:51:50.356 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/grafana/amd64 with version 10.2.1
2024-11-18 22:51:50.659 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/esphome/esphome-hassio with version 2024.10.3
2024-11-18 22:51:51.724 INFO (MainThread) [supervisor.api.middleware.security] /dns/info access from a0d7b954_influxdb
2024-11-18 22:52:04.285 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'mqtt-client'
2024-11-18 22:52:04.285 INFO (MainThread) [supervisor.auth] Home Assistant not running, checking cache
2024-11-18 22:52:20.379 INFO (MainThread) [supervisor.core] Start Home Assistant Core
2024-11-18 22:52:20.388 INFO (SyncWorker_1) [supervisor.docker.manager] Starting homeassistant
2024-11-18 22:52:20.500 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
2024-11-18 22:52:23.722 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state startup
2024-11-18 22:52:23.723 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-11-18 22:52:25.545 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-11-18 22:52:25.548 INFO (MainThread) [supervisor.homeassistant.core] Home Assistant Core state changed to APIState(core_state='NOT_RUNNING', offline_db_migration=False)
2024-11-18 22:52:29.029 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'mqtt-broker'
2024-11-18 22:52:29.032 INFO (MainThread) [supervisor.auth] Home Assistant not running, checking cache
2024-11-18 22:52:45.572 INFO (MainThread) [supervisor.homeassistant.core] Home Assistant Core state changed to APIState(core_state='STARTING', offline_db_migration=False)
2024-11-18 22:52:50.579 INFO (MainThread) [supervisor.homeassistant.core] Home Assistant Core state changed to APIState(core_state='RUNNING', offline_db_migration=False)
2024-11-18 22:52:50.579 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
2024-11-18 22:52:50.579 INFO (MainThread) [supervisor.addons.manager] Phase 'application' starting 2 add-ons
2024-11-18 22:52:50.828 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/sabeechen/hassio-google-drive-backup-amd64 with version 0.112.1
2024-11-18 22:52:51.058 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-piper with version 1.5.2
2024-11-18 22:52:51.615 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from cebe7a76_hassio_google_drive_backup
2024-11-18 22:52:51.618 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
2024-11-18 22:52:51.620 INFO (MainThread) [supervisor.api.middleware.security] /mounts access from cebe7a76_hassio_google_drive_backup
2024-11-18 22:52:51.621 INFO (MainThread) [supervisor.api.middleware.security] /addons access from cebe7a76_hassio_google_drive_backup
2024-11-18 22:52:51.625 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
2024-11-18 22:52:51.626 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
2024-11-18 22:52:51.627 INFO (MainThread) [supervisor.api.middleware.security] /backups/9ab889bb/info access from cebe7a76_hassio_google_drive_backup
[... 141 lines cut ...]
version_api | ok
installed_addons | Samba share (12.3.2), Let's Encrypt (5.2.3), InfluxDB (5.0.1), Grafana (10.2.1), Mosquitto broker (6.4.1), Home Assistant Google Drive Backup (0.112.1), Piper (1.5.2), ESPHome (2024.10.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 4
views | 13
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | November 8, 2024 at 21:52
-- | --
current_recorder_run | November 18, 2024 at 22:52
estimated_db_size | 410.51 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

Jordan-Szwejda, 2024-11-20:
I've just managed to resolve an issue with Home Assistant updates. The solution is rather unusual, and I suspect there might be a bug within the system. Here's a breakdown of the problem and my workaround:

To enable integration with Google Nest, I expose Home Assistant to my public IP address and use Let's Encrypt to generate valid SSL certificates periodically. This requires HTTP-based validation, so I forward port 80 from my router to port 8090 on the Home Assistant host, where Let's Encrypt is configured for HTTP challenges.

However, with this forwarding rule active, Home Assistant OS is unable to perform updates. As soon as I disable the rule and reboot the host, updates resume normally.

I hope this information proves helpful to others encountering similar update issues or provides insights for developers to identify and rectify this peculiar Home Assistant behavior.

emerysteele, 2024-11-29:
Confirming same issue. Had port forwarding (80/HTTP) setup on my router to allow lets encrypt to work.

Did a bunch of stuff trying to troubleshoot. Even setup a new HAOS instance on a completely different machine with different IP address from my main instance. And had issues even on that machine trying to update the core, getting error: `'HomeAssistantCore.update' blocked from execution, no host internet connection`

Disabling the port 80 forwarding rule on my router & rebooting both HomeAssistant instances got updates working again.

Also seems to affect certain integrations. Specifically I was having issues with the TotalConnect integration.

sairon, 2024-12-04:
That doesn't make any sense to me, in fact it makes me believe the problem is in the firewall configuration of the router. Can you @Jordan-Szwejda and @emerysteele share what router are you using and how you configure the forwarding rule?

The host internet test is performed by NetworkManager periodically (every 10 minutes) by doing an HTTP request to `http://checkonline.home-assistant.io/online.txt` - you can check if you can access it from HAOS, and from other machines on the network, when the firewall/forwarding rule is enabled. Forwarding port 80 of your router's WAN interface to a device in LAN should not affect _outgoing_ traffic on that port.

Jordan-Szwejda, 2024-12-04:
I've been using MikroTik RouterOS 7.8 (https://www.mikrotik.com/). Here are the firewall rules I configured on the router. It's worth mentioning that I have a hairpin NAT rule set up. This allows me to access Home Assistant using the same domain address from both the external network (internet) and the local network. In both cases, I can use the same HTTPS URL for Home Assistant with a valid certificate.
[firewall_rules_mikrotik.txt](https://github.com/user-attachments/files/1
[image]
8013693/firewall_rules_mikrotik.txt)

Let me know if you need anything more.

sairon, 2024-12-04:
@Jordan-Szwejda At first sight the rules are looking fine and AFAIK nothing changed recently that should change the behavior there, but you know, you can never rule out the butterfly effect :sweat_smile: Anyway the host internet connection is transparently taken from the NetworkManager and you can check `ha network info` if it currently thinks it's online. Anyway, if the issue comes back again, performing the checks I suggested could take us further.

emerysteele, 2024-12-05:
I am using OPNSense.

Did some further troubleshooting, guess the port forwarding is a bit different than I'm used to. Thought it would only apply to inbound WAN traffic, but without setting specifically to 'WAN address' for destination, it also applies to outbound WAN traffic as well.

Guess I didn't notice cause most stuff is HTTPS/port 443 nowadays for web traffic.

[image]

Jordan-Szwejda, 2024-12-05:
When I enable the forward rule then `network info` command starts returning 'host_internet: false' after a while whereas
`curl https://checkonline.home-assistant.io/online.txt` returns 'NetworkManager is online' response all the time.
Screenshot attached:

[image]

emerysteele, 2024-12-05:
@Jordan-Szwejda try setting your In. Interface List to 'WAN' for the port 80 forwarding rule.

[image]

Jordan-Szwejda, 2024-12-06:
@emerysteele Filtering 'In Interface' to WAN helps in this particular situation. Another thing is what really happens there that HAOS thinks there is no internet connection.
According to what @sairon said about NetworkManager there should not be broken internet connection reported due to the fact that I am able to retrieve correct response from http://checkonline.home-assistant.io/online.txt as it was presented on the screenshot in the previous comment.

emerysteele, 2024-12-06:
You used https in your curl command which is port 443, not 80.

Jordan-Szwejda, 2024-12-07:
Right, if I use http instead of https then I get an error. The reason why I used https is that I found this url in the supervisor repository defined for connectivity check: https://github.com/home-assistant/supervisor/blob/main/supervisor/supervisor.py

[image]

Anyway, that's a router configuration issue and I believe the thread can be closed. Thank you for participating in this thread and your help.

closed 2024-12-07 by Jordan-Szwejda