home-assistant/operating-system #3684: ODROID N2+ don't boot

Hadramos, 2024-11-20:
### Describe the issue you are experiencing

Hello,

I received my ODROID N2+ with 64GB eMMC Module N2 and I did a fresh install of Home Assistant with this link.

https://www.home-assistant.io/installation/odroid

My ODROID does not boot, I have this message:

Card did not respond to voltage select! : -110
unable to select a mode : -5
MMC Device 2 not found
no mic device at slot 2

However the module works perfectly when I use the USB adapter and connect it to my Linux Debian.

I have configured the switch on the Odroid N2 on MMC.

With Ubuntu OS Image I have no problem…
https://dn.odroid.com/S922X/ODROID-N2/Ubuntu/ubuntu-20.04-4.9-minimal-odroid-n2-20220228.img.xz

I just retested by installing the HA image haos_odroid-n2-13.2.img with balenaEtcher but my ODROID does not boot.

I tested with the ODROID image ubuntu-22.04-4.9-mate-odroid-n2-20220629.img && ubuntu-22.04-4.9-mate-odroid-n2-20220629.img and I have no boot problem.

Isn’t there a problem on the HA image with an eMMC module on an ODROID N2+?

Do you have the same problem?

Regards

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

core-2024.11.0 && core-2024.11.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

ODROID N2+
eMMC Linux RED 64Gb

### Steps to reproduce the issue

1. Install haos_odroid-n2-13.2.img into my eMMC with USB Adapter
2. Boot with eMMC bur don't boot

### Anything in the Supervisor logs that might be useful for us?

```txt
Card did not respond to voltage select! : -110
unable to select a mode : -5
MMC Device 2 not found
no mic device at slot 2
```

### Anything in the Host logs that might be useful for us?

```txt
Card did not respond to voltage select! : -110
unable to select a mode : -5
MMC Device 2 not found
no mic device at slot 2
```

### System information

Version	core-2024.11.0
Type d'installation	Home Assistant OS
Développement	false
Supervisor	true
Docker	true
Utilisateur	root
Environnement virtuel	false
Version de Python	3.12.4
Famille du système d'exploitation	Linux
Version du système d'exploitation	6.6.54-haos
Architecture du processeur	aarch64
Fuseau horaire	Europe/Paris
Répertoire de configuration	/config

sairon, 2024-11-20:
I'm not sure what you mean by "eMMC Linux RED 64Gb" but the symptoms look very much like #3250. Can you check with the latest 14.0.rc1?

Hadramos, 2024-11-20:
This module : https://www.hardkernel.com/shop/64gb-emmc-module-n2-linux/

Very well I will test with the latest release 14.0.rc1 but I just retested and I have the same problem at boot with the following versions 13.0 - 13.1 - 13.2

Hadramos, 2024-11-20:
I like HA and I have been using it for a long time with another ODROID N2 (not +) but with an ssd card. I recently bought another ODROID N2+ with the eMMC module to have more power and that there is only HA installed on it. I will be there if you need more testing

Hadramos, 2024-11-20:
I didn't know that not all versions of HA were compatible with a Kingston eMMC module.
[image]

sairon, 2024-11-20:
That doesn't seem right, after the `unable to select a mode` line, there should be `Card failed to initialize at 24000000 Hz, disabling meson_gx hack.` one. It seems that you didn't flash the 14.0.rc1 to the eMMC properly, this is not U-Boot from that version.

Hadramos, 2024-11-20:
Sorry, i changed my screen print with my camera

Hadramos, 2024-11-20:
[image]

Hadramos, 2024-11-21:
Hi @sairon it's ok for you my last picture?

sairon, 2024-11-21:
@Hadramos Yes, it's clear. It seems that for some reason the 64 GB module behaves differently than other Kingston eMMCs that we have tested (32 GB and 128 GB variants, where the 128 GB one even worked with older HAOS versions). I will try to get my hands on the 64 GB module as well and check what's going on. Unfortunately, the interim solution would be to use other eMMCs that are proven to be working - if you still can return this one, I will suggest doing that.

Can you just double-check it really has is a Kingston chip? It should look like this with the sticker removed, but even with the sticker the part of the "Kingston" logo should be prominent:
[image]

Hadramos, 2024-11-21:
Yes, I can confirm that it is indeed a Kingston module. I looked yesterday. 
I'll take a photo for you this evening, I'm not there today.

Hadramos, 2024-11-21:
Do you have any sites other than Hardkernel to order a 128GB eMMC? With Hardkernel I have to pay customs fees

sairon, 2024-11-21:
Check the official [distributors list](https://www.hardkernel.com/distributors/). I got mine from RPishop.cz (they're also shipping internationally to Europe) but there might be some distributor closer to you.

Hadramos, 2024-11-22:
Hi, attached a photo of my eMMC Kingston module

[image]

sairon, 2024-11-25:
Hardkernel kindly provided me with 64GB eMMC modules for testing, unfortunately, I'm unable to reproduce the issue here with my N2+ - unlike 128 GB modules, it also has the same issue as 32 GB modules but it's mitigated with the fallback. Tried that couple of times and always booted fine.
```
Card did not respond to voltage select! : -110
unable to select a mode : -5
Card failed to initialize at 24000000 Hz, disabling meson_gx hack.
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
3451 bytes read in 2 ms (1.6 MiB/s)
## Executing script at 08000000
switch to partitions #0, OK
mmc1(part 0) is current device
loading env...
```

It appears the eMMC chip has the same marking as yours (but it's hard to read - wiping it from the glue residue with alcohol can help).

[image]

What revision of the ODROID you have (written next to the USB ports)? For reference, my is `REV 0.6 20220517`.

Hadramos, 2024-11-26:
Hello and thank you.

The revision of my ODROID is **REV 0.6 20220517**

I took a photo of my eMMC which is much better.
The only difference I see is the last line

**4019005.01**

[image]
[image]

closed 2025-03-04 by github-actions[bot]