home-assistant/operating-system #3690: UPB PIM (USB) no drivers in HAOS 13.2 x86-64

dtomcat, 2024-11-21:
### Describe the issue you are experiencing

Migrated HA from Pi3B+ to intel NUC. Both are HAOS 13.2.  Migration went well except the UPB does not load a driver (does on pi). Tried rebooting, different ports, and different cables. “dmesg” shows it sees it and labels it properly.  But stops there and no drivers loaded 
[image]
 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

13.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Simply Automated inc UPB PIM (USB)

### Steps to reproduce the issue

1. Plug in UPB PIM to USB port
2. Read dmesg

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2024.11.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.54-haos
arch | x86_64
timezone | America/Denver
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | July 18, 2025 at 6:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | us-east-1-11.ui.nabu.casa
certificate_status | ready
instance_id | fb60cc31251f4967b199891e207e5df6
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.11.4
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 54.3 GB
disk_used | 7.7 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (6.6.0), Mosquitto broker (6.4.1), OpenThread Border Router (2.12.1), File editor (5.8.0), Terminal & SSH (9.15.0), FTP (5.1.1), ESPHome Device Compiler (2024.11.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | November 11, 2024 at 11:20 AM
-- | --
current_recorder_run | November 21, 2024 at 12:58 PM
estimated_db_size | 269.21 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

sairon, 2024-11-22:
What other driver should be loaded? It is registered using the `cypress_m8` driver which treats it as "HID->COM RS232 Adapter". I don't see any specific handling in the Raspberry Pi kernel, so I'd expect that it works there the same there. As such, it should appear e.g. in `/dev/by-serial/by-id`.

dtomcat, 2024-11-22:
Apologies… that’s the image from pi.  Nuc image:
[image]

sairon, 2024-11-22:
And my apologies as well, at first I have not checked the kernel config for generic-x86-64 and `CONFIG_USB_SERIAL_CYPRESS_M8` is indeed not enabled there. That will be an easy fix.

closed 2024-11-22 by sairon

dtomcat, 2024-11-22:
Thank you.  I’ll test it when released    I’ll report back with results, but I’m sure it’ll be good

Closed #3690<https://github.com/home-assistant/operating-system/issues/3690> as completed via d57e507<https://github.com/home-assistant/operating-system/commit/d57e50776499f209e013efa0db6cb3c7f46f51c3>.

dtomcat, 2024-11-25:
Confirmed working. Thank you again!

PR #3693 (2024-11-22): Enable Cypress M8 USB serial driver for all targets
Build cypress_m8 driver as module for all targets - some of them had it in their base defconfig while some not. It is required e.g. for UPB PIM (Powerline Interface Module).

Fixes #3690

## Summary by CodeRabbit

- **New Features**
  - Added support for Cypress M8 USB serial devices, enhancing compatibility with additional USB hardware.
  
- **Improvements**
  - Expanded USB-related configurations to improve functionality and support for various USB devices.
  - Updated multimedia subsystem options for better video and audio processing capab [... cut]
commit d57e5077 (2024-11-22): Enable Cypress M8 USB serial driver for all targets (#3693)

Build cypress_m8 driver as module for all targets - some of them had it in
their base defconfig while some not. It is required e.g. for UPB PIM (Powerline
Interface Module).

Fixes #3690
commit 55bbddad (2024-12-03): Enable Cypress M8 USB serial driver for all targets (#3693)

Build cypress_m8 driver as module for all targets - some of them had it in
their base defconfig while some not. It is required e.g. for UPB PIM (Powerline
Interface Module).

Fixes #3690

(cherry picked from commit d57e50776499f209e013efa0db6cb3c7f46f51c3)