home-assistant/operating-system #3692: Use GRUB2 legacy loader for 2 older Chips

MYF540, 2024-11-22:
### Describe the issue you are experiencing

Same issue as described in #3305. This time, however, it affects an HP T640 Thin Client with a Ryzen Embedded R1505G CPU (_Originally posted by @okh-mzny in https://github.com/home-assistant/operating-system/issues/3305#issuecomment-2381015846_) and a Lenovo ThinkCentre M715q with a Ryzen 5 2400GE (_Originally posted by @MYF540 in https://github.com/home-assistant/operating-system/issues/3305#issuecomment-2489297609_). 
Just reported this Issue because the other one is already closed.

Ryzen 5 2400GE
`grub> smbios --type 4 --get-qword 8
1696726757278813968`

Ryzen Embedded R1505G
`grub> smbios --type 4 --get-qword 8 
1696726757278814081`

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.33

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Lenovo ThinkCentre M715q Ryzen 5 2400GE and HP T640 Thin Client Ryzen Embedded R1505G

### Steps to reproduce the issue

Try to install or update newer versions of HA on mentioned bare metal.

### Anything in the Supervisor logs that might be useful for us?

```txt
no logs
```

### Anything in the Host logs that might be useful for us?

```txt
no logs
```

SellSan, 2024-12-03:
I have the same problem on my HP T610 WW Thin Client. AMD G-T56N Processor.
I'm not sure this helps, but mine is:
grub> smbios --type 4 --get-qword 8
22534628645272575

MYF540, 2024-12-12:
@sairon you seem to have created the previous patch related to this issue. Could you please add the following three processor IDs as well? 
(https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/patches/grub2/0002-loader-efi-linux-use-legacy-loader-for-broken-Intel-.patch)

```
|| processor_id == 0x178bfbff00730f00   // AMD Ryzen 5 2400GE
|| processor_id == 0x178bfbff00730f81   // AMD Ryzen Embedded R1505G
|| processor_id == 0x500f01007ffffff);  // AMD G-T56N
```

I can convert the CPU IDs to hex, but I’m not sure if there are any other parts of your patch that would also need adjustments.

Thanks in advance!

iosonogerva, 2025-01-26:
I probably posted a [comment ](https://github.com/home-assistant/operating-system/issues/3305#issuecomment-2612892627)in the wrong issue(#3305) the other day, so i will repost here:

> Hi everyone, i was trying to give home assistant a try by installing it on an old NUC i have lying around but i stumbled upon the problem described here. The NUC is a DE3815TYKE with an Intel Atom E3815. The output from `smbios --type 4 --get-qword 8` is `13829424153406670451`

I can also add i-m trying to install the latest (at this time) version: `haos_generic-x86-64-14.1`
Sorry for the duplicate comment.
Thanks in advance.

closed 2025-05-04 by github-actions[bot]