home-assistant/operating-system #3699: RPI 5 network at 100Mbps instead of 1Gbps

jacauc, 2024-11-27:
### Describe the issue you are experiencing

I am running HA OS on an RPI 5 with NVMe SSD drive. 
The network cable is connected to a 1Gbps switch and no matter what I do on the switch (Auto, forced 1Gbps Full Duplex), the HA OS always negotiates at 100Mbps FD.

How can I make HA OS utilize the Gigabit potential of the hardware? 
This is particularly noticeable on my internet Speed Tests which are all "capped" at 100Mbps.

I've tried updating HAOS to the latest Beta and that also had no effect.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

14.0 rc2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Raspberry Pi 5

rpi5-64

nvme-KINGSTON_SNV2S500G

usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240124083207-if00

### Steps to reproduce the issue

1. Connect RPI5 with HAOS to 1Gbps network port
2. Check actual network speed

### Anything in the Supervisor logs that might be useful for us?

```txt
NA
```

### Anything in the Host logs that might be useful for us?

```txt
NA
```

### System information

## System Information

version | core-2024.11.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.51-haos-raspi
arch | aarch64
timezone | Europe/Lisbon
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1467
Downloaded Repositories | 30

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 45

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.0.rc2
-- | --
update_channel | beta
supervisor_version | supervisor-2024.11.4
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 457.7 GB
disk_used | 20.7 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (19.0.0), Mosquitto broker (6.4.1), Duck DNS (1.18.0), Let's Encrypt (5.2.5), File editor (5.8.0), ESPHome Device Compiler (2024.11.1), Home Assistant Google Drive Backup (0.112.1), Assist Microphone (1.2.0), Music Assistant Server (2.3.3), Studio Code Server (5.17.3), Uptime Kuma (0.12.4), Node-RED (18.1.1), AdGuard Home (5.1.5), Zigbee2MQTT (1.41.0-1), Plex Media Server (3.6.0), Silicon Labs Flasher (0.3.1), Cloudflared (5.2.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 18
views | 15
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 16 November 2024 at 22:38
-- | --
current_recorder_run | 27 November 2024 at 07:07
estimated_db_size | 183.04 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

jacauc, 2024-11-27:
might be related to this old post: https://github.com/home-assistant/core/issues/107849

sairon, 2024-12-02:
Sorry, but we can't move forward without that. You don't have to "blow away your primary HA instance", all you need is an micro SD card flashed with Raspberry Pi OS. Since auto-negotiation happens on the physical layer, and to my knowledge there's nothing specific to HAOS in this part of this stack, checking with Raspberry Pi OS is the easiest way to confirm this is not an upstream driver issue or a hardware defect.

jacauc, 2024-12-02:
My RPi is configured to boot from NVMe, so it's not as easy as that. Thanks anyways. We'll close the issue if we can't move forward. I'll figure out a way to move the instance to a VM instead.

sairon, 2024-12-02:
Even then, it should be possible to perform the test by simply flashing RPi OS to an SD card (and then HA OS as well), and checking what link speed is negotiated in both cases. The NVMe shouldn't make a difference.

closed 2024-12-02 by jacauc

francois-le-ko4la, 2024-12-17:
I get it with a Rasp 4.
https://github.com/francois-le-ko4la/rasp-fix-eth0/